raspberrypi/bookworm-feedback #251: getty prevents serial port usage

ZL4JY, 2024-05-14:
Despite disabling the console using raspi-config, getty still takes control of both ttyS0 and ttyAMA0.

Version:

Raspberry Pi OS with desktop
Release date: March 15th 2024
System: 64-bit
Kernel version: 6.6
Debian version: 12 (bookworm)

ZL4JY, 2024-07-01:
RP 5

lurch, 2024-07-02:
I'm unable to reproduce this. Could you please give **exact** reproduction steps? (e.g. what does "Follow usual steps to enable ttyS0 or ttyAMA0" mean?)

Here's what I did (on a fully up-to-date 64-bit Raspberry Pi OS Bookworm image) on a Pi5:
* ran `sudo raspi-config`
* selected "3 Interface Options"
* selected "I6 Serial Port"
* answered "No" when asked "Would you like a login shell to be accessible over serial?"
* answered "Yes" when asked "Would you like the serial port hardware to be enabled?"
* selected "OK" and "Finish" to exit raspi-config
* rebooted as prompted
* after reboot, `ls -l /dev/ttyAMA0` shows the permissions as `crw-rw----`
* ran the `sudo chmod 777 /dev/ttyAMA0` that you suggested
* `ls -l /dev/ttyAMA0` now shows the permissions as `crwxrwxrwx`
* I waited a couple of minutes, and `ls -l /dev/ttyAMA0` still shows the permissions as `crwxrwxrwx`
* `systemctl status serial-getty@ttyAMA0` lists the service as both `disabled` and `inactive (dead)`