raspberrypi/bookworm-feedback #252: Apt installs python3-pil when asked to install python3-pillow

ChrisHodgesUK, 2024-05-14:
### Describe the bug

Pillow has capabilities that PIL doesn't, in my case ImageTk.  However (even after removing python3-pil with apt), it's not possible to install pillow.  The failure is rather subtle because it claims a successful installation, but buried in the installation output is `Note, selecting 'python3-pil' instead of 'python3-pillow'`, and sure enough I get the same `ImportError: cannot import name 'ImageTk' from 'PIL'  ` when I try to use pillow.

It should install the package I asked for, not guess at a lesser substitute.  If there's a good reason not to install the package I wanted, it should throw an error at installation, not sneakily install something else

### Steps to reproduce the behaviour

`sudo apt remove python3-pil`
`sudo apt install python3-pillow`

To confirm:
`python`
`>>> from PIL import ImageTk`

### Device (s)

Raspberry Pi 3 Mod. B+, Raspberry Pi 5

### System

Raspberry Pi reference 2023-12-05
Generated using pi-gen, https://github.com/RPi-Distro/pi-gen, 02b371de1a87bbe70cfcbd498527067a7d2ee6ca, stage4

2024/04/20 11:53:30 
Copyright (c) 2012 Broadcom
version d1744d21 (release) (embedded)

Linux raspberrypi 6.6.28+rpt-rpi-v8 raspberrypi/linux#1 SMP PREEMPT Debian 1:6.6.28-1+rpt1 (2024-04-22) aarch64 GNU/Linux

popcornmix, 2024-05-14:
Moved to bookworm-feedback repo as this issue has nothing to do with linux kernel.

XECDesign, 2024-05-14:
It doesn't have anything to do without anything we've done with bookworm either. The issue should be taken up with Debian and they will have good reasons they've decided to do it this way.

XECDesign, 2024-05-14:
Taking a closer look, python3-pil pillow and the ImageTk bit comes from the python3-pil.imagetk package.

ChrisHodgesUK, 2024-05-14:
> Taking a closer look, python3-pil IS pillow and the ImageTk bit comes from the python3-pil.imagetk package.

More subtle than I (could have) guessed, which also makes for a pretty simple workaround.  The different packaging such that Pillow (deb) is a subset of Pillow (via pip) still doesn't seem like sensible/expected behaviour.

ChrisHodgesUK, 2024-05-14:
> Moved to bookworm-feedback repo as this issue has nothing to do with linux kernel.

Oops, sorry.  The link I followed didn't make that clear.  Thanks for moving it.

popcornmix, 2024-05-14:
@ChrisHodgesUK what link did you follow?

closed 2024-05-14 by lurch