raspberrypi/bookworm-feedback #253: Wayfire: cannot close mousepad text editor window when changed window position

qrp73, 2024-05-17:
Steps to reproduce:
1) Start mousepad (menu Accessories->Text Editor)
2) enter some text, for example "111"
3) move mousepad window down in such way so it's middle part appears outside of screen
4) Click on window close button

Expected result: Save dialog appears on the screen within bounds

Actual result: Save dialog appears outside of the screen with no way to click on it and no way to close mousepad window
[image]

lurch, 2024-07-01:
Can confirm this behaviour, it seems as though the "Do you want to save the changes before closing?" dialog appears outside of the visible-area of the screen if there's more of the mousepad window "outside" of the screen than "inside" the screen.

However there **is** a workaround - if you hold down the Windows key on your keyboard (or the Raspberry key on a Pi keyboard) whilst clicking&dragging the mousepad window, you can drag the mousepad window so that it's back in the "visible" part of the screen, and then when you release the mouse-button the dialog "pops" back to being visible on the screen.

lurch, 2024-07-01:
I've just tested this with the beta Labwc compositor (see https://forums.raspberrypi.com/viewtopic.php?t=370005 ), and that doesn't exhibit this problem - the "Do you want to save the changes before closing?" dialog is always visible on-screen regardless of how much of the mousepad window is off-screen.

lurch, 2025-01-14:
Raspberry Pi OS now uses labwc by default, and under labwc the "Do you want to save the changes before closing?" dialog always appears on-screen, regardless of how much of the mousepad window is off-screen. So closing this issue.

closed 2025-01-14 by lurch