raspberrypi/bookworm-feedback #254: Appearance Settings / pipanel fails on headless Raspi. bug?

sgalke, 2024-05-17:
After latest bookworm update pipanel does not open the "Appearance Settings" on headless Desktop.
invoked from shell results in following error on RPi 3A+ / latest bookworm 32 Bit:

(pipanel:25196): Gtk-CRITICAL **: 20:53:49.309: gtk_file_chooser_select_filename: assertion 'filename != NULL' failed

In contrast: no error on *not yet* updated headless RPi 3B+ / bookworm 64 Bit, "Appearance Settings" window pops up.

sgalke, 2024-05-17:
... running X11 in both cases.

sgalke, 2024-05-20:
verified today on fresh install of bookworm 32-bit on pi400 *after* update: the same error ("Gtk-CRITICAL....") occurs.

sgalke, 2024-05-22:
... tested on Pi5: same error!

sgalke, 2024-05-22:
workaround under bookworm / X11:

change in /boot/firmware/config.txt:

```
# Enable DRM VC4 V3D driver
dtoverlay=vc4-fkms-v3d
[pi5]
dtoverlay=vc4-kms-v3d
[all]

```
this does not help on Pi5, but on older models the Appearance Setting will again show up on headless PI's.
maybe you have to set screen resolution in config.txt explicit:

```
hdmi_force_hotplug=1
hdmi_ignore_edid=0xa5000080
hdmi_group=2
hdmi_mode=82

```

Joaav, 2024-06-14:
I have the same issue.

Setting dtoverlay=vc4-kms-v3d did not change anything, with dtoverlay=vc4-fkms-v3d set up there is no display visible in VNC.

Any ideas how to fix it?

kirichenec, 2024-06-25:
same on pi5

Joaav, 2024-06-26:
I ended up plugging an HDMI dummy into the rpi as a walk around and it’s working well now. What is interesting is that whole GUI started working faster - probably no scaling issues caused that

marko-pi, 2024-07-05:
Is there a possibility to do these things by editing some file?  I just want to remove picture.

quinnell, 2024-07-07:
Still no fixes for Pi5?

Joaav, 2024-07-07:
> Is there a possibility to do these things by editing some file?  I just want to remove picture.

Just plug it into something for a moment then

marko-pi, 2024-07-08:
> > Is there a possibility to do these things by editing some file?  I just want to remove picture.
> 
> @marko-pi Just plug it into something for a moment then

Today, me and most of my colleagues use notebooks, so I don't have HDMI monitor at disposal.

Removing the picture by editing some setting file would be very beneficial.

ghollingworth, 2024-07-09:
If this is only a problem on X then it's unlikely to get fixed any time soon.  Why don't you just use Wayfiire/labwc instead as compositor?

sgalke, 2024-07-09:
i needed realvnc under all circumstances. 
vnc was not working under wayland, so i switched to x11. 
meanwhile vnc works reliably under wayland, so my problem is no longer existent. 
i close the issue.

closed 2024-07-09 by sgalke

zedstarr, 2024-07-12:
I did see this problem on a Pi5 running bookworm/wayland. Not sure if it's significant but my route to desktop was via a bookworm Lite OS install then I installed "xserver-xorg raspberrypi-ui-mods" over the top.

Today as part of some other work I booted the Pi with the official Raspberry Pi 7" touch screen display attached (HDMI disconnected) - it proved too small to be useful at 800 x 480 and with no way of changing settings via the GUI I powered off, disconnected the touch display and went back to the 1920 x 1080 HDMI monitor I'd been using previously.

I then noticed if ssh'd into the Pi from my Fedora laptop with "ssh -X" I could run pipanel and the dialog box would display on the laptop's screen fine. Back on the Pi running "pipanel" would still fail with `Gtk-CRITICAL **: 11:43:00.186: gtk_file_chooser_select_filename: assertion 'filename != NULL' failed` but I noticed an extra `sh: 1: wlr-randr: not found
` I hadn't seen before. Installed wlr-randr and pipanel now runs/displays on the Pi... 

So - not sure if installing desktop over Lite OS image is significant in causing the error in the first place or auto-detection of 2nd display helped things along in highlighting wlr-randr missing? Pretty sure the wlr-randr error wasn't there initially.