raspberrypi/bookworm-feedback #256: wmctrl fails with "cannot execute: required file not found"

HamishB, 2024-05-22:
### Describe the bug

Hi, I'm trying to run `wmctrl` on a RPi 3B with Raspian-Bookworm to tell a window to maximize (it's a kiosk display), but it is failing:

```
 wmctrl 
-bash: /usr/bin/wmctrl: cannot execute: required file not found
```

Works fine on regular Debian Bookworm amd64 on my desktop computer.
I don't have a non-RPi arm Debian system to test it on.

Some websearching indicated that maybe it was a problem with 32/64 bit cross compiling, or a failure to find the dynamic loader.

Debian bug #362068 speaks of 64/32bit changes in wmctrl breaking other stuff.

Nothing jumping out at me in ldd. Other stuff on the RPi all seems to work fine.

```
$ file `which wmctrl`
/usr/bin/wmctrl: ELF 32-bit LSB executable, ARM, EABI5 version 1 (SYSV), dynamically linked, interpreter /lib/arm-linux-gnueabihf/ld-linux.so.3, for GNU/Linux 2.6.26, BuildID[sha1]=c73e024bc544b26de8c591016e5ffa6de17dddb3, stripped
```

```
$ ldd `which wmctrl`
        linux-vdso.so.1 (0x7eede000)
        /usr/lib/arm-linux-gnueabihf/libarmmem-${PLATFORM}.so => /usr/lib/arm-linux-gnueabihf/libarmmem-v7l.so (0x76f20000)
        libX11.so.6 => /lib/arm-linux-gnueabihf/libX11.so.6 (0x76dab000)
        libXmu.so.6 => /lib/arm-linux-gnueabihf/libXmu.so.6 (0x76d80000)
        libglib-2.0.so.0 => /lib/arm-linux-gnueabihf/libglib-2.0.so.0 (0x76c64000)
        libc.so.6 => /lib/arm-linux-gnueabihf/libc.so.6 (0x76aeb000)
        /lib/arm-linux-gnueabihf/ld-linux.so.3 => /lib/ld-linux-armhf.so.3 (0x76f3e000)
        libxcb.so.1 => /lib/arm-linux-gnueabihf/libxcb.so.1 (0x76ab0000)
        libXt.so.6 => /lib/arm-linux-gnueabihf/libXt.so.6 (0x76a5d000)
        libXext.so.6 => /lib/arm-linux-gnueabihf/libXext.so.6 (0x76a30000)
        libpcre2-8.so.0 => /lib/arm-linux-gnueabihf/libpcre2-8.so.0 (0x76980000)
        libm.so.6 => /lib/arm-linux-gnueabihf/libm.so.6 (0x76939000)
        libXau.so.6 => /lib/arm-linux-gnueabihf/libXau.so.6 (0x76920000)
        libXdmcp.so.6 => /lib/arm-linux-gnueabihf/libXdmcp.so.6 (0x76900000)
        libSM.so.6 => /lib/arm-linux-gnueabihf/libSM.so.6 (0x768e0000)
        libICE.so.6 => /lib/arm-linux-gnueabihf/libICE.so.6 (0x768b0000)
        libbsd.so.0 => /lib/arm-linux-gnueabihf/libbsd.so.0 (0x76880000)
        libuuid.so.1 => /lib/arm-linux-gnueabihf/libuuid.so.1 (0x76a52000)
        libmd.so.0 => /lib/arm-linux-gnueabihf/libmd.so.0 (0x76860000)
```

```
$ ls -l /lib/arm-linux-gnueabihf/ld-linux-armhf.so.3
-rwxr-xr-x 1 root root 167464 May  7 19:37 /lib/arm-linux-gnueabihf/ld-linux-armhf.so.3
```

### Steps to reproduce the behaviour

```
$ sudo apt install wmctrl
$ wmctrl
-bash: /usr/bin/wmctrl: cannot execute: required file not found
```

### Device (s)

Raspberry Pi 3 Mod. B

### System

```
$ cat /etc/rpi-issue 
Raspberry Pi reference 2023-10-10
Generated using pi-gen, https://github.com/RPi-Distro/pi-gen, fb56ad562991cf3ae5c96ab50983e1deeaefc7b6, stage4
```

```
$ vcgencmd version
Aug 10 2023 15:35:01 
Copyright (c) 2012 Broadcom
version 03dc77429335caee083e22ddc8eec09c07f12a7a (clean) (release) (start)
```

```
$ uname -a
Linux seispi 6.1.0-rpi4-rpi-v7 raspberrypi/linux#1 SMP Raspbian 1:6.1.54-1+rpt2 (2023-10-05) armv7l GNU/Linux
```

### Additional context

Thanks.

popcornmix, 2024-05-22:
Moved to bookworm-feedback repo as this issue has nothing to do with linux kernel.

jcapona, 2024-05-30:
hey @HamishB  - just ran into the same issue.

The binary fails to execute because it needs `/lib/arm-linux-gnueabihf/ld-linux-armhf.so.3` (according to `ldd`) but the file doesn't exist.
```
$ file /lib/arm-linux-gnueabihf/ld-linux.so.3
/lib/arm-linux-gnueabihf/ld-linux.so.3: cannot open `/lib/arm-linux-gnueabihf/ld-linux.so.3' (No such file or directory)
```

however, if you check the output of `ldd` you'll see that this file should be a link to `/lib/ld-linux-armhf.so.3` which in my case exists:
```
$ file /lib/ld-linux-armhf.so.3
/lib/ld-linux-armhf.so.3: symbolic link to arm-linux-gnueabihf/ld-linux-armhf.so.3
```

so your issue will be fixed if you create the missing file as a link to this library:
```
$ sudo ln -s /lib/ld-linux-armhf.so.3 /lib/arm-linux-gnueabihf/ld-linux.so.3
$ DISPLAY=:0 wmctrl -l
0x01e00003 -1 pi pcmanfm
0x01c00006 -1 pi panel
0x02a00003  0 pi pi@raspberrypi: ~
```

Not sure if this is an issue for the RPi OS devs or one for the maintainers of `wmctrl` since it's compiled using a 'non standard' location of the libraries (or maybe an older version of libc where this file was provided?). 

The missing file was provided by `libc` until `bullseye` but it disappeared on the `bookworm` release of the package; check:
- https://packages.debian.org/bookworm/armhf/libc6/filelist
- https://packages.debian.org/bullseye/armhf/libc6/filelist

HamishB, 2024-05-31:
Yeah, adding the missing symlink got it working, thanks!

```
# cd /usr/lib/arm-linux-gnueabihf
# ln -s ld-linux-armhf.so.3 ld-linux.so.3
```

lurch, 2024-06-06:
@XECDesign is currently on holiday, but when he gets back: Is this an "us problem" or an "upstream Raspbian" problem? Or is it a bug in upstream wmctrl?

XECDesign, 2024-06-12:
It's a Raspbian bug, but the fix is just to rebuild the package, so I've fixed it on our end instead.

closed 2024-07-01 by lurch