raspberrypi/bookworm-feedback #257: gstreamer-1.22.0 ximagesink is broken on R. Pi: needs a tiny backport from 1.22.10

fduncanh, 2024-05-25:
When touchscreen support was added to ximagesink (only needed in cases where the monitor doesn't support xvimagesink, typically seen on small 400x860 screens often used with Pi .... )   an initialization was not done, which always causes a segfault on the pi with bookworm's gstreamer1.0-plugins-base.

* the patch needed is just to add   " = {0}"  on  one line, and is already merged into gstreamer.
see

https://gitlab.freedesktop.org/gstreamer/gstreamer/-/merge_requests/6631/diffs

XECDesign, 2024-05-27:
We don't currently patch gstreamer in bookworm, so it's a stock Debian package. If the bug is filed and fixed there, it will flow down into Raspberry Pi OS.

@6by9, or does this look like a high priority issue that we should fix on our end?

6by9, 2024-06-01:
It's not a terribly high priority fix in my book.

You've flagged if for a backport in your comment at https://gitlab.freedesktop.org/gstreamer/gstreamer/-/issues/3470#note_2371743. Is that going to happen? If so then it should appear eventually as Debian pick up the 1.22 update. Otherwise it's possibly worth flagging it on the Debian bug tracker as a potential patch they might wish to carry.

fduncanh, 2024-06-01:
It was backported from development to 1.24.3 and 1.22.11    

I will try and get Debian bugs to backport it, but who knows?   Its a very small fix, and a patch is available on our site, with instructions for users.   https://github.com/FDH2/UxPlay/wiki/patches-for-(1)-broken--videoflip-method-and-(2)-broken-ximagesink---in-GStreamer-1.22.0#2-ximagesink-support

* probably not "high priority" but also only requires a small effort.

It was unnoticed since october 2022 because it doesn't break  x86_64.  It does break arm because random uninitialized
junk  usually triggers an X11parsing error (at least on R Pi) when ximagesink states.   No idea why x86 is immune.

* why would ximagesink as opposed to xvimagesink get used?      

* the bug emerged when someone was trying to develop a device with a Pi CM4 board and a  small 480x800
monitor which didn't support XV  X11 extensions.   

I tracked the but down to when it first appeared

Probably a rare use use.

https://github.com/FDH2/UxPlay/issues/277

https://github.com/FDH2/UxPlay/issues/280

lurch, 2024-07-02:
> It's not a terribly high priority fix in my book.

> Probably a rare use case.

In that case I'll close this issue.

closed 2024-07-02 by lurch

XECDesign, 2024-07-10:
> I humbly suggest reconsidering backporting this.

Is [this](https://gitlab.freedesktop.org/gstreamer/gstreamer/-/merge_requests/6631/diffs) the patch you're talking about?

buhman, 2024-07-10:
Yes, I think https://gitlab.freedesktop.org/gstreamer/gstreamer/-/merge_requests/6631/diffs is both trivial and highly valuable for me.