raspberrypi/bookworm-feedback #259: Strange kernel bug issue.

qrp73, 2024-05-30:
Just catch some strange error when tried to start firefox.  Memory was almost free, there was just vlc process playing music. I started firefox but it's window didn't appears. I tried to start it again, still not appears, system monitor showed that firefox process is running with 0% cpu usage, but it's window don't appears. Then I tried to reboot and it leads to some strange kernel errors. 

Here is the log:
```
May 30 05:24:50 raspi kernel: vc4-drm gpu: swiotlb buffer is full (sz: 524288 bytes), total 32768 (slots), used 212 (slots)
May 30 05:24:50 raspi kernel: vc4-drm gpu: swiotlb buffer is full (sz: 524288 bytes), total 32768 (slots), used 84 (slots)
May 30 05:24:50 raspi kernel: vc4-drm gpu: swiotlb buffer is full (sz: 524288 bytes), total 32768 (slots), used 212 (slots)
May 30 05:24:50 raspi kernel: vc4-drm gpu: swiotlb buffer is full (sz: 524288 bytes), total 32768 (slots), used 84 (slots)
May 30 05:25:35 raspi kernel: ------------[ cut here ]------------
May 30 05:25:35 raspi kernel: WARNING: CPU: 1 PID: 15102 at drivers/media/common/videobuf2/videobuf2-core.c:2024 __vb2_queue_cancel+0x220/0x2a0 [videobuf2_co>
May 30 05:25:35 raspi kernel: Modules linked in: rfcomm snd_seq_dummy snd_hrtimer snd_seq cmac algif_hash aes_arm64 aes_generic algif_skcipher af_alg bnep sn>
May 30 05:25:35 raspi kernel: CPU: 1 PID: 15102 Comm: ThreadPoolSingl Tainted: G         C         6.6.31-v8+ #1
May 30 05:25:35 raspi kernel: Hardware name: Raspberry Pi 4 Model B Rev 1.5 (DT)
May 30 05:25:35 raspi kernel: pstate: 80000005 (Nzcv daif -PAN -UAO -TCO -DIT -SSBS BTYPE=--)
May 30 05:25:35 raspi kernel: pc : __vb2_queue_cancel+0x220/0x2a0 [videobuf2_common]
May 30 05:25:35 raspi kernel: lr : __vb2_queue_cancel+0x38/0x2a0 [videobuf2_common]
May 30 05:25:35 raspi kernel: sp : ffffffc081b4bae0
May 30 05:25:35 raspi kernel: x29: ffffffc081b4bae0 x28: ffffffd1b6f0de18 x27: ffffff80b84d4000
May 30 05:25:35 raspi kernel: x26: ffffff8047355d30 x25: 0000000000000000 x24: 0000000040045613
May 30 05:25:35 raspi kernel: x23: 0000000000000000 x22: ffffff80e0700298 x21: ffffff80e0700298
May 30 05:25:35 raspi kernel: x20: 000000000000000a x19: ffffff80e0700298 x18: ffffffffffffffff
May 30 05:25:35 raspi kernel: x17: 0000000000000002 x16: ffffffd2038d4350 x15: ffffffc101b4b717
May 30 05:25:35 raspi kernel: x14: 0000000000000023 x13: ffffff8046f40028 x12: 0000000000000000
May 30 05:25:35 raspi kernel: x11: ffffff8091c16a50 x10: ffffff8091c16910 x9 : ffffffd203929e14
May 30 05:25:35 raspi kernel: x8 : ffffff8091c16938 x7 : 0000000000000001 x6 : 07fc000000000000
May 30 05:25:35 raspi kernel: x5 : 0000000041001000 x4 : 0000000080150014 x3 : ffffffaef6efd000
May 30 05:25:35 raspi kernel: x2 : 0000000100000000 x1 : 0000000000000000 x0 : 0000000000000007
May 30 05:25:35 raspi kernel: Call trace:
May 30 05:25:35 raspi kernel:  __vb2_queue_cancel+0x220/0x2a0 [videobuf2_common]
May 30 05:25:35 raspi kernel:  vb2_core_streamoff+0x2c/0xe0 [videobuf2_common]
May 30 05:25:35 raspi kernel:  vb2_streamoff+0x24/0x80 [videobuf2_v4l2]
May 30 05:25:35 raspi kernel:  v4l2_m2m_streamoff+0x50/0x160 [v4l2_mem2mem]
May 30 05:25:35 raspi kernel:  v4l2_m2m_ioctl_streamoff+0x20/0x38 [v4l2_mem2mem]
May 30 05:25:35 raspi kernel:  v4l_streamoff+0x2c/0x40 [videodev]
May 30 05:25:35 raspi kernel:  __video_do_ioctl+0x194/0x400 [videodev]
May 30 05:25:35 raspi kernel:  video_usercopy+0x1e4/0x780 [videodev]
May 30 05:25:35 raspi kernel:  video_ioctl2+0x20/0x40 [videodev]
May 30 05:25:35 raspi kernel:  v4l2_ioctl+0x48/0x70 [videodev]
May 30 05:25:35 raspi kernel:  __arm64_sys_ioctl+0xb0/0x100
May 30 05:25:35 raspi kernel:  invoke_syscall+0x50/0x128
May 30 05:25:35 raspi kernel:  el0_svc_common.constprop.0+0x48/0xf8
May 30 05:25:35 raspi kernel:  do_el0_svc+0x28/0x40
May 30 05:25:35 raspi kernel:  el0_svc+0x40/0xf8
May 30 05:25:35 raspi kernel:  el0t_64_sync_handler+0x13c/0x158
May 30 05:25:35 raspi kernel:  el0t_64_sync+0x190/0x198
May 30 05:25:35 raspi kernel: ---[ end trace 0000000000000000 ]---
May 30 05:25:35 raspi kernel: videobuf2_common: driver bug: stop_streaming operation is leaving buf 000000007803eae9 in active state
May 30 05:25:35 raspi kernel: videobuf2_common: driver bug: stop_streaming operation is leaving buf 00000000c2e505c9 in active state
May 30 05:25:35 raspi kernel: videobuf2_common: driver bug: stop_streaming operation is leaving buf 000000003c501769 in active state
May 30 05:25:35 raspi kernel: videobuf2_common: driver bug: stop_streaming operation is leaving buf 000000008863bc2f in active state
May 30 05:25:35 raspi kernel: videobuf2_common: driver bug: stop_streaming operation is leaving buf 000000005d2ecabb in active state
May 30 05:25:35 raspi kernel: videobuf2_common: driver bug: stop_streaming operation is leaving buf 00000000e1df21d7 in active state
May 30 05:25:35 raspi kernel: videobuf2_common: driver bug: stop_streaming operation is leaving buf 000000000825d10f in active state
May 30 05:25:35 raspi kernel: ------------[ cut here ]------------
May 30 05:25:35 raspi kernel: WARNING: CPU: 1 PID: 12952 at drivers/media/common/videobuf2/videobuf2-core.c:1049 vb2_buffer_done+0x29c/0x2c0 [videobuf2_commo>
May 30 05:25:35 raspi kernel: Modules linked in: rfcomm snd_seq_dummy snd_hrtimer snd_seq cmac algif_hash aes_arm64 aes_generic algif_skcipher af_alg bnep sn>
May 30 05:25:35 raspi kernel: CPU: 1 PID: 12952 Comm: kworker/1:0 Tainted: G        WC         6.6.31-v8+ #1
May 30 05:25:35 raspi kernel: Hardware name: Raspberry Pi 4 Model B Rev 1.5 (DT)
May 30 05:25:35 raspi kernel: Workqueue: events buffer_work_cb [bcm2835_mmal_vchiq]
May 30 05:25:35 raspi kernel: pstate: 80000005 (Nzcv daif -PAN -UAO -TCO -DIT -SSBS BTYPE=--)
May 30 05:25:35 raspi kernel: pc : vb2_buffer_done+0x29c/0x2c0 [videobuf2_common]
May 30 05:25:35 raspi kernel: lr : ip_buffer_cb+0xb8/0x208 [bcm2835_codec]
May 30 05:25:35 raspi kernel: sp : ffffffc085babd10
May 30 05:25:35 raspi kernel: x29: ffffffc085babd10 x28: ffffffd204c269c0 x27: ffffffd204c26000
May 30 05:25:35 raspi kernel: x26: ffffff8047234940 x25: ffffff8047234900 x24: ffffff80b84d4000
May 30 05:25:35 raspi kernel: x23: ffffff8046194800 x22: ffffff80e0700298 x21: ffffffd1b6f4a780
May 30 05:25:35 raspi kernel: x20: ffffff8046f40e28 x19: ffffff8046194800 x18: ffffffffffffffff
May 30 05:25:35 raspi kernel: x17: 0000000100000001 x16: ffffffd2036f5a18 x15: ffffffc101b4b777
May 30 05:25:35 raspi kernel: x14: 0000000000000003 x13: ffffffc081b4b77d x12: 0000000000000000
May 30 05:25:35 raspi kernel: x11: 00000000000000cf x10: 0000000000001a30 x9 : ffffffd1b7112b48
[... 352 lines cut ...]
May 30 05:27:25 raspi kernel:  do_el0_svc+0x28/0x40
May 30 05:27:25 raspi kernel:  el0_svc+0x40/0xf8
May 30 05:27:25 raspi kernel:  el0t_64_sync_handler+0x13c/0x158
May 30 05:27:25 raspi kernel:  el0t_64_sync+0x190/0x198
May 30 05:27:25 raspi kernel: Code: 79402c00 79400461 cb000040 79400863 (b9400a62) 
May 30 05:27:25 raspi kernel: ---[ end trace 0000000000000000 ]---
```

Device Raspberry Pi 4 v1.5

```
$ cat /etc/rpi-issue
Raspberry Pi reference 2023-09-22
Generated using pi-gen, https://github.com/RPi-Distro/pi-gen, 40f37458ae7cadea1aec913ae10b5e7008ebce0a, stage4

$ vcgencmd version
May  3 2024 19:32:18 
Copyright (c) 2012 Broadcom
version fef7849c350cb250be836afc6d43c742f9ebc97a (clean) (release) (start)

$ uname -a
Linux raspi 6.6.31-v8+ #1 SMP PREEMPT Fri May 24 16:51:56 UTC 2024 aarch64 GNU/Linux
```

CPU temperature at the moment of issue was 40°C, voltage also was ok, no throttling and no undervoltage. Power supply is original Raspberry Pi 4.

lurch, 2024-07-02:
@6by9 Should this be moved to the Linux repo?

6by9, 2024-07-02:
Probably not as it's not clear what is going on.

` WARNING: CPU: 1 PID: 15102 at drivers/media/common/videobuf2/videobuf2-core.c:2024 __vb2_queue_cancel+0x220/0x2a0 [videobuf2_co>' is a known one of some buffers going missing.
https://github.com/raspberrypi/linux/issues/3791
https://github.com/raspberrypi/linux/issues/5857 is almost the same log raised by qrp73
It's odd that we then get the buffer callbacks after everything should have been shutdown.

Why the system then triggers `Unable to handle kernel NULL pointer dereference at virtual address 000000000000000c` OOPS is likely to be something independent.

qrp73, 2024-07-02:
I catch this issue when there was present some issue with kernel freeze during hw accelerated video decoding. Since issue with kernel freeze is fixed with some update, it is possible that this issue also not relevant anymore. At least I didn't catch it again for a long time.

ghollingworth, 2024-07-09:
In which case reopen if you see it again

closed 2024-07-09 by ghollingworth