raspberrypi/bookworm-feedback #261: adb

spookyfirehorse, 2024-06-08:
Hi and Hello

pi bookworm on pi zero W  arm1176jzf-s

adb over apt installled 

in deutsch üngültiger maschienen befehl 

may in english unknown machincode

i also tried dpkg-buildpackage but ther is also some error

thanks for your great work 

and sorry for my english

spooky

XECDesign, 2024-06-10:
At first I thought it was because our version of android-platform-tools was a problem. I've pulled it from the armhf repo, but the issue is still there with Raspbian's build, but this time in android-libboringssl.

I believe the issue should be reported [here](https://bugs.launchpad.net/raspbian/+bugs) instead.

closed 2024-06-10 by XECDesign

0awe, 2024-12-06:
In case anyone else stumbles across this issue, it has also been logged [here](https://bugs.launchpad.net/raspbian/+bug/2091145).