raspberrypi/bookworm-feedback #266: [rpi zero 2w] Disconnect/connect USB devices on the fly leads to unrecoverable error in USB driver

qrp73, 2024-06-22:
Steps to reproduce:
1) Connect USB hub with keyboard, mouse and USB sound card
2) Play some video with music ```SDL_VIDEODRIVER=kmsdrm ffplay music.mp4```
3) Connect and disconnect USB hub and may be try to connect different type of USB sound card
4) Repeat 3 in a loop

Expected result: all works ok

Actual result: after some connect/disconnect cycles USB driver going to some unrecoverable error and cannot detect USB devices anymore. 

Here is the log:
```
[  534.605670] input: Synaptics CX31993 384Khz HIFI AUDIO Consumer Control as /devices/platform/soc/3f980000.usb/usb1/1-1/1-1.4/1-1.4:1.3/0003:3302:3352.0008/input/input11
[  534.666587] hid-generic 0003:3302:3352.0008: input,hidraw3: USB HID v1.11 Device [Synaptics CX31993 384Khz HIFI AUDIO] on usb-3f980000.usb-1.4/input3
[  579.431911] usb 1-1.1: USB disconnect, device number 13
[  584.597152] usb 1-1.2: USB disconnect, device number 14
[  601.453062] usb 1-1.4: USB disconnect, device number 16
[  601.453137] usb 1-1.4: Unable to submit urb #0: -19 at snd_usb_queue_pending_output_urbs
[  612.179921] usb 1-1.1: new full-speed USB device number 17 using dwc_otg
[  612.288494] usb 1-1.1: New USB device found, idVendor=1b3f, idProduct=2008, bcdDevice= 1.00
[  612.288530] usb 1-1.1: New USB device strings: Mfr=1, Product=2, SerialNumber=0
[  612.288540] usb 1-1.1: Product: USB Audio Device
[  612.288548] usb 1-1.1: Manufacturer: GeneralPlus
[  612.331058] input: GeneralPlus USB Audio Device as /devices/platform/soc/3f980000.usb/usb1/1-1/1-1.1/1-1.1:1.3/0003:1B3F:2008.0009/input/input12
[  612.390824] hid-generic 0003:1B3F:2008.0009: input,hidraw0: USB HID v2.01 Device [GeneralPlus USB Audio Device] on usb-3f980000.usb-1.1/input3
[  648.284319] usb 1-1.1: USB disconnect, device number 17
[  682.578684] usb 1-1.1: new high-speed USB device number 18 using dwc_otg
[  682.699853] usb 1-1.1: New USB device found, idVendor=3302, idProduct=3352, bcdDevice= 0.01
[  682.699910] usb 1-1.1: New USB device strings: Mfr=1, Product=2, SerialNumber=0
[  682.699922] usb 1-1.1: Product: CX31993 384Khz HIFI AUDIO
[  682.699930] usb 1-1.1: Manufacturer: Synaptics
[  682.728118] input: Synaptics CX31993 384Khz HIFI AUDIO Consumer Control as /devices/platform/soc/3f980000.usb/usb1/1-1/1-1.1/1-1.1:1.3/0003:3302:3352.000A/input/input13
[  682.793176] hid-generic 0003:3302:3352.000A: input,hidraw0: USB HID v1.11 Device [Synaptics CX31993 384Khz HIFI AUDIO] on usb-3f980000.usb-1.1/input3
[  683.889873] usb 1-1.1: USB disconnect, device number 18
[  684.890676] usb 1-1.1: new high-speed USB device number 19 using dwc_otg
[  685.011363] usb 1-1.1: New USB device found, idVendor=3302, idProduct=3352, bcdDevice= 0.01
[  685.011394] usb 1-1.1: New USB device strings: Mfr=1, Product=2, SerialNumber=0
[  685.011403] usb 1-1.1: Product: CX31993 384Khz HIFI AUDIO
[  685.011410] usb 1-1.1: Manufacturer: Synaptics
[  685.039294] input: Synaptics CX31993 384Khz HIFI AUDIO Consumer Control as /devices/platform/soc/3f980000.usb/usb1/1-1/1-1.1/1-1.1:1.3/0003:3302:3352.000B/input/input14
[  685.103221] hid-generic 0003:3302:3352.000B: input,hidraw0: USB HID v1.11 Device [Synaptics CX31993 384Khz HIFI AUDIO] on usb-3f980000.usb-1.1/input3
[  807.765524] usb 1-1.2: new low-speed USB device number 20 using dwc_otg
[  807.877808] usb 1-1.2: New USB device found, idVendor=046d, idProduct=c051, bcdDevice=30.00
[  807.877848] usb 1-1.2: New USB device strings: Mfr=1, Product=2, SerialNumber=0
[  807.877861] usb 1-1.2: Product: USB-PS/2 Optical Mouse
[  807.877871] usb 1-1.2: Manufacturer: Logitech
[  807.897891] input: Logitech USB-PS/2 Optical Mouse as /devices/platform/soc/3f980000.usb/usb1/1-1/1-1.2/1-1.2:1.0/0003:046D:C051.000C/input/input15
[  807.899085] hid-generic 0003:046D:C051.000C: input,hidraw3: USB HID v1.10 Mouse [Logitech USB-PS/2 Optical Mouse] on usb-3f980000.usb-1.2/input0
[  808.820989] usb 1-1.1: USB disconnect, device number 19

[  814.057413] ERROR::dwc_otg_hcd_urb_enqueue:501: Not connected

[  814.085543] usb 1-1: USB disconnect, device number 12
[  814.085573] usb 1-1.2: USB disconnect, device number 20
[  814.146858] usb 1-1.3: USB disconnect, device number 15
[  819.689456] Indeed it is in host mode hprt0 = 00021501
[  819.877474] usb 1-1: new high-speed USB device number 21 using dwc_otg
[  819.877783] Indeed it is in host mode hprt0 = 00001101
[  820.105964] usb 1-1: New USB device found, idVendor=3302, idProduct=3352, bcdDevice= 0.01
[  820.106001] usb 1-1: New USB device strings: Mfr=1, Product=2, SerialNumber=0
[  820.106011] usb 1-1: Product: CX31993 384Khz HIFI AUDIO
[  820.106019] usb 1-1: Manufacturer: Synaptics
[  820.133496] input: Synaptics CX31993 384Khz HIFI AUDIO Consumer Control as /devices/platform/soc/3f980000.usb/usb1/1-1/1-1:1.3/0003:3302:3352.000D/input/input16
[  820.198539] hid-generic 0003:3302:3352.000D: input,hidraw0: USB HID v1.11 Device [Synaptics CX31993 384Khz HIFI AUDIO] on usb-3f980000.usb-1/input3
[  849.799453] usb usb1-port1: disabled by hub (EMI?), re-enabling...
[  849.799505] usb 1-1: USB disconnect, device number 21
[  850.009429] Indeed it is in host mode hprt0 = 00001101
[  850.209200] usb 1-1: new high-speed USB device number 22 using dwc_otg
[  850.209540] NYET/NAK/ACK/other in non-error case, 0x00000002
[  850.209568] NYET/NAK/ACK/other in non-error case, 0x00000002
[  850.209598] NYET/NAK/ACK/other in non-error case, 0x00000002
[  850.209632] Indeed it is in host mode hprt0 = 00001101
[  850.397198] usb 1-1: device descriptor read/64, error -71
[  850.505374] NYET/NAK/ACK/other in non-error case, 0x00000002
[  850.505503] NYET/NAK/ACK/other in non-error case, 0x00000002
[  850.505546] NYET/NAK/ACK/other in non-error case, 0x00000002
[  850.505582] NYET/NAK/ACK/other in non-error case, 0x00000002
[  850.505648] NYET/NAK/ACK/other in non-error case, 0x00000002
[  850.505679] NYET/NAK/ACK/other in non-error case, 0x00000002
[  850.505716] NYET/NAK/ACK/other in non-error case, 0x00000002
[  850.505761] Indeed it is in host mode hprt0 = 00001101
[  850.693197] usb 1-1: device descriptor read/64, error -71
[  850.801291] Indeed it is in host mode hprt0 = 00001101
[  850.989191] usb 1-1: new high-speed USB device number 23 using dwc_otg
[  850.989611] Indeed it is in host mode hprt0 = 00001101
[  851.177260] usb 1-1: device descriptor read/64, error -71
[  851.285623] Indeed it is in host mode hprt0 = 00001101
[  851.473215] usb 1-1: device descriptor read/64, error -71
[  851.581279] usb usb1-port1: attempt power cycle
```

lurch, 2024-07-01:
1. Is there a reason why you need to constantly unplug and replug your USB keyboard 30 times? :wink: 
2. This sound like a [Linux kernel](https://github.com/raspberrypi/linux) issue. Could you transfer it please @P33M ?

qrp73, 2024-07-01:
> 1. Is there a reason why you need to constantly unplug and replug your USB keyboard 30 times? 😉

Yes, this is very mandatory feature to be able connect and disconnect USB devices on the fly. This is a typical use-case for USB port.

Note that the issue happens with any kind of USB device. I'm used USB keyboard just as example for demonstration as a typical USB device. But the same issue happens with any other kind of USB devices, like USB mouse, USB sound card, USB flash drives, etc. 

First I notice, that USB port stops to work after some period of usage. It happens when I worked with different USB devices, so I started investigation what is going on and I found that USB driver randomly going into unrecoverable error at USB device connect/disconnect time. And this issue can be easily reproduced with simple connect/disconnect USB device in a loop.
 
>     2. This sound like a [Linux kernel](https://github.com/raspberrypi/linux) issue. Could you transfer it please @P33M ?

I don't know how to transfer it between different repositories. How can I do it?

lurch, 2024-07-01:
Thank you for the additional background details :+1: 

> I don't know how to transfer it between different repositories. How can I do it?

You can't - that's why I asked @P33M (who's one of our kernel developers) to transfer it :slightly_smiling_face: