raspberrypi/bookworm-feedback #267: Not able to mount filesystem at boot via NFSv4

NCLI, 2024-06-22:
There is a bug in Debian which prevents us from using NFSv4 for mounting a remote filesystem at boot time. This means that users have to fall back to the less secure NFSv3 if they want to boot their pi off of a network drive.

Upstream has been aware of [the issue](https://bugs.debian.org/cgi-bin/bugreport.cgi?bug=409272) for almost two decades, but apparently [it used to work in raspios](https://bugs.launchpad.net/ubuntu/+source/linux-raspi/+bug/1954716).

Any chance of getting this working again? I really don't like having to use NFSv3 just for netbooting my Pis.

XECDesign, 2024-06-24:
You may get it working again by disabling initramfs. Unfortunately, it's a bit fiddly and I haven't tested this at all, so please take a backup first.

1. In `/etc/default/raspi-firmware`, set `SKIP_INITRAMFS_GEN=no` to prevent raspi-firmware from genetating initramfs.
2. in `/etc/initramfs-tools/update-initramfs.conf` set `update_initramfs=no` to prevent initramfs-tools from updating initramfs.
3. In `/boot/firmware/config.txt` set `auto_initramfs=0` to prevent the firmware from looking for initramfs files.
4. Run the following to delete existing ones:
```
sudo update-initramfs -d
sudo rm -f /boot/firmware/initramfs_*
```

Not all of these steps are strictly required, but this should remove all attempts to do anything with initramfs on the system.

NCLI, 2024-06-26:
My system has at this point only loaded /bolt via tftp, so step 4 was the only step I could see having a meaningful effect. This just won me a complaint from the kernel that no init was found.

ETA-2026, 2025-09-20:
Actually, the only required step to boot using NFS version 4+ is to:

set `auto_initramfs=1` to `#auto_initramfs=1` in `/boot/firmware/config.txt` and reboot.