raspberrypi/bookworm-feedback #268: Lack of natural scrolling for mice

mgriego, 2024-06-22:
One of the more annoying aspects of the new Wayland-compositor-based desktop UI in Raspberry Pi OS is the lack of natural scrolling for mice. I personally use natural scrolling on every desktop, so not having it on the Pi is jarring. I submitted an upstream patch to add natural scrolling support for mice to the `wayfire` compositor: https://github.com/WayfireWM/wayfire/pull/2370. Since `wayfire` for Raspberry Pi OS is not on the current version, can we backport this into the Raspberry Pi OS `wayfire` release? At the moment, I'm using a locally-patched Debian package I've created. The patch pretty much directly applies to the older `wayfire` version. I would have submitted a patch, but none of the branches on the https://github.com/raspberrypi-ui/wayfire repository look like they directly correspond to the source package for `wayfire` on Raspberry Pi OS...

lurch, 2024-07-01:
I asked a colleague about this, and they said that although we don't yet have this in our version of Wayfire, it _is_ available in the beta Labwc compositor (see https://forums.raspberrypi.com/viewtopic.php?t=370005 ) via the `<libinput><device><naturalScroll> [yes|no]` config option, if you'd like to give that a test?

popcornmix, 2024-07-01:
I think that forum thread may be visible only to beta testers.
There are similar instructions here:
https://forums.raspberrypi.com/viewtopic.php?p=2219975#p2219975

mgriego, 2024-07-08:
So the expectation is that Raspberry Pi OS is going to switch to the labwc compositor as the default going forward?

ghollingworth, 2024-07-09:
Yes