raspberrypi/bookworm-feedback #269: XRDP and VNC inconsistenties

SensorAnalyticsAus, 2024-06-24:
I have RPI Imager installs of Bookworm on a PI5 and a PI3B. Both also have `xrdp` installed, *VNC* enabled, and *W1 X11     Openbox window manager with X11 backend* option selected within `raspi-config`. PI5 can be connected over `xrdp` while *VNC* comes up with *Cannot currently show the desktop* after successfully authenticating to *VNC*. 

OTH PI3B connection over `xrdp` fails with MS Remote Desktop client opening up a black screen which after a few minutes timesout to a grey popup ending with a message *error connecting to user session* with an *OK* button. Clicking on it leads to another popup screen for re-typing in the password, and this also fails after a long wait. Last few entries in `.xorgxrdp-11.log`:
```
[867399.538] (WW) xf86CloseConsole: KDSETMODE failed: Bad file descriptor
[867399.538] (WW) xf86CloseConsole: VT_GETMODE failed: Bad file descriptor
[867399.541] (EE) Server terminated with error (1). Closing log file.
```
paradoxically, *VNC* connection works on PI3B.

#### RPIs OS and HW info
PI5 
`uname -a
Linux rpi5 6.6.31+rpt-rpi-2712 #1 SMP PREEMPT Debian 1:6.6.31-1+rpt1 (2024-05-29) aarch64 GNU/Linux`
```
cat /proc/cpuinfo
...
Revision	: d04170
Serial		: c4c6119e310d22ba
Model		: Raspberry Pi 5 Model B Rev 1.0
```

PI3B 
`uname -a
Linux rpi3bw 6.6.31+rpt-rpi-v8 #1 SMP PREEMPT Debian 1:6.6.31-1+rpt1 (2024-05-29) aarch64 GNU/Linux`
```
cat /proc/cpuinfo
...
Revision	: a22082
Serial		: 00000000a37635af
Model		: Raspberry Pi 3 Model B Rev 1.2
```

I can get to gui desktop on both the PI's (one way or the other) but this inconsistency isn't re-assuring.

lurch, 2024-07-01:
Using the X11 backend is deprecated on Raspberry Pi 5. Are you able to use the Wayfire backend and WayVNC instead?

SensorAnalyticsAus, 2024-07-02:
I have switched to Wayfire backend on PI5, no change i.e. xrdp works and vnc shows 'cannot currently show the desktop' message. How does one use WayVNC? Can't see it in `raspi-config` only option is to enable VNC (which is done).

lurch, 2024-07-02:
https://www.raspberrypi.com/documentation/computers/remote-access.html#vnc

I believe that if you use `raspi-config` to enable VNC, it'll automatically use WayVNC if you're using Wayfire or vncserver-x11 if you're using X11. So if you switched to X11 and then enabled VNC, and then switched back to Wayfire, _perhaps_ it's still trying to use vncserver-x11 with Wayfire? (which obviously won't work).
Does it work if (after you've switched to Wayfire) you use raspi-config to disable VNC (i.e. disable vncserver-x11), reboot, and then use raspi-config to enable VNC again (i.e. enable WayVNC)?

Note that when using WayVNC you _might_ also need to use a different VNC client to what you were using with vncserver-x11 - the documentation link I provided above recommends TigerVNC.

SensorAnalyticsAus, 2024-07-02:
I was able to switch to WayVNC with your docu link and  information, thanks. However tigerVNC client (1.3.1 for MacOS) connects to a blank grey screen. 
[image]

lurch, 2024-07-11:
Works fine for me with TigerVNC Viewer v1.13.1 on Linux. I'll check with one of my Mac-using colleagues when he's back in the office next week. Did you also mean 1.13.1, or are you actually using [1.3.1](https://github.com/TigerVNC/tigervnc/releases/tag/v1.3.1) from June 2014? :wink:

SensorAnalyticsAus, 2024-07-11:
I've indeed installed TigerVNC Viewer v1.13.1, as you guessed precisely, on  intel macos 13.6.7 Thanks, that'll be good to know.

lurch, 2024-07-23:
@SensorAnalyticsAus Can you try testing this again please, using a freshly-written Bookworm image on your Pi 5?

SensorAnalyticsAus, 2024-07-24:
I was thinking of doing the same, have ordered a sdcard.

SensorAnalyticsAus, 2024-07-24:
> @SensorAnalyticsAus Can you try testing this again please, using a freshly-written Bookworm image on your Pi 5?

Done a fresh install of Bookworm 64bit with Desktop (Recommended ver.) on a new sdcard with latest PI Imager for PI5. Both VNC and XRDP connections are working now 👍  Only one minor bug encountered, the *Shutdown* *Reboot* *Logout* dialog box doesn't work within XRDP. It does work within VNC though. Anyway, a good outcome; this glitch's immaterial.  Thanks

andys73, 2024-07-24:
I am using a bookworm Desktop 64 bit fresh install on a Pi3B .

I get the same via XRDP

[  5368.877] (WW) xf86CloseConsole: KDSETMODE failed: Bad file descriptor
[  5368.877] (WW) xf86CloseConsole: VT_GETMODE failed: Bad file descriptor
[  5368.879] (EE) Server terminated with error (1). Closing log file.

Via VNC I get a blank screen using latest tiger VNC 1.14
[image]

andys73, 2024-07-24:
this was solved by downgrating to tinyvnc 1.13.1...

RDP still does not work but I can now remotely connect to my bookworm pi :D

lurch, 2024-07-24:
Okay, closing this issue.

closed 2024-07-24 by lurch

asfasty, 2025-02-01:
ok, after spending days to make xrdp work which definitely is working on the bulleseye edition I do not understand why this is closed. Can you please elaborate why? I cannot repeat all the config files I have tried - but I can definitely say - rdp from windows works with pre-bookworm and I just do not get it - why. My edge tabs are showing around 30+ tabs of what I have tried but I get the impression rdp from Windows is not wanted. If so I am  happy to be told so in order to use another app to access devices which ususally have been accessible by using rdcman

spl237, 2025-02-01:
Bullseye uses X11 as its windowing system. Bookworm uses Wayland, which is a completely different windowing system. RDP accesses the windowing system, so it looks as if while RDP was compatible with X11, it is not compatible with Wayland.

If you wish to remote access a Bookworm Pi, use VNC - there is a version of VNC which is compatible with Wayland, and this is already installed on Bookworm.

asfasty, 2025-02-02:
Thank you very much for this detailed explanation. I could not find this anywhere else explained so straightforward. 👍 :)

ghollingworth, 2025-02-03:
Or just use Pi Connect, since it "Just Works ^tm"