raspberrypi/bookworm-feedback #270: Realtek wifi driver is broken

SensorAnalyticsAus, 2024-06-24:
Support for Realtek `8821cu` chipset which is almost ubiquitous in cheap wifi dongles is broken in PiOS. On PI5 I was able to get such dongles to be recognised but on PI3B (running an older kernel) not even that. Soft AP mode, which is a key reason to have such a dongle, does not work regardless. Fix is installing the driver from `https://github.com/morrownr/8821cu-20210916.git`. If this repo's driver can be made part of the Bookworm, will save having to do separate updates and def give PiOS a big legup in supporting the headless mode for production uses.

popcornmix, 2024-06-24:
The answer for supporting an out-of-tree kernel driver is always no.

There tends to be a lot of maintenance work on these when we rebase on newer kernel versions.

You should make sure any devices have *upstream* kernel support before buying.

SensorAnalyticsAus, 2024-06-24:
Fair enough. What about fixing the broken driver then?

6by9, 2024-06-25:
You haven't provided any significant details in how you believe the driver is broken.

The Realtek wireless driver used in the Pi kernel is unchanged from the mainline Linux kernel tree, so I would expect you to get the same behaviour on any other system (eg x86) running a recent Linux kernel. 
Support for 8821cu appears to have been added in 6.2 via https://github.com/torvalds/linux/commit/aff5ffd718de23cb8603f2e229204670e2644334, so if your Pi3 kernel predates that (you haven't said how much older your kernel is there) then I'm not surprised it doesn't work.

If Realtek haven't upstreamed the Soft AP mode that you want, then that's not down to us, hence popcornmix's comment of buying devices that have upstream Linux kernel support for any features you want.

SensorAnalyticsAus, 2024-06-25:
> You haven't provided any significant details in how you believe the driver is broken.
> 
> The Realtek wireless driver used in the Pi kernel is unchanged from the mainline Linux kernel tree, so I would expect you to get the same behaviour on any other system (eg x86) running a recent Linux kernel. Support for 8821cu appears to have been added in 6.2 via [torvalds/linux@aff5ffd](https://github.com/torvalds/linux/commit/aff5ffd718de23cb8603f2e229204670e2644334), so if your Pi3 kernel predates that (you haven't said how much older your kernel is there) then I'm not surprised it doesn't work.
> 

`uname -a
Linux rpi5 6.6.31+rpt-rpi-2712 #1 SMP PREEMPT Debian 1:6.6.31-1+rpt1 (2024-05-29) aarch64 GNU/Linux`

And all my Realtek wifi devices work perfectly after installing morrownr's 8821cu driver.

> If Realtek haven't upstreamed the Soft AP mode that you want, then that's not down to us, hence popcornmix's comment of buying devices that have upstream Linux kernel support for any features you want.

Ok I can see your limitations, it was feedback; not a features request.

lurch, 2024-07-01:
> The answer for supporting an out-of-tree kernel driver is always no.