raspberrypi/bookworm-feedback #272: Hig cpu usage by lxdpe-pi-shutdown-helper process

berdinal, 2024-06-26:
### Describe the bug

**Expected behaviour:**
The CPU usage should remain at a normal level when the shutdown menu is opened.

**Actual behavour:**
When the shutdown menu is opened, the lxdpe-pi-shutdown-helper process causes the CPU to run at 100%.

### Steps to reproduce the behaviour

Press the key combination Ctrl+Alt+Delete and observe the load on the processor.

### Device (s)

Raspberry Pi 5

### System

Raspberry Pi reference 2023-10-10
Generated using pi-gen, https://github.com/RPi-Distro/pi-gen, 962bf483c8f326405794827cce8c0313fd5880a8, stage4

2024/04/20 11:53:30 
Copyright (c) 2012 Broadcom
version d1744d21 (release) (embedded)

Linux rpi5 6.6.28+rpt-rpi-2712 raspberrypi/linux#1 SMP PREEMPT Debian 1:6.6.28-1+rpt1 (2024-04-22) aarch64 GNU/Linux

### Additional context

he issue occurs consistently every time the shutdown menu is opened.
I had the dialog open for over a week and the CPU stayed at 100% and increased the temperature to over 80C. So it doesn't go away after some time.

popcornmix, 2024-06-26:
Moving to bookworm-feedback as this issue is not related to the linux kernel.

lurch, 2024-07-02:
I've checked with a colleague, and he says that this is because the shutdown-dialog is busy-looping watching for the power-button on the Pi5 to be pressed. (If the power-button is pressed while the shutdown-dialog is open, the Pi5 then powers down)
There are no plans to change this behaviour, because a brief CPU-usage spike when you're about to shutdown or reboot anyway doesn't seem particularly harmful?

Please feel free to add additional comments if this is causing you _actual_ problems.

closed 2024-07-02 by lurch