raspberrypi/bookworm-feedback #274: Raspbian GNU/Linux 11 (bullseye) has incorrect Mexican Timezones

rodllag, 2024-07-03:
Most of the country is now GMT-6 (has been for some time), but when selecting Mexico City the timezone defaults to CDT (GMT-5)...

6by9, 2024-07-03:
This repo is for feedback/issues on Bookworm, not general repository related issues.

I would strongly suspect that the timezone files will be unchanged from mainline Debian.
https://packages.debian.org/bullseye/tzdata implies that there were updates
-  Fri, 24 Mar 2023 "17-mexico-dst-chihuahua.patch: The northern edge of Chihuahua changes to US timekeeping."
- Sat, 29 Oct 2022 "15-mexico-dst.patch: Mexico will no longer observe DST except near the US border. Chihuahua moves to year-round -06 on 2022-10-30."
Both Bullseye and Bookworm appear to be on the same version as the latest changelog entries are "New upstream version 2024a"

Check your version of the package via "apt policy tzdata".

rodllag, 2024-07-03:
You were right, I was running a 2021 package for some reason. 

Had to run `sudo apt upgrade tzdata` and all is well now.

closed 2024-07-03 by rodllag