raspberrypi/bookworm-feedback #275: Huge memory leak

qrp73, 2024-07-11:
I'm using raspberry pi 4 with 4GB RAM and latest bookworm 64 with wayland.

I notice that after some period of runtime (it may take about 6 hours) with using different apps, there is a huge memory leak. When I close all user processes, the gnome-system-monitor shows that there is 1.2 GB RAM in usage, but on process list page I don't see processes with so large memory consumption.

btop shows 1.12 GB memory in use, but also there is no processes which consume a lot of memory.

Here is output of free:
```
$ free
               total        used        free      shared  buff/cache   available
Mem:         3862616     1137532     1615380      680648     1867836     2725084
Swap:              0           0           0
```

For some unknown reason, htop shows memory usage 392M/3.68G:
[image]

When I reboot the system, memory usage returns to about 450 MB. 

Since there is no information about processes which consume so many memory, I'm confused. How to check where is memory leak?

lurch, 2024-07-11:
>  (it may take about 6 hours) with using different apps

Unless you can come up with a repeatable reproduction scenario, unfortunately I don't think there's anything we can do to investigate this.

> It looks that the memory leak is in shared memory, is it correct? How to check which process cause it?
Since there is no information about processes which consume so many memory, I'm confused. How to check where is memory leak?

I guess this is just a general Linux problem, rather than anything Raspberry Pi specific? A quick Google search finds e.g. https://github.com/0voice/kernel_memory_management/blob/main/%E2%9C%8D%20%E6%96%87%E7%AB%A0/5%20useful%20tools%20to%20detect%20memory%20leaks%20with%20examples.md

I'll close this issue for now, but feel free to follow-up if you do find a good way of reproducing this.

closed 2024-07-11 by lurch

qrp73, 2024-07-11:
> Unless you can come up with a repeatable reproduction scenario, unfortunately I don't think there's anything we can do to investigate this.

This is a stable repeatable reproducion scenario, just use different apps and see how shared memory usage smoothly grows up.

popcornmix, 2024-07-11:
I've found [this script](https://www.zyxware.com/articles/4446/show-total-memory-usage-by-each-application-in-your-ubuntu-or-any-gnu-linux-system) useful on my ubuntu laptop.
It's standard linux so should work on a Pi.

qrp73, 2024-07-11:
> I've found [this script](https://www.zyxware.com/articles/4446/show-total-memory-usage-by-each-application-in-your-ubuntu-or-any-gnu-linux-system) useful on my ubuntu laptop. It's standard linux so should work on a Pi.

The problem here is that leaked memory is missing from process list. It is not available to use and there is no info about processes that using it. I just found that it is displayed as shared memory in ```free``` command output.

popcornmix, 2024-07-11:
If you continue to wait, do things fail catestrophically with out-of-memory errors?

qrp73, 2024-07-12:
yes, usually it leads to system freeze due to out of memory with wayfire crashing. But if you're just waiting it don't grows. It grows when you working with different apps. And I'm not sure which app is the root of cause because there is no way to see which app eats that memory. It looks like some kind of kernel memory leak.

popcornmix, 2024-07-12:
I think it's clear this isn't something that effects most users, or the forums would be swamped with reports.
So there is something specific to your use case that provokes it.

If you can provide instructions, starting with a clean RPiOS image, followed by a sequence of steps (perhaps a script that launches and quits a specific app) that results in an out of memory failure we can investigate.

lurch, 2024-07-12:
Have you checked the size of your tmpfs filesystems? https://unix.stackexchange.com/questions/307015/what-is-the-meaning-of-shared-memory-in-the-free-command

qrp73, 2024-07-13:
> Have you checked the size of your tmpfs filesystems?

Thanks, no I didn't checked this part of memory and this seems like a plausible explanation for where memory leaks. Needs more investigation to check it.

ashah7, 2024-11-04:
did you ever manage to get to the bottom of this
I'm seeing similar issues on the Raspberry pi using bookworm