raspberrypi/bookworm-feedback #277: RPI5 RTC wakealarm not waking up the pi

quitmeyer, 2024-07-17:
In the past I have enabled the wakealarm like this

$ sudo sh -c "echo 0 > /sys/class/rtc/rtc0/wakealarm" 
$ sudo sh -c "echo `date '+%s' -d '+ 3 minutes'` > /sys/class/rtc/rtc0/wakealarm"
And then if i do "sudo halt" or sudo shutdown 

it will shut down the pi, and wake it up 3 minutes later.

For some reason this does not work anymore.

I just put a new bookworm image on my pi, and tried this and it failed to wake itself up.
I also edited the config to make the battery charge and do the true low power mode thing. But I had made these changes previously and waking up worked fine.

 Was there an update or change or something?

I have the rechargeable battery connected to it, and it is getting constant 5V power anyway. 
I don't think anything else has changed with this hardware, but maybe i screwed up something?

Anyone else have this problem?

lurch, 2024-07-17:
I just tried following [these instructions](https://www.raspberrypi.com/documentation/computers/raspberry-pi.html#real-time-clock-rtc) myself, and it worked absolutely fine. Is your bootloader configured correctly?

popcornmix, 2024-07-17:
I don't believe this has ever worked.
```
$ sudo sh -c "echo date '+%s' -d '+ 3 minutes' > /sys/class/rtc/rtc0/wakealarm"
```
I don't think you want the 'echo' in that command. The output of:
```
pi@pi5:~ $ echo date '+%s' -d '+ 3 minutes'
date +%s -d + 3 minutes
```
makes no sense. You probably want:
```
pi@pi5:~ $ date '+%s' -d '+ 3 minutes'
1721221322
```

But as the linked instructions show
```
echo +180 | sudo tee /sys/class/rtc/rtc0/wakealarm
```
is a much easier way to get 3 minutes (and you'll no longer need the `sudo sh -c "echo 0 > /sys/class/rtc/rtc0/wakealarm`).

quitmeyer, 2024-07-17:
> I don't believe this has ever worked.

Interesting!
i have a different pi that i just ran a test to see if i had a typo or something and it has an image from back in April
(from my old post here)
https://raspberrypi.stackexchange.com/questions/148532/rpi5-rtc-wakealarm-not-waking-up-the-pi

and the
```
$ sudo sh -c "echo 0 > /sys/class/rtc/rtc0/wakealarm" 
$ sudo sh -c "echo `date '+%s' -d '+ 3 minutes'` > /sys/class/rtc/rtc0/wakealarm"
```
totally worked! popped right back on in 3 mins. 

I ran your example too, and it also worked on the older pi from April, and it also worked fine :)

Now these two are different physical things (though the same hardware setup, same battery on the RTC, same pi's just different images. But maybe on the one that doesn't work 

- physically there is something wrong? Can the RTC die in a pi? (im guessing this is very unlikely)
- i screwed up something in setting up its image.

I'll dig deeper and report back

quitmeyer, 2024-07-17:
> physically there is something wrong? Can the RTC die in a pi? (im guessing this is very unlikely)

So i just took sd card image from the pi that would not wake itself back up, and i put it into the other pi that used to have an older functional image. This other pi totally can wake itself up now with either image. **So does this mean it is a physical problem?** does one pi just have something busted about its hardware?

popcornmix, 2024-07-17:
To be sure, swap just the pi over (so power supply, sdcard, hdmi cable, RTC battery, usb devices etc from the working pi are now attached to non-working pi), rather than just the sdcard.

If the failure follows the Pi, then we can rule everything else out.

The Pi does also contain bootloader, and bootloader config, so you should compare those.
Run
```
vcgencmd bootloader_version
sudo rpi-eeprom-config
```
and check they match.

lurch, 2024-07-17:
> I don't believe this has ever worked.
> 
> ```
> $ sudo sh -c "echo date '+%s' -d '+ 3 minutes' > /sys/class/rtc/rtc0/wakealarm"
> ```

Ahhh, looks like the backticks in the very first comment got lost because @quitmeyer didn't put the whole thing in triple-backticks :slightly_smiling_face: 
I suspect the intention was:
```
$ sudo sh -c "echo `date '+%s' -d '+ 3 minutes'` > /sys/class/rtc/rtc0/wakealarm"
```

quitmeyer, 2024-07-19:
> vcgencmd bootloader_version
> sudo rpi-eeprom-config

When I do that i get this
[image]

which i don't see anything to match

but if i run rpi eeprom update it says there's an update (and the Jan 5 thing matches the vcgencmd thing
[image]

maybe ill update it and see if it starts working?

quitmeyer, 2024-07-19:
> vcgencmd bootloader_version

Oh WAIT my eeprom settings are totally different!

it doesn't have this stuff:
```

POWER_OFF_ON_HALT=1
WAKE_ON_GPIO=0

```
but my other pis with the exact same image DO have those options configured. 

Do these settings not transfer with the IMAGE!?!? **I may have just learned something really important.** Does each pi need to have me go:

`
sudo -E rpi-eeprom-config --edit`

and change those settings? Does the image on the SD card not carry that information?

quitmeyer, 2024-07-19:
It works!!! I didn't know the eeprom config was specific to each pi I guess! Some of the pis around my house must have just worked because i manually did that at some point, and others didn't wake themselves up because I never edited their eeprom manually.

**Thanks for all your help everyone!**
To recap, my solution was that I had to manually edit the EEPROM stuff on each pi.

Also can someone confirm if my thinking is correct:

If I have a Pi programmed with a schedule to wake itself up, and I want to clone this pi's functionality I need to:
1 Clone the SD card image 
2 manually edit the eeprom for each one?

Bonus question, is there a way to make an image that automatically installs the same EEPROM configuration to the pi once it is going? (I have 30+ of these to make, Lol)

closed 2024-07-19 by quitmeyer

quitmeyer, 2024-07-22:
> Bonus question, is there a way to make an image that automatically installs the same EEPROM configuration to the pi once it is going? (I have 30+ of these to make, Lol)

As an Update, I created a script that is part of my scheduling script that runs whenever the pi is started up. It checks to see if the two lines are already correct in the EEPROM, and if not, it sets it for you!

you can see at the top of this file
https://github.com/Digital-Naturalism-Laboratories/Mothbox/blob/main/Software/Scheduler.py

So in theory i think that should work!

JamesH65, 2024-07-22:
The EEPROM is specific to the Pi, its non-volatile storage on the device itself. The image sits on the SD card (usually), so can be transferred from Pi to Pi. 

So yes, you clone the SD card then make the appropriate EEPROM changes.

quitmeyer, 2024-07-22:
FYI here's my code for auto-setting the EEPROM (if you put this script in a cron command that launches on @reboot )

```
### ~~~~~~~~~~ TODO: SET THE EEPROM TO WAKE UP ~~~~~~~~~~~~~~
def check_eeprom_settings():
    """Checks the current EEPROM settings and returns a dictionary of settings."""
    output = subprocess.check_output(["sudo", "rpi-eeprom-config"]).decode("utf-8")
    settings = {}
    for line in output.splitlines():
        match = re.match(r"(\w+)=(\d+)", line)
        if match:
            settings[match.group(1)] = match.group(2)
    return settings

def set_eeprom_settings(settings):
    """Sets the specified EEPROM settings."""
    config_lines = []
    for key, value in settings.items():
        config_lines.append(f"{key}={value}")

    config_content = "\n".join(config_lines)
    with open("/tmp/eeprom_config.txt", "w") as f:
        f.write(config_content)

    subprocess.run(["sudo", "rpi-eeprom-config", "--apply", "/tmp/eeprom_config.txt"])

if(rpiModel==5):
  desired_settings = {"POWER_OFF_ON_HALT": "1", "WAKE_ON_GPIO": "0"}
  current_settings = check_eeprom_settings()

  if all(current_settings.get(key) == value for key, value in desired_settings.items()):
      print("EEPROM settings are already correct.")
  else:
    for key, value in desired_settings.items():
        if key not in current_settings or current_settings[key] != value:
            current_settings[key] = value
    set_eeprom_settings(current_settings)
    print("EEPROM settings updated.")

```

timg236, 2024-07-22:
If you want to automatically apply bootloader configuration settings then I'd suggest using the hooks mechanism in 
rpi-eeprom-update
https://github.com/raspberrypi/rpi-eeprom/blob/master/rpi-eeprom-update#L159