raspberrypi/bookworm-feedback #279: Firmwire issue with wifi-chip on Zero2w with *some* access-points

lianergoist, 2024-07-20:
There is a problem with the firmware in Bookworm for the wifi-chip on Zero2w. It is a known problem for many users, and it has been discussed in the raspberry pi forum: https://forums.raspberrypi.com/viewtopic.php?t=366155

It has been a problem for a while, when people has upgraded to Bookworm. And it still exist in both 32 and 64 bit versions. But, apparently only on some APs... I can connect to the AP on my android phone, but not to the router from my ISP. 

I don't have the skills to narrow down the exact problem, but a user has reported: "The router logs show that RPi is trying to connect, but everything ends with a Handshake error".

The reason why I think it is a firmware problem in Bookworm for the wifi-chip on Zero2ws: 

I create the sd-card in rpi-imager app (32 or 64 bit). It does not matter if I choose to configure wifi in rpi-imager or do it later in raspi-config. If I configure in rpi-manager (set ssid and password) there is simply no network after logging in. If I use raspi-config, it tries to connect, but fails. 

If I move that sd-card to a Zero1w (only the 32-bit OS of course) and boot up, there are no problems. 

If I use a wifi-dongle, there are no problems.

So, there is only a problem when I use the wifi-chip on Zero2w. And again, only on some APs...

I hope someone will look into it. 

I am not looking for a workaround. I simply will report a problem some users have, that should be fixed.

iustin, 2024-07-20:
Can confirm I have the same issue. zero1w works, pi4 works, zero2w doesn't work. Same AP, same wifi settings. Gave up and using a very thirsty pi4 instead.

JamesH65, 2024-07-22:
I believe this is being looked in to @ Pi

learmj, 2024-07-22:
Please can you post the output of ```dmesg | grep brcmfmac``` and ```sudo apt-cache policy firmware-brcm80211``` from your zero2w.
What would also help is if you created ```/etc/modprobe.d/brcmfmac.conf``` with the following contents...
```options brcmfmac debug=0x104404```
...reboot, then attach the dmesg output here encapsulating the failed association with the AP.

Regarding:

> It has been a problem for a while, when people has upgraded to Bookworm. 

You don't say upgraded from what. The problem may _appear_ to have started happening on an upgrade to bookworm _if_ the previous distro had a different version of firmware compared to bookworm. FYI the tip of bullseye currently has the same version of firmware as bookworm for these chips.

lianergoist, 2024-07-22:
> Please can you post the output of `dmesg | grep brcmfmac` 

```
[   12.178076] brcmfmac: F1 signature read @0x18000000=0x1541a9a6
[   12.238092] brcmfmac: brcmf_fw_alloc_request: using brcm/brcmfmac43430-sdio for chip BCM43430/1
[   12.239172] usbcore: registered new interface driver brcmfmac
[   12.545042] brcmfmac: brcmf_c_process_txcap_blob: no txcap_blob available (err=-2)
[   12.545826] brcmfmac: brcmf_c_preinit_dcmds: Firmware: BCM43430/1 wl0: Jun 14 2023 07:27:45 version 7.45.96.s1 (gf031a129) FWID 01-70bd2af7 es7
[   16.271034] brcmfmac: brcmf_cfg80211_set_power_mgmt: power save enabled
```

> and `sudo apt-cache policy firmware-brcm80211` from your zero2w. 

```
firmware-brcm80211:
  Installeret: 1:20230625-2+rpt2
  Kandidat:    1:20230625-2+rpt2
  Versionstabel:
 *** 1:20230625-2+rpt2 500
        500 http://archive.raspberrypi.com/debian bookworm/main arm64 Packages
        500 http://archive.raspberrypi.com/debian bookworm/main armhf Packages
        100 /var/lib/dpkg/status
     20230210-5 500
        500 http://deb.debian.org/debian bookworm/non-free-firmware arm64 Packages
        500 http://deb.debian.org/debian bookworm/non-free-firmware armhf Packages
```

> What would also help is if you created `/etc/modprobe.d/brcmfmac.conf` with the following contents... `options brcmfmac debug=0x104404` ...reboot, then attach the dmesg output here encapsulating the failed association with the AP.
> 

I have attached the output file... 
[dmesg_grep_brcmfmac_debug-0x104404.txt](https://github.com/user-attachments/files/16331763/dmesg_grep_brcmfmac_debug-0x104404.txt)

> Regarding:
>
>
> You don't say upgraded from what. The problem may _appear_ to have started happening on an upgrade to bookworm _if_ the previous distro had a different version of firmware compared to [...]

I have not upgraded myself, but I have seen this thread:

https://forums.raspberrypi.com/viewtopic.php?p=2130615#p2147719

learmj, 2024-07-22:
Thanks for the log. Unfortunately, it does not appear to show an attempt at association/connection to an AP. I would expect to see ```brcmf_cfg80211_connect``` after ```brcmf_notify_escan_complete ESCAN Completed scan: Done``` for an attempt to connect to a network. Maybe dmesg has overflowed and it's just scanning after having given up at connecting, in which case ```journalctl -k``` would show everything since kernel boot. Maybe the upper layers of the networking stack are also interfering.

If you'd like to try a more 'hands on' approach to connect manually, you could try the following steps while retaining the modprobe settings:
```
sudo systemctl stop wpa_supplicant.service
sudo systemctl stop NetworkManager.service
sudo killall wpa_supplicant
sudo rfkill unblock all
```
Create /etc/wpa_supplicant/wpa_supplicant.conf if it doesn't exist, for example:
```
ctrl_interface=DIR=/var/run/wpa_supplicant GROUP=netdev
update_config=1
```
Start the supplicant:
```
sudo wpa_supplicant -d -c /etc/wpa_supplicant/wpa_supplicant.conf -iwlan0
```
Then use wpa_cli to connect:
```
wpa_cli -i wlan0
> scan
> scan_results
> add_network
0
> set_network 0 ssid "theSSID"
> set_network 0 psk "thepassphrase"
> enable_network 0
> status
```
...where the '0' is the network number returned from add_network. It may be greater than 0 for you.

It should have tried to connect at this point, so hopefully you'll be able to see sign of an error in dmesg / journalctl -k.

You can trigger the following using wpa_cli, although this obviously only helps if you can connect initially:
```
> disconnect
> reconnect
```

lianergoist, 2024-07-22:
> If you'd like to try a more 'hands on' approach to connect manually, you could try the following steps while retaining the modprobe settings:

I am happy to help. Just ask 
> 
[...]
> sudo wpa_supplicant -d -c /etc/wpa_supplicant/wpa_supplicant.conf -iwlan0

This fails. Or rather, it don'tfinish, and I had to Ctrl-z and kill it.

This is the output. I kill it at "EAPOL: External notification - portValid=0".
```

wpa_supplicant v2.10
random: getrandom() support available
Successfully initialized wpa_supplicant
Initializing interface 'wlan0' conf '/etc/wpa_supplicant/wpa_supplicant.conf' driver 'default' ctrl_interface 'N/A' bridge 'N/A'
Configuration file '/etc/wpa_supplicant/wpa_supplicant.conf' -> '/etc/wpa_supplicant/wpa_supplicant.conf'
Reading configuration file '/etc/wpa_supplicant/wpa_supplicant.conf'
ctrl_interface='DIR=/var/run/wpa_supplicant GROUP=netdev'
update_config=1
nl80211: Using driver-based roaming
nl80211: Supported cipher 00-0f-ac:1
nl80211: Supported cipher 00-0f-ac:5
nl80211: Supported cipher 00-0f-ac:2
nl80211: Supported cipher 00-0f-ac:4
nl80211: Supported cipher 00-0f-ac:6
nl80211: Using driver-based off-channel TX
nl80211: Supported vendor command: vendor_id=0x1018 subcmd=1
nl80211: Use separate P2P group interface (driver advertised support)
nl80211: Enable multi-channel concurrent (driver advertised support)
nl80211: use P2P_DEVICE support
nl80211: key_mgmt=0xd0f enc=0x10f auth=0x7 flags=0x20000003804fac8 rrm_flags=0x0 probe_resp_offloads=0x0 max_stations=0 max_remain_on_chan=5000 max_scan_ssids=10
nl80211: interface wlan0 in phy phy0
nl80211: Set mode ifindex 2 iftype 2 (STATION)
nl80211: Subscribe to mgmt frames with non-AP handle 0x55a2feef00
nl80211: Register frame type=0xd0 (WLAN_FC_STYPE_ACTION) nl_handle=0x55a2feef00 match=0104 multicast=0
nl80211: Register frame type=0xd0 (WLAN_FC_STYPE_ACTION) nl_handle=0x55a2feef00 match=040a multicast=0
nl80211: Register frame type=0xd0 (WLAN_FC_STYPE_ACTION) nl_handle=0x55a2feef00 match=040b multicast=0
nl80211: Register frame type=0xd0 (WLAN_FC_STYPE_ACTION) nl_handle=0x55a2feef00 match=040c multicast=0
nl80211: Register frame type=0xd0 (WLAN_FC_STYPE_ACTION) nl_handle=0x55a2feef00 match=040d multicast=0
nl80211: Register frame type=0xd0 (WLAN_FC_STYPE_ACTION) nl_handle=0x55a2feef00 match=090a multicast=0
nl80211: Register frame type=0xd0 (WLAN_FC_STYPE_ACTION) nl_handle=0x55a2feef00 match=090b multicast=0
nl80211: Register frame type=0xd0 (WLAN_FC_STYPE_ACTION) nl_handle=0x55a2feef00 match=090c multicast=0
nl80211: Register frame type=0xd0 (WLAN_FC_STYPE_ACTION) nl_handle=0x55a2feef00 match=090d multicast=0
nl80211: Register frame type=0xd0 (WLAN_FC_STYPE_ACTION) nl_handle=0x55a2feef00 match=0409506f9a09 multicast=0
nl80211: Register frame type=0xd0 (WLAN_FC_STYPE_ACTION) nl_handle=0x55a2feef00 match=7f506f9a09 multicast=0
nl80211: Register frame type=0xd0 (WLAN_FC_STYPE_ACTION) nl_handle=0x55a2feef00 match=0409506f9a1a multicast=0
nl80211: Register frame type=0xd0 (WLAN_FC_STYPE_ACTION) nl_handle=0x55a2feef00 match=0800 multicast=0
nl80211: Register frame type=0xd0 (WLAN_FC_STYPE_ACTION) nl_handle=0x55a2feef00 match=0801 multicast=0
nl80211: Register frame type=0xd0 (WLAN_FC_STYPE_ACTION) nl_handle=0x55a2feef00 match=12 multicast=0
nl80211: Register frame type=0xd0 (WLAN_FC_STYPE_ACTION) nl_handle=0x55a2feef00 match=06 multicast=0
nl80211: Register frame type=0xd0 (WLAN_FC_STYPE_ACTION) nl_handle=0x55a2feef00 match=0a07 multicast=0
nl80211: Register frame type=0xd0 (WLAN_FC_STYPE_ACTION) nl_handle=0x55a2feef00 match=0a11 multicast=0
nl80211: Register frame type=0xd0 (WLAN_FC_STYPE_ACTION) nl_handle=0x55a2feef00 match=0a0b multicast=0
nl80211: Register frame type=0xd0 (WLAN_FC_STYPE_ACTION) nl_handle=0x55a2feef00 match=0a1a multicast=0
nl80211: Register frame type=0xd0 (WLAN_FC_STYPE_ACTION) nl_handle=0x55a2feef00 match=1101 multicast=0
nl80211: Register frame type=0xd0 (WLAN_FC_STYPE_ACTION) nl_handle=0x55a2feef00 match=1102 multicast=0
nl80211: Register frame type=0xd0 (WLAN_FC_STYPE_ACTION) nl_handle=0x55a2feef00 match=0505 multicast=0
nl80211: Register frame type=0xd0 (WLAN_FC_STYPE_ACTION) nl_handle=0x55a2feef00 match=0500 multicast=0
nl80211: Register frame type=0xd0 (WLAN_FC_STYPE_ACTION) nl_handle=0x55a2feef00 match=1301 multicast=0
nl80211: Register frame type=0xd0 (WLAN_FC_STYPE_ACTION) nl_handle=0x55a2feef00 match=1305 multicast=0
nl80211: Register frame type=0xd0 (WLAN_FC_STYPE_ACTION) nl_handle=0x55a2feef00 match=7e506f9a1a multicast=0
rfkill: initial event: idx=1 type=1 op=0 soft=0 hard=0
netlink: Operstate: ifindex=2 linkmode=1 (userspace-control), operstate=5 (IF_OPER_DORMANT)
Add interface wlan0 to a new radio phy0
nl80211: Regulatory information - country=DK (DFS-ETSI)
nl80211: 2400-2483 @ 40 MHz 20 mBm
nl80211: 5150-5250 @ 80 MHz 23 mBm (no outdoor)
nl80211: 5250-5350 @ 80 MHz 20 mBm (no outdoor) (DFS)
nl80211: 5470-5725 @ 160 MHz 26 mBm (DFS)
[... 1792 lines cut ...]
[  174.406149] 00000000: 00 01 01 01 04                                   .....
[  174.406158] brcmfmac: brcmf_fweh_handle_if_event action: 1 ifidx: 0 bsscfgidx: 1 flags: 1 role: 4
[  174.406169] brcmfmac: brcmf_fweh_handle_if_event adding wl0 (da:3a:dd:74:90:c9)
[  174.406179] brcmfmac: brcmf_add_if allocate non-netdev interface
[  174.406921] brcmfmac: check_vif_up device is not ready : status (0)
[  174.406945] brcmfmac: check_vif_up device is not ready : status (0)
[  174.421337] brcmfmac: CONSOLE: 000161.633 wl0: wl_open
[... cut ...]
[  205.758161] brcmfmac: brcmf_fweh_event_worker   version 2 flags 0 status 0 reason 0
[  205.758178] brcmutil: event payload, len=5
[  205.758193] 00000000: 00 02 01 01 04                                   .....
[  205.758204] brcmfmac: brcmf_fweh_handle_if_event action: 2 ifidx: 0 bsscfgidx: 1 flags: 1 role: 4
[  205.758246] brcmfmac: brcmf_p2p_ifp_removed P2P: device interface removed
[  205.758272] brcmfmac: check_vif_up device is not ready : status (0)
[  205.834957] brcmfmac: brcmf_btcoex_set_mode DHCP session ends
```

learmj, 2024-07-22:
> Or rather, it don'tfinish,

You'll need to use a different shell to run wpa_cli in unless wpa_supplicant is invoked with ```-B```. If you pass ```-B``` to wpa_supplicant it will fork to background allowing you to use the same shell in which to run wpa_cli.

lianergoist, 2024-07-23:
> > Or rather, it don'tfinish,
> 
> You'll need to use a different shell to run wpa_cli in unless wpa_supplicant is invoked with `-B`. If you pass `-B` to wpa_supplicant it will fork to background allowing you to use the same shell in which to run wpa_cli.

Okay, I have been running the wpa_cli commands.
```
wpa_cli -i wlan0
> scan
> scan_results
> add_network
0
> set_network 0 ssid "theSSID"
> set_network 0 psk "thepassphrase"
> enable_network 0
> status
```
After that this is dmesg:
[dmesg.txt](https://github.com/user-attachments/files/16346592/dmesg.txt)

and this is  journalctl - k -b -0 (only latest boot):
[journalctl-k-b-0.txt](https://github.com/user-attachments/files/16346591/journalctl-k-b-0.txt)

Hope you can use it, and please let me know if you want me do try other things.

learmj, 2024-07-23:
Thanks. That's what I needed. If you append ```feature_disable=0x2000``` to the line in your ```/etc/modprobe.d/brcmfmac.conf``` does it connect successfully?

lianergoist, 2024-07-23:
> Thanks. That's what I needed. If you append `feature_disable=0x2000` to the line in your `/etc/modprobe.d/brcmfmac.conf` does it connect successfully?

No, I am sorry.

[journalctl-k-b-0.txt](https://github.com/user-attachments/files/16348188/journalctl-k-b-0.txt)
[dmesg.txt](https://github.com/user-attachments/files/16348190/dmesg.txt)

learmj, 2024-07-23:
I think my typo confused things. Your recent journalctl log still shows the feature (PSK offload) as enabled. I'm guessing the syntax in your modprobe conf file is incorrect. Try this:
```
options brcmfmac debug=0x104404 
options brcmfmac feature_disable=0x2000
```
You can confirm the feature is disabled if you **don't** see the following in the log when attempting to connect:
```
brcmfmac: brcmf_cfg80211_connect using PSK offload
```
I suspect disabling this feature will allow you to connect to your AP, but let's see.

lianergoist, 2024-07-23:
> I suspect disabling this feature will allow you to connect to your AP, but let's see.

YES!! Now it's working! :-)

clds84, 2024-07-26:
> I think my typo confused things. Your recent journalctl log still shows the feature (PSK offload) as enabled. I'm guessing the syntax in your modprobe conf file is incorrect. Try this:
>
> ```
> options [...]

Hello,

Any chance you could help me address my issue as well? I've been in the same boat trying to use my hotspot on my iphone as a connection I've setup with networkmanager and while my older setups work just fine, I'm having issues with Bookworm in trying to connect to it. Unfortunately, the two lines in brcmfmac.conf aren't solving it.

I put the following code in /etc/modprobe.d/brcmfmac.conf

```
options brcmfmac debug=0x104404 
options brcmfmac feature_disable=0x2000
 ```
 
If I search for the `brcmfmac: brcmf_cfg80211_connect using PSK offload`, I only find:
```
[   13.969052] brcmfmac: brcmf_cfg80211_connect ie (00000000ff702a51), ie_len (39)
```

learmj, 2024-07-26:
Hi. Can you attach your entire kernel boot log encapsulating the failed association with (1) only the ```options brcmfmac debug=0x104404``` line in the modprobe conf and (2) if you can get it, the log output from wpa_supplicant with some debug enabled (you can either kill and restart it as per instructions above adding ```-d``` to the args, or restart it via systemd, after adding ```-d``` to the ```ExecStart``` line in ```/usr/lib/systemd/system/wpa_supplicant.service```).
Which older setups with your same zero2w were able to connect to your hotspot?

clds84, 2024-07-26:
> options brcmfmac debug=0x104404

Do you normally debug this with a wired network connection?

This disconnects me from ssh session as I would expected
```
sudo systemctl stop wpa_supplicant.service
sudo systemctl stop NetworkManager.service
sudo killall wpa_supplicant
sudo rfkill unblock all
```

I'm running headless so I connected to a display to continue, and added 
```
sudo wpa_supplicant -B -d -c  /etc/wpa_supplicant/wpa_supplicant.conf -iwlan0
```
I included the -B so it would run in the background, but it only works with -B before -d and -c

I see my usual network to connect to when I run

```
wpa_cli -i wlan0
> scan
> scan_results
```

however, I don't see my hotspot after toggling it on and running scan, scan_results. There are about 5 results of the 12 or so that don't have ssid so I don't know if that would be one of them. Would you like me to connect to my usually or are we supposed to be attempting the following add network command with the hotspot?

Forgive the noob question, but if successfully connected to something, am I then supposed to ssh again so I can run dmesg and journalctl so I can get you the outputs?

learmj, 2024-07-26:
> Do you normally debug this with a wired network connection?

Yes. I use a USB OTG cable attached to the zero2w and have a USB-Ethernet adapter plugged into it.
I don't know why you can't see your hotspot in the scan. Maybe it's on 5GHz only. The zero2w only supports 2.4GHz. Pi5 is Dual Band 802.11ac.

learmj, 2024-07-29:
> Any harm in keeping that brcmfmac.conf file in there? Don't quite understand still what that does.

It just makes the brcmfmac driver output extra debug. If nothing needs debugged, you don't need it. Removing it will prevent lots of debug being written to disk and spamming your kernel log.

qrp73, 2024-08-03:
Do I understand correctly, that adding ```options brcmfmac feature_disable=0x2000``` leads to a higher CPU load for WiFi connection due to performing tasks on the main CPU instead of WiFi adapter?

learmj, 2024-08-05:
Correct. It's considered a workaround.

6ffm70, 2024-08-07:
Not sure if related, but I'm getting a similar error message with RPOS Bookworm (32bit lite) on a Zero 2 W when starting a hotspot running on the Zero:
```
me@zero2w:~ $ sudo nmcli connection up hotspot
Passwords or encryption keys are required to access the wireless network 'hotspot-ssid'.
Warning: password for '802-11-wireless-security.psk' not given in 'passwd-file' and nmcli cannot ask without '--ask' option.
Error: Connection activation failed: Secrets were required, but not provided
Hint: use 'journalctl -xe NM_CONNECTION=<uuid> + NM_DEVICE=wlan0' to get more details.
```
The hotspot worked fine before I created and started a bridge with nmcli. bringing the bridge down and then starting the hotspot gives the same result.

Output of `journalctl -xe NM_CONNECTION=<uuid> + NM_DEVICE=wlan0`

```
Aug 07 13:32:37 zero2w NetworkManager[518]: <info>  [1723030357.6216] device (wlan0): state change: failed -> disconnected (reason 'none', sys-iface-state: 'managed')
Aug 07 13:32:37 zero2w NetworkManager[518]: <info>  [1723030357.6216] device (wlan0): state change: failed -> disconnected (reason 'none', sys-iface-state: 'managed')
Aug 07 13:32:37 zero2w NetworkManager[518]: <info>  [1723030357.6216] device (wlan0): state change: failed -> disconnected (reason 'none', sys-iface-state: 'managed')
Aug 07 13:32:57 zero2w NetworkManager[518]: <info>  [1723030377.1466] device (wlan0): state change: disconnected -> unmanaged (reason 'removed', sys-iface-state: 'removed')
Aug 07 13:37:36 zero2w NetworkManager[518]: <info>  [1723030656.5305] device (wlan0): driver supports Access Point (AP) mode
Aug 07 13:37:36 zero2w NetworkManager[518]: <info>  [1723030656.5350] manager: (wlan0): new 802.11 Wi-Fi device (/org/freedesktop/NetworkManager/Devices/9)
Aug 07 13:37:36 zero2w NetworkManager[518]: <info>  [1723030656.5739] device (wlan0): state change: unmanaged -> unavailable (reason 'managed', sys-iface-state: 'external')
Aug 07 13:37:36 zero2w NetworkManager[518]: <info>  [1723030656.9042] device (wlan0): supplicant interface state: internal-starting -> disconnected
Aug 07 13:37:36 zero2w NetworkManager[518]: <info>  [1723030656.9132] device (wlan0): state change: unavailable -> disconnected (reason 'supplicant-available', sys-iface-state: 'managed')
Aug 07 13:43:01 zero2w NetworkManager[518]: <info>  [1723030981.8323] device (wlan0): state change: disconnected -> unmanaged (reason 'removed', sys-iface-state: 'removed')
Aug 07 13:43:17 zero2w NetworkManager[518]: <info>  [1723030997.4637] device (wlan0): driver supports Access Point (AP) mode
Aug 07 13:43:17 zero2w NetworkManager[518]: <info>  [1723030997.4679] manager: (wlan0): new 802.11 Wi-Fi device (/org/freedesktop/NetworkManager/Devices/11)
Aug 07 13:43:17 zero2w NetworkManager[518]: <info>  [1723030997.5056] device (wlan0): state change: unmanaged -> unavailable (reason 'managed', sys-iface-state: 'external')
Aug 07 13:43:17 zero2w NetworkManager[518]: <info>  [1723030997.7544] device (wlan0): supplicant interface state: internal-starting -> disconnected
Aug 07 13:43:17 zero2w NetworkManager[518]: <info>  [1723030997.7638] device (wlan0): state change: unavailable -> disconnected (reason 'supplicant-available', sys-iface-state: 'managed')
Aug 07 13:43:27 zero2w NetworkManager[518]: <info>  [1723031007.1180] device (wlan0): Activation: starting connection 'hotspot' (<uuid>)
Aug 07 13:43:27 zero2w NetworkManager[518]: <info>  [1723031007.1195] device (wlan0): state change: disconnected -> prepare (reason 'none', sys-iface-state: 'managed')
Aug 07 13:43:27 zero2w NetworkManager[518]: <info>  [1723031007.1415] device (wlan0): state change: prepare -> config (reason 'none', sys-iface-state: 'managed')
Aug 07 13:43:27 zero2w NetworkManager[518]: <info>  [1723031007.1433] device (wlan0): Activation: (wifi) access point 'hotspot' has security, but secrets are required.
Aug 07 13:43:27 zero2w NetworkManager[518]: <info>  [1723031007.1435] device (wlan0): state change: config -> need-auth (reason 'none', sys-iface-state: 'managed')
Aug 07 13:43:27 zero2w NetworkManager[518]: <warn>  [1723031007.1628] device (wlan0): no secrets: No agents were available for this request.
Aug 07 13:43:27 zero2w NetworkManager[518]: <info>  [1723031007.1629] device (wlan0): state change: need-auth -> failed (reason 'no-secrets', sys-iface-state: 'managed')
Aug 07 13:43:27 zero2w NetworkManager[518]: <info>  [1723031007.1659] device (wlan0): released from master device bridge0
Aug 07 13:43:27 zero2w NetworkManager[518]: <warn>  [1723031007.1674] device (wlan0): Activation: failed for connection 'hotspot'
Aug 07 13:43:27 zero2w NetworkManager[518]: <info>  [1723031007.1700] device (wlan0): state change: failed -> disconnected (reason 'none', sys-iface-state: 'managed')
Aug 07 13:48:41 zero2w NetworkManager[518]: <info>  [1723031321.6079] device (wlan0): Activation: starting connection 'hotspot' (<uuid>)
Aug 07 13:48:41 zero2w NetworkManager[518]: <info>  [1723031321.6088] device (wlan0): state change: disconnected -> prepare (reason 'none', sys-iface-state: 'managed')
Aug 07 13:48:41 zero2w NetworkManager[518]: <info>  [1723031321.6388] device (wlan0): state change: prepare -> config (reason 'none', sys-iface-state: 'managed')
Aug 07 13:48:41 zero2w NetworkManager[518]: <info>  [1723031321.6408] device (wlan0): Activation: (wifi) access point 'hotspot' has security, but secrets are required.
[... cut ...]
Aug 07 13:48:41 zero2w NetworkManager[518]: <info>  [1723031321.8542] device (wlan0): state change: need-auth -> failed (reason 'no-secrets', sys-iface-state: 'managed')
Aug 07 13:48:41 zero2w NetworkManager[518]: <info>  [1723031321.8587] device (wlan0): released from master device bridge0
Aug 07 13:48:41 zero2w NetworkManager[518]: <warn>  [1723031321.8603] device (wlan0): Activation: failed for connection 'hotspot'
Aug 07 13:48:41 zero2w NetworkManager[518]: <info>  [1723031321.8627] device (wlan0): state change: failed -> disconnected (reason 'none', sys-iface-state: 'managed')
```

learmj, 2024-08-07:
> when starting a hotspot running on the Zero:
> The hotspot worked fine before I created and started a bridge with nmcli. bringing the bridge down and then starting the hotspot gives the same result.

This seems like a different issue. There is no evidence to suggest that it's caused by the zero2 wlan firmware related problem this ticket is intended to track.

> no secrets: No agents were available for this request.
Aug 07 13:48:41 zero2w NetworkManager[518]: <info>  [1723031321.8542] device (wlan0): state change: need-auth -> failed (reason 'no-secrets', sys-iface-state: 'managed')

At a guess, maybe network manager has reused an existing connection which did not require security, but now does. Try cleaning out your NM connections and starting afresh.

Derpinbird, 2024-08-20:
I have the same issue. New image didn't solve the problem either. tried nmtui and nmcli but no luck

learmj, 2024-08-28:
> I have the same issue.

Which? The hotspot or the PSK offload timeout? I'd prefer to keep this ticket specifically for the latter.

SergeyRim, 2024-09-20:
I have the same issue with my Raspberry Pi Zero 2W and Mikrotik hap ax2 router.
Create file /etc/modprobe.d/brcmfmac.conf with
`options brcmfmac feature_disable=0x2000`
solves the issue.

ibn-e-battuta, 2024-10-10:
Any update on this?

learmj, 2024-10-10:
It has proven somewhat tricky to track down exactly where the issue is, or rather specifically what triggers the 4-way handshake to fail when that transaction is offloaded to the firmware. There is some evidence to suggest that's AP related.

ibn-e-battuta, 2024-10-11:
Oh okay, I got 2 Pi Zero 2Ws, I can't get them to work with the official Raspberry Pi OS Bookworm. I had to use Ubuntu images. On Ubuntu WIFI is very stable. But Ubuntu doesn't have rpicam-apps, which sucks for my use case.

ibn-e-battuta, 2024-10-14:
[   17.866354] brcmfmac: F1 signature read @0x18000000=0x1541a9a6
[   17.882078] brcmfmac: brcmf_fw_alloc_request: using brcm/brcmfmac43430-sdio for chip BCM43430/1
[   17.885109] usbcore: registered new interface driver brcmfmac
[   18.337399] brcmfmac: brcmf_c_process_txcap_blob: no txcap_blob available (err=-2)
[   18.338697] brcmfmac: brcmf_c_preinit_dcmds: Firmware: BCM43430/1 wl0: Jun 14 2023 07:27:45 version 7.45.96.s1 (gf031a129) FWID 01-70bd2af7 es7
[   19.587316] brcmfmac: brcmf_cfg80211_set_power_mgmt: power save enabled

learmj, 2024-10-14:
Is that for both distros?
Are the versions of wpa_supplicant the same in both distros, too?
```sudo wpa_supplicant -v``` will show you what's installed and ```sudo apt-cache policy wpasupplicant``` will show you what's available.

ibn-e-battuta, 2024-10-14:
> [ 17.866354] brcmfmac: F1 signature read @0x18000000=0x1541a9a6 [ 17.882078] brcmfmac: brcmf_fw_alloc_request: using brcm/brcmfmac43430-sdio for chip BCM43430/1 [ 17.885109] usbcore: registered new [...]

this is for ubuntu 24.04

SergeyRim, 2024-10-14:
I have a PiKVM, here are my data:

sudo dmesg | grep brcmfmac
```
[   12.550875] brcmfmac: F1 signature read @0x18000000=0x1541a9a6
[   12.566737] brcmfmac: brcmf_fw_alloc_request: using brcm/brcmfmac43430-sdio for chip BCM43430/1
[   12.580320] usbcore: registered new interface driver brcmfmac
[   13.007770] brcmfmac: brcmf_c_process_txcap_blob: no txcap_blob available (err=-2)
[   13.019152] brcmfmac: brcmf_c_preinit_dcmds: Firmware: BCM43430/1 wl0: Jun 14 2023 07:27:45 version 7.45.96.s1 (gf031a129) FWID 01-70bd2af7 es7
[   17.796563] brcmfmac: brcmf_cfg80211_set_power_mgmt: power save enabled
```

sudo wpa_supplicant -v
```
wpa_supplicant v2.10-hostap_2_10+
Copyright (c) 2003-2022, Jouni Malinen <j@w1.fi> and contributors
```
uname -a
`Linux pikvm 6.6.45-7-rpi #1 SMP Mon Sep  2 09:16:27 UTC 2024 armv7l GNU/Linux`

SergeyRim, 2024-10-14:
Installed Ubuntu 24.01 LTS via Pi imager -> the same issue with WiFi, not abe to connect to mikrotik router without a fix. So the is no difference between Ubuntu and Raspberry Pi OS.

ibn-e-battuta, 2024-10-14:
`options brcmfmac feature_disable=0x2000`

I am unable to connect to wifi with this.

SergeyRim, 2024-10-14:
> `options brcmfmac feature_disable=0x2000`
> 
> I am unable to connect to wifi with this.
Then you have another issue not related to this thread.

@learmj Is any way how we can troubleshoot this issue with connecting some router? It is a very annoying to add a fix every time when install new image.

fgma, 2024-10-14:
Same problem here. I tried connecting from different pi zero w/w 2 devices  to different mikrotik devices. Connecting to my phone as a hotspot works. Connecting to mikrotik APs doesn't. After adding the workaround and rebooting I get a working wifi connection to the mikrotik devices.

learmj, 2024-10-15:
Thanks @SergeyRim 

> So the is no difference between Ubuntu and Raspberry Pi OS.

Good to know. That's what I had hoped.

> @learmj Is any way how we can troubleshoot this issue with connecting some router? It is a very annoying to add a fix every time when install new image.

I hear you. I've only reproduced it with a Mikrotik hAP ac lite so far, but I'm sure there must be interop problems with other APs. If you're able to get any logging output from the AP itself when the connection fails, that might be useful (you can get syslog output from a Mikrotik for example). I'm running version RouterOS ```'v6.49.11 (stable)'```.

Curious - have you adjusted any settings on your Mikrotik (eg changed security profile/config since factory defaults)? Obviously restoring to factory defaults may be undesirable in many cases, but if you were able to zap it back to defaults I'd be interested to know if the problem persists (or has persisted since day 1 with the factory default security profile).

fgma, 2024-10-16:
@learmj Here are my logs collected on a mAP lite runnning routeros 7.16.1:

```20:12:13 wireless,debug wlan1: xx:xxxx:xx:xx:xx attempts to associate
20:12:13 wireless,debug wlan1: xx:xxxx:xx:xx:xx not in local ACL, by default accept
20:12:13 wireless,info xx:xxxx:xx:xx:xx@wlan1: connected, signal strength -61
20:12:13 wireless,info xx:xxxx:xx:xx:xx@wlan1: disconnected, unicast key exchange failed, signal strength -65
20:12:24 wireless,debug wlan1: xx:xxxx:xx:xx:xx attempts to associate
20:12:24 wireless,debug wlan1: xx:xxxx:xx:xx:xx not in local ACL, by default accept
20:12:24 wireless,info xx:xxxx:xx:xx:xx@wlan1: connected, signal strength -60
20:12:24 wireless,info xx:xxxx:xx:xx:xx@wlan1: disconnected, unicast key exchange failed, signal strength -71
20:12:38 wireless,debug wlan1: xx:xxxx:xx:xx:xx attempts to associate
20:12:38 wireless,debug wlan1: xx:xxxx:xx:xx:xx not in local ACL, by default accept
20:12:38 wireless,info xx:xxxx:xx:xx:xx@wlan1: connected, signal strength -74
20:12:38 wireless,info xx:xxxx:xx:xx:xx@wlan1: disconnected, unicast key exchange failed, signal strength -67
20:12:49 wireless,debug wlan1: xx:xxxx:xx:xx:xx attempts to associate
20:12:49 wireless,debug wlan1: xx:xxxx:xx:xx:xx not in local ACL, by default accept
20:12:49 wireless,info xx:xxxx:xx:xx:xx@wlan1: connected, signal strength -61
20:12:49 wireless,info xx:xxxx:xx:xx:xx@wlan1: disconnected, unicast key exchange failed, signal strength -65
20:13:03 wireless,debug wlan1: xx:xxxx:xx:xx:xx attempts to associate
20:13:03 wireless,debug wlan1: reject xx:xxxx:xx:xx:xx, banned (last failure - unicast key exchange failed)
20:13:14 wireless,debug wlan1: xx:xxxx:xx:xx:xx attempts to associate
20:13:14 wireless,debug wlan1: reject xx:xxxx:xx:xx:xx, banned (last failure - unicast key exchange failed)```

I got exactly the same issue with a hAP ax3 als running 7.16.1. My wireless configuration is completely custom so I don't know the difference to a factory default setup. In case it would really help narrowing down the issue I could reset the mAP lite to factory defaults.

learmj, 2024-10-17:
Thanks. I'd say this is a 4-way handshake error, indicative what I've seen with my mikrotik. As you pointed out, disabling the 4-way handshake in the wlan firmware prevents the problem from occurring.

chrismaster, 2024-11-07:
Same here with mikrotik ax3.
Adding options brcmfmac feature_disable=0x2000 didn't help.
Since I have enabled WPA2 PSK and WPA3 PSK @ mikrotik ax3, where is the best chance to get it goint? WPA3 Personal vs WPA & WPA2 Pers over nmtui ?
rpi zero 2 with wpi os lite 64-bit 22.10.24

poeggi, 2024-11-08:
Just reporting that I also see this issue.

Latest RaspiOS Lite release (22.Oct.2024), on a Zero 2 W. (Also tested the non-lite and earlier releases)
As routers/WLAN AP, I tested w/ FritzBox 6690, FritzBox 6670 and an iPhone14 pro "Personal Hotspot" (Option "maximize compatibility" set to enabled).

I could see all all SSID (sudo nmcli d wifi list), but connect to none ("Error ... Secrets were required, but not provided.")

I disable power-save, fixed some oddities with the locale config, tried to set up WiFi manually. Disabled and re-enabled WPA3 and PMF on my AP. Nothing seemed to help until I today found this issue.

I modified the brcmfmac.conf (as outlined earlier in this thread) and finally WiFi works.

lukaszkieronski, 2024-12-09:
> `options brcmfmac feature_disable=0x2000`
i believe it should be
 `options brcmfmac feature_disable=0x82000`
at least that is what worked for me

Strykar, 2025-04-01:
I am facing this issue with a Mikrotik [cAP aX](https://mikrotik.com/product/cap_ax) access point only, my RPi Zero 2W connects to my Android's hotspot just fine.

I suspect it has to do with WPA3 and Management Frames being required by the AP.
```
> scan_results
bssid / frequency / signal level / flags / ssid
34:71:46:04:c5:d5	2462	-44	[WPA2-PSK-CCMP][ESS]	wrlte
58:07:f8:d4:c1:d9	2437	-78	[WPA2-PSK-CCMP][ESS]	Airtel_deep_1723
78:9a:18:21:85:b9	2412	-88	[WPA2-SAE+FT/SAE-CCMP][SAE-H2E][ESS]	wrtea
```
I have tried all the solutions in this issue and https://github.com/raspberrypi/linux/issues/4976 but wifi won't connect.
RPi Zero 2W [wpa_supplicant and wpa_cli logs](https://gist.github.com/Strykar/7c5c89093fc533ade9c73a24585e1bbf).

nazgul6, 2025-04-05:
@Strykar literally just faced this issue on RPi Zero 2W trying to connect to Hap AX3 on ROS 7.18.2.

What worked was adding brcmfmac.conf - i.e., run **sudo nano /etc/modprobe.d/brcmfmac.conf** and add this line **options brcmfmac feature_disable=0x2000** and nothing else (note, it is 0x**2**000, not 0x82000 as suggested by others). Seems to persist reboots of the RPI OS. I need to figure out DietPi and PiKVM next...hopefully same solution will work...

FWIW hap ax3 wifi settings (excluding password and country settings for privacy):
set [ find default-name=wifi2 ] channel.band=2ghz-n .frequency=2437,2412 .width=20mhz .hide-ssid=yes .mode=ap .ssid="SSID"  disabled=no security.authentication-types=\
    wpa2-psk,wpa3-psk .connect-priority=0 .disable-pmkid=no

TheAdamWarlock, 2025-04-15:
> [@Strykar](https://github.com/Strykar) literally just faced this issue on RPi Zero 2W trying to connect to Hap AX3 on ROS 7.18.2.
>
> What worked was adding brcmfmac.conf - i.e., run '**sudo nano [...]

This works. I was facing the same issue.

marcsutton443, 2025-07-04:
I had been battling this issue for a painfully long amount of time and I have discovered that the issue is there if the signal is **merely within the range**. Here's how it was haunting me:

First Pi Zero 2W connected to Router 1: hAP ac, running ROS 6.49 with WPA2. Working happily since 2 years.
Second Pi Zero 2W tried to connect to Router 2 which was merely within the range of the 1st: hAP ax2 with ROS 7.19, running WPA2/3 with 802.11w. Fails to connect.

Just to see if there's any interference with router 1, I turn off WiFi on Router 1 and turn it back on again. And guess what? The first Pi Zero 2W starts failing to connect to Router 1. There was **nothing** that was changed on this Pi. There was **nothing** changed on Router 1, other than literally just the wifi being turned off and on. Rebooting router doesn't help. Rebooting Pi doesn't help. The only thing that helps is turning off Router 2, then rebooting the Pi connecting to Router 1.

This is the craziest issue I've ever seen. The mere presence of a router can wreck the WiFi on a device, but it's not even an attack. It's a mere broadcast!

The routers were few feet away from each other, using different frequencies and working absolutely fine with other devices like laptops and mobile phones. The only thing that was having issues were the Pi Zero 2W's.