raspberrypi/bookworm-feedback #283: NetworkManager/wpasupplicant: Ad-Hoc connections don't works

qrp73, 2024-07-30:
Tried to create ad-hoc connection based on [this instruction](https://www.thinkpenguin.com/gnu-linux/how-setup-ad-hoc-network), but it don't works on raspi-os bookworm.

Also tried some tweaks from other manuals with no result.

Steps to reproduce:
1. Open terminal
2. ```sudo nano /etc/NetworkManager/system-connections/ADEL.nmconnection``` and put the following content:
```
[connection]
id=ADEL
uuid=21058ee7-2cf5-8df5-baa4-87d4dca5ea55
type=wifi
autoconnect=false
interface-name=wlan0

[wifi]
band=bg
channel=7
mode=adhoc
mtu=1532
ssid=ADEL

[wifi-security]
group=ccmp;
key-mgmt=wpa-psk
pairwise=ccmp;
proto=rsn;
psk=1234567890

[ipv4]
address1=192.168.2.1/24
method=manual

[ipv6]
addr-gen-mode=stable-privacy
method=disabled

[proxy]
```
3. Press ```Ctrl+O```, ```Enter``` and ```Ctrl+X``` to save and exit
4. ```sudo chmod 600 /etc/NetworkManager/system-connections/ADEL.nmconnection```
5. ```sudo systemctl restart NetworkManager```
6. ```sudo nmcli radio wifi on```
7. ```sudo nmcli connection up ADEL```

Expected result: ADEL connection is up

Actual result: ```$ sudo nmcli connection up ADEL 
Error: Connection activation failed: 802.1X supplicant took too long to authenticate
Hint: use 'journalctl -xe NM_CONNECTION=7e435e77-ebe3-4348-b96f-f6c42dd69e20 + NM_DEVICE=wlan0' to get more details.```

paulchaffey, 2025-01-18:
Hi qrp73,
have you found a solution for this ?
Thx, Paul.

qrp73, 2025-01-19:
no, ad-hoc mode is broken in NetworkManager.