raspberrypi/bookworm-feedback #287: screen blanking in GUI login screen or lock screen not working as before

c-w-e, 2024-08-09:
Hi, 

not sure if this is a bug or works as designed.  

I was used to have the PI running with GUI started waiting in graphical login screen. When logged in I used "/usr/bin/dm-tool lock" to lock the screen while away using bullseye. This is not working in Bookworm anymore. 

In bullseye (using X11) the "screen blanking" setting was als respected on the login screen as well as the lock screen which it seems not to be in bookworm/Wayland.

Was not able to find exactly this details in the documentation on raspberry.com.

What I did:
I updated one of my Pis (Pi4) to the latest bookworm release (fresh install). screen blanking did not work anymore in the above mentioned scenario.
Before it was bullseye running X11 perfectly.

Is this expected behavior or how to get the screen blanking setting also working for log in and lock screen.

BR,
c-w-e

c-w-e, 2024-11-01:
With the new software release mentioned in https://www.raspberrypi.com/news/a-new-release-of-raspberry-pi-os/ with switching to labwc (and the fix for wayvnc - https://github.com/labwc/labwc/issues/2279#issuecomment-2448043931) there is a working setup.

Enabling screen blanking via "Raspberry PI Configuration" tool for user session works like a sharm (since wayvnc got the fix). Having a gui for adding the idle time would be nice, but editing the users labwc autostart is very simple to once you find out how that works and how Wayland/labwc handles the screen blanking scenario.

Having it manually enabled for the login screen is then easy, as you can edit /usr/share/labwc/autostart and add a similar line with swayidle & wlopm

Maybe a later enhancement to have the login screen automatically configured with the same setting in "Raspberry PI Configuration".

closed 2024-11-01 by c-w-e

MattiJarvinen, 2024-11-01:
@c-w-e note that lightdm lock screen opens a new VTY, and is completely skippable by changing back to the original VTY.

I switched to swaylock to fix that issue with lightdm & labwc.

MattiJarvinen, 2024-11-02:
For anyone wondering how to do it:

```
swayidle -w \
lock 'swaylock -f' \
timeout 300 'loginctl lock-session' \
timeout 500 'wlopm --off \*' \
resume 'wlopm --on \*'  &
```

I have some display resume problems currently so wlopm lines might cause problems. That or argon40 case or the HDMI cable or the old display or...