raspberrypi/bookworm-feedback #288: [regression]: "cannot get freq (v2/v3): err -110", "cannot set freq 44100 (v2/v3): err -110" after latest apt upgrade

qrp73, 2024-08-11:
today I executed apt upgrade, it completed ok. But I notice, that now sound periodically disappears with the following errors in the dmesg log:
```
[59082.276833] usb 1-1.4.4: 1:3: cannot get freq (v2/v3): err -110
[59087.397083] usb 1-1.4.4: 1:3: cannot set freq 44100 (v2/v3): err -110
[59092.516999] usb 1-1.4.4: uac_clock_source_is_valid(): cannot get clock validity for id 9
[59092.517028] usb 1-1.4.4: clock source 9 is not valid, cannot use
[59097.637106] usb 1-1.4.4: 1:3: cannot get freq (v2/v3): err -110
[59102.757202] usb 1-1.4.4: 1:3: cannot set freq 44100 (v2/v3): err -110
[59107.877231] usb 1-1.4.4: uac_clock_source_is_valid(): cannot get clock validity for id 9
[59107.877247] usb 1-1.4.4: clock source 9 is not valid, cannot use
[59112.997365] usb 1-1.4.4: 1:3: cannot get freq (v2/v3): err -110
[59117.457953] usb 1-1.4.4: 1:3: cannot set freq 44100 (v2/v3): err -71
[59117.462239] usb 1-1.4.4: uac_clock_source_is_valid(): cannot get clock validity for id 9
[59117.462257] usb 1-1.4.4: clock source 9 is not valid, cannot use
[59117.462617] usb 1-1.4.4: 1:3: cannot get freq (v2/v3): err -71
[59117.462988] usb 1-1.4.4: 1:3: cannot set freq 44100 (v2/v3): err -71
[59117.549356] usb 1-1.4.4: USB disconnect, device number 8
[59120.837190] usb 1-1.4.4: new high-speed USB device number 9 using xhci_hcd
[59120.958535] usb 1-1.4.4: New USB device found, idVendor=3302, idProduct=3352, bcdDevice= 0.01
[59120.958551] usb 1-1.4.4: New USB device strings: Mfr=1, Product=2, SerialNumber=0
[59120.958555] usb 1-1.4.4: Product: CX31993 384Khz HIFI AUDIO
[59120.958558] usb 1-1.4.4: Manufacturer: Synaptics
[59120.986902] input: Synaptics CX31993 384Khz HIFI AUDIO Consumer Control as /devices/platform/scb/fd500000.pcie/pci0000:00/0000:00:00.0/0000:01:00.0/usb1/1-1/1-1.4/1-1.4.4/1-1.4.4:1.3/0003:3302:3352.0006/input/input10
[59121.045560] hid-generic 0003:3302:3352.0006: input,hidraw3: USB HID v1.11 Device [Synaptics CX31993 384Khz HIFI AUDIO] on usb-0000:01:00.0-1.4.4/input3
```
When I disconnect sound DAC from USB and connect again, the sound works again.

This issue is regression in apt upgrade, because before latest apt upgrade the USB sound DAC worked ok.

I'm using CX31993 sound DAC dongle: https://www.aliexpress.com/item/1005005719157767.html

qrp73, 2024-08-11:
If it helps, the google shows similar issue here: https://guillaumeplayground.net/teac-ud-503-ud-501-snd-usb-audio-patch/
Maybe it is related, because it says that the issue happens when playback format is changed.

I notice that in my case the issue happens when application starts playing music or when player switch between different audio file, such events may lead to change sound playback sample rate.

qrp73, 2024-09-07:
I hadn't encountered this error for a long time, and I was almost sure that it had been fixed with some update. But no, it's still there...

Yesterday, I encountered it again. However, I'm not sure what condition causes it. It seems to happen when a new app tries to play a sound, but there might be some additional condition that triggers it. I'm not sure what that trigger is. It probably occurs when the app tries to play sound at a 44100 Hz sample rate, but I'm not certain. More testing is needed...

corus87, 2024-09-15:
I've come across the same issue with my new [Respeaker Lite](https://www.seeedstudio.com/ReSpeaker-Lite-p-5928.html) on a raspberry pi zero 2. I first thought about an problem with the Respeaker firmware but after finding this issue, I would assume an PI problem too.
 
I bought the Respeaker a couple of weeks ago, so I have no clue if this issue started with packages updates. 

I've described my issue [on the Seeed forum](https://forum.seeedstudio.com/t/respeaker-lite-usb-connections-issues-stops-working-after-couple-of-minutes/281341).

qrp73, 2024-09-27:
I suspect that this issue is somehow related with some memory management issue. Because it usually happens when some app starts to consume a lot of memory. I usually catch it with Octave, it also happens in audacity and other app. And it happens when I start operation which consume a lot of memory very quickly.

In this case it happens when I started Octave script which consume about 2 GB memory:
```
[44897.727131] bcm2835-codec bcm2835-codec: bcm2835_codec_flush_buffers: Timeout waiting for buffers to be returned - 0 outstanding
[106409.261811] usb 1-1.4.4: 1:3: cannot get freq (v2/v3): err -110
[106414.382049] usb 1-1.4.4: 1:3: cannot set freq 44100 (v2/v3): err -110
[106419.502225] usb 1-1.4.4: uac_clock_source_is_valid(): cannot get clock validity for id 9
[106419.502239] usb 1-1.4.4: clock source 9 is not valid, cannot use
[106424.622432] usb 1-1.4.4: 1:3: cannot get freq (v2/v3): err -110
[106429.742664] usb 1-1.4.4: 1:3: cannot set freq 44100 (v2/v3): err -110
[106438.450735] usb 1-1.4.4: uac_clock_source_is_valid(): cannot get clock validity for id 9
[106438.450753] usb 1-1.4.4: clock source 9 is not valid, cannot use
[106443.567342] usb 1-1.4.4: 1:3: cannot get freq (v2/v3): err -110
[106448.687485] usb 1-1.4.4: 1:3: cannot set freq 44100 (v2/v3): err -110
[106453.807671] usb 1-1.4.4: uac_clock_source_is_valid(): cannot get clock validity for id 9
[106453.807687] usb 1-1.4.4: clock source 9 is not valid, cannot use
[106458.927975] usb 1-1.4.4: 1:3: cannot get freq (v2/v3): err -110
[106464.048123] usb 1-1.4.4: 1:3: cannot set freq 44100 (v2/v3): err -110
[106469.168334] usb 1-1.4.4: uac_clock_source_is_valid(): cannot get clock validity for id 9
[106469.168350] usb 1-1.4.4: clock source 9 is not valid, cannot use
[106474.288439] usb 1-1.4.4: 1:3: cannot get freq (v2/v3): err -110
[106479.408644] usb 1-1.4.4: 1:3: cannot set freq 44100 (v2/v3): err -110
[106489.649135] usb 1-1.4.4: uac_clock_source_is_valid(): cannot get clock validity for id 9
[106489.649150] usb 1-1.4.4: clock source 9 is not valid, cannot use
[106494.769270] usb 1-1.4.4: 1:3: cannot get freq (v2/v3): err -110
[106499.889634] usb 1-1.4.4: 1:3: cannot set freq 44100 (v2/v3): err -110
[106555.187639] usb 1-1.4.4: uac_clock_source_is_valid(): cannot get clock validity for id 9
[106555.187656] usb 1-1.4.4: clock source 9 is not valid, cannot use
[106560.308249] usb 1-1.4.4: 1:3: cannot get freq (v2/v3): err -110
[106565.429164] usb 1-1.4.4: 1:3: cannot set freq 44100 (v2/v3): err -110
[106570.548398] usb 1-1.4.4: uac_clock_source_is_valid(): cannot get clock validity for id 9
[106570.548414] usb 1-1.4.4: clock source 9 is not valid, cannot use
[106575.672558] usb 1-1.4.4: 1:3: cannot get freq (v2/v3): err -110
[106580.788689] usb 1-1.4.4: 1:3: cannot set freq 44100 (v2/v3): err -110
[106585.908860] usb 1-1.4.4: uac_clock_source_is_valid(): cannot get clock validity for id 9
[106585.908875] usb 1-1.4.4: clock source 9 is not valid, cannot use
[106591.029048] usb 1-1.4.4: 1:3: cannot get freq (v2/v3): err -110
[106596.149323] usb 1-1.4.4: 1:3: cannot set freq 44100 (v2/v3): err -110
[106606.389585] usb 1-1.4.4: uac_clock_source_is_valid(): cannot get clock validity for id 9
[106606.389600] usb 1-1.4.4: clock source 9 is not valid, cannot use
[106611.509741] usb 1-1.4.4: 1:3: cannot get freq (v2/v3): err -110
[106616.630123] usb 1-1.4.4: 1:3: cannot set freq 44100 (v2/v3): err -110
[113653.930661] usb 1-1.4.4: uac_clock_source_is_valid(): cannot get clock validity for id 9
[113653.930680] usb 1-1.4.4: clock source 9 is not valid, cannot use
[113659.049818] usb 1-1.4.4: 1:3: cannot get freq (v2/v3): err -110
[113664.169820] usb 1-1.4.4: 1:3: cannot set freq 44100 (v2/v3): err -110
[113669.290504] usb 1-1.4.4: uac_clock_source_is_valid(): cannot get clock validity for id 9
[113669.290525] usb 1-1.4.4: clock source 9 is not valid, cannot use
[113674.409947] usb 1-1.4.4: 1:3: cannot get freq (v2/v3): err -110
[113679.533671] usb 1-1.4.4: 1:3: cannot set freq 44100 (v2/v3): err -110
[113684.650312] usb 1-1.4.4: uac_clock_source_is_valid(): cannot get clock validity for id 9
[113684.650343] usb 1-1.4.4: clock source 9 is not valid, cannot use
[113689.770198] usb 1-1.4.4: 1:3: cannot get freq (v2/v3): err -110
[113694.890343] usb 1-1.4.4: 1:3: cannot set freq 44100 (v2/v3): err -110
[113705.130441] usb 1-1.4.4: uac_clock_source_is_valid(): cannot get clock validity for id 9
[113705.130458] usb 1-1.4.4: clock source 9 is not valid, cannot use
[113710.250563] usb 1-1.4.4: 1:3: cannot get freq (v2/v3): err -110
[113715.370795] usb 1-1.4.4: 1:3: cannot set freq 44100 (v2/v3): err -110
[115173.807282] usb 1-1.4.4: uac_clock_source_is_valid(): cannot get clock validity for id 9
[115173.807303] usb 1-1.4.4: clock source 9 is not valid, cannot use
[115178.927416] usb 1-1.4.4: 1:3: cannot get freq (v2/v3): err -110
[115184.047765] usb 1-1.4.4: 1:3: cannot set freq 44100 (v2/v3): err -110
[115189.167713] usb 1-1.4.4: uac_clock_source_is_valid(): cannot get clock validity for id 9
[115189.167777] usb 1-1.4.4: clock source 9 is not valid, cannot use
[115194.287568] usb 1-1.4.4: 1:3: cannot get freq (v2/v3): err -110
[115199.407605] usb 1-1.4.4: 1:3: cannot set freq 44100 (v2/v3): err -110
[115204.531749] usb 1-1.4.4: uac_clock_source_is_valid(): cannot get clock validity for id 9
[115204.531773] usb 1-1.4.4: clock source 9 is not valid, cannot use
[115209.647668] usb 1-1.4.4: 1:3: cannot get freq (v2/v3): err -110
[... 51 lines cut ...]
[... cut ...]
[116272.056231] usb 1-1.4.4: Product: CX31993 384Khz HIFI AUDIO
[116272.056234] usb 1-1.4.4: Manufacturer: Synaptics
[116272.083977] input: Synaptics CX31993 384Khz HIFI AUDIO Consumer Control as /devices/platform/scb/fd500000.pcie/pci0000:00/0000:00:00.0/0000:01:00.0/usb1/1-1/1-1.4/1-1.4.4/1-1.4.4:1.3/0003:3302:3352.0006/input/input10
[116272.144434] hid-generic 0003:3302:3352.0006: input,hidraw3: USB HID v1.11 Device [Synaptics CX31993 384Khz HIFI AUDIO] on usb-0000:01:00.0-1.4.4/input3
```

After reconnecting USB sound card it starts to work again.

florian-asche, 2025-06-17:
Linked to https://github.com/respeaker/ReSpeaker_Lite/issues/8#issuecomment-2981011872

lurch, 2025-06-18:
@florian-asche The original report was about a different USB DAC to the one that you're using. Could you open a separate issue please? Thanks.

florian-asche, 2025-06-18:
> [@florian-asche](https://github.com/florian-asche) The original report was about a different USB DAC to the one that you're using. Could you open a separate issue please? Thanks.

@lurch in can do that, but do you want to open a issue for different hardware with the same issue?
I also have "cannot get freq" in my error logs. I think it is the same problem, only difference is, that my hardware is not working on reconnect, which can be a problem by my "soundcard".

lurch, 2025-06-19:
Yes, please open a separate issue. The error might "look" similar, but the USB and audio stacks are quite complex, so the actual underlying issue may be entirely different.