raspberrypi/bookworm-feedback #289: ValueError: (11, 'Resource temporarily unavailable')  when calling python3 math library

githubdoe, 2024-08-13:
the following Python3 line causes resource temporarily unavailable exception when running in a thread  but only every so often.  On the order of several minutes apart for a thread that is calling it about 10 times a second. 

 delta = math.degrees(math.acos(math.sin(
                            dec1) * math.sin(dec2) + math.cos(dec1) * math.cos(dec2)*math.cos(ra1 - ra2)))

lurch, 2024-08-14:
One of the replies in https://stackoverflow.com/questions/13340893/oserror-errno-11-resource-temporarily-unavailable-what-causes-this suggests that maybe you're creating too many threads?

However this seems to be a "general Python programming problem" rather than a "Raspberry Pi OS Bookworm problem", so I'm going to close this issue and suggest that you ask on https://forums.raspberrypi.com/viewforum.php?f=32 instead.

closed 2024-08-14 by lurch

githubdoe, 2024-08-14:
Thank you for your feedback.