raspberrypi/bookworm-feedback #292: Wastebasket (trash) icon does not reappear on desktop after removing tick and later reticking it via " Appearance settings"

MuppsGit, 2024-08-22:
Wastebasket icon does not reappear on desktop after removing tick and later reticking it in "Appearance settings" --> Desktop --> Wastebasket

spl237, 2024-09-06:
I am fiddling with the desktop at the moment and did see a crash associated with changing the desktop items in Appearance Settings when I was testing yesterday, so I have seen something similar - will investigate.

spl237, 2024-09-06:
Are you still using the default wayfire compositor, or have you used raspi-config to switch to labwc? (If this question makes no sense to you, then you are still using wayfire!) Many thanks.

closed 2024-09-11 by spl237

MuppsGit, 2024-09-27:
Hi Simon. Excuses for the late answer and thank you for trying to replicate the issue and find the cause.
Still using wayfire. Ive tried going back to defaults in Appearance settings but still didnt work. Maybe as u say the config file is corrupted in some way. Will see if i can find it and otherwise do a fresh install. Thanks again.

popcornmix, 2024-09-27:
A quick test here and it seems to work fine. Wastebasket disappears and appears with the checkbox.