raspberrypi/bookworm-feedback #296: audacity stopped to work after latest apt upgrade

qrp73, 2024-09-19:
Steps to reproduce:

1)  ```sudo apt update && sudo apt upgrade```
2)  Start audacity

Expected result: audacity is started and works (as it worked before apt upgrade)

Actual result: audacity is in dead loop at app splash

Please fix it

popcornmix, 2024-09-19:
I've just tried `sudo apt install audacity` on a fresh 64-bit RPiOS bookworm image and it launched okay.
Also tried on an older image (after `apt full-upgrade`) and it launched there too.

Do you have a USB audio device connected? Does it start without that connected?

popcornmix, 2024-09-19:
Note:
```
$ audacity

(process:1862): Gdk-CRITICAL **: 17:50:25.896: gdk_screen_get_root_window: assertion 'GDK_IS_SCREEN (screen)' failed

(process:1862): Gdk-CRITICAL **: 17:50:25.896: gdk_window_get_display: assertion 'GDK_IS_WINDOW (window)' failed

(process:1862): Gdk-CRITICAL **: 17:50:25.896: gdk_cursor_new_from_pixbuf: assertion 'GDK_IS_DISPLAY (display)' failed
```
appear on a working launch, so are likely unrelated to your issue.

qrp73, 2024-09-19:
I think it is freezes in dead loop due to error with sound card. Because when I disconnect the sound card from USB, it starts ok.

It appears that if I start youtube in firefox first (after reboot) the error don't happens.

popcornmix, 2024-09-19:
Any complaints about USB in dmesg?

It sounds more like a kernel issue than an audacity one.

qrp73, 2024-09-19:
yes, it seems like kernel issue with usb.

just tried to reboot and now it works ok. I suspect it may be related to specific video which was played in youtube, may be due to some specific sample rate.

qrp73, 2024-09-27:
It looks that this is the same issue as this one: https://github.com/raspberrypi/bookworm-feedback/issues/288

lurch, 2024-10-02:
Closing as duplicate of #288

closed 2024-10-02 by lurch