raspberrypi/bookworm-feedback #297: maxima segmentation violation error

foompah, 2024-09-22:
Steps to reproduce:

sudo apt install maxima
Start maxima
Expected result: maxima is started and works

Actual result: Segmentation violation error

app log:
$ maxima
mprotect failure: 0xb29000 41320448 : Invalid argument

Unrecoverable error: Segmentation violation..
Aborted

lurch, 2024-09-24:
This package comes directly from upstream Debian, so you should probably report it there? https://packages.debian.org/bookworm/maxima

foompah, 2024-09-28:
RPiOS 32-bit doesn’t have this bug. Ubuntu 64-bit doesn't have this bug. The wxmaxima flatpak doesn't have this bug. This is all on a RPi 5 8GB.

ghollingworth, 2024-09-29:
It's likely to be a 16kbyte page issue.

closed 2024-09-29 by ghollingworth

popcornmix, 2024-09-29:
> It's likely to be a 16kbyte page issue.

If it runs with `kernel=kernel8.img` in config.txt, then that would confirm this theory.

foompah, 2024-09-29:
Yes, with kernel=kernel8.img, it runs.

martignoni, 2025-05-23:
> This package comes directly from upstream Debian, so you should probably report it there? https://packages.debian.org/bookworm/maxima

See https://bugs.debian.org/cgi-bin/bugreport.cgi?bug=1106352 (thanks to @sangwinc).