raspberrypi/bookworm-feedback #299: Logging out of desktop kills SSH server session on same Pi

andrum993, 2024-09-29:
There is a feature in the current Bookworm version of Raspberry Pi OS which will actively kills all user processes when the user in question logs off from the graphical desktop. This feature goes beyond that optionally implemented in systemd, which can be configured to kill all user processes that belong to the same session. In this case, any process owned by the user who logs off from the graphical desktop will be killed without warning. This was apparently added in the interests of security, however in my view this is actually a security threat, since it amounts to a form of denial of service. Please remove this feature. If you do not wish to remove this feature, then please at least give us the option to switch it off, and document how to perform this configuration. If this feature is retained, then it should of course also be documented.

(I would have linked to the GitHub issue where this feature was revealed, but unfortunately I can't find it).

lurch, 2024-09-30:
Which exact OS version are you using? https://downloads.raspberrypi.com/raspios_armhf/release_notes.txt suggests that this behaviour was added in the 2024-03-12 release:
```
Shutdown assistant now closes all user processes when logging out
```
but then reverted in the 2024-07-04 release:
```
pishutdown - revert to original use of pkill to close desktop
```

andrum993, 2024-09-30:
Apologies - this one is user error. I must have been shutting down the Pi instead of logging out when I checked it recently, having checked a few weeks ago when I did not get kicked out of my SSH session. I've checked with both a clean install of the 2024-07-04 image, and the 2024-03-12 image fully upgraded using apt, and both show the correct behaviour, i.e. the SSH session stays up. Sorry!

closed 2024-09-30 by andrum993