raspberrypi/bookworm-feedback #303: pcmanfm - File Manager - persistent issue with Copy-Paste functionality (Paste is not available after first Copy)

qrp73, 2024-10-10:
The "Paste" option is unavailable after single click on "Copy" menu, requiring the user to click "Copy" menu at least twice in order to use "Paste" menu.

###Steps to Reproduce:

1) Open File Manager (pcmanfm).
2) Right-click on a file and select "Copy" from the context menu.
3) Double-click on another folder to change the current directory.
4) Right-click in the new folder and attempt to select "Paste" from the context menu.

**Expected Result:** The "Paste" option should be available and should paste the previously copied file into the new folder.

**Actual Result:** The "Paste" option is unavailable. The user is forced to select "Copy" at least twice before being able to paste the file.

**Impact:** This issue has been present for a long time and is highly disruptive to workflow, requiring unnecessary repetition of actions to complete a basic task. It greatly affects usability and efficiency in file management.

**Recommendation:** Fix the issue so that the "Paste" become accessible after single select "Copy" menu, without requiring the user to re-select "Copy" again.

lurch, 2024-10-10:
I'm not able to reproduce this - it all works perfectly fine for me. Can you please provide more details about which OS image / desktop environment / Raspberry Pi / etc. you're using?

qrp73, 2024-10-10:
Raspberry Pi 4B rev 1.5
```
$ uname -a && cat /etc/rpi-issue && vcgencmd version
Linux raspi 6.6.51+rpt-rpi-v8 #1 SMP PREEMPT Debian 1:6.6.51-1+rpt3 (2024-10-08) aarch64 GNU/Linux
Raspberry Pi reference 2023-09-22
Generated using pi-gen, https://github.com/RPi-Distro/pi-gen, 40f37458ae7cadea1aec913ae10b5e7008ebce0a, stage4
Aug 30 2024 19:17:39 
Copyright (c) 2012 Broadcom
version 2808975b80149bbfe86844655fe45c7de66fc078 (clean) (release) (start)
```

As I remember this issue was present just out of the box after all Raspberry Pi OS installation.
Currently I also installed clipman, but this issue was present before installation.

This is very strange that you cannot reproduce it. Do you use context menu by right click mouse?

lurch, 2024-10-10:
> I'm using Detailed List View

Ah! That seems to be the problem. If I click on the "View as detailed list" button on the toolbar, then the right-click "Paste" menu indeed appears to stop working. Although the Edit->Paste menu _does_ still seem to work, so I guess that's a workaround for now whilst we investigate this problem...

popcornmix, 2024-10-10:
Just tried and a copied a file and pasted it into a new directory just fine. This was using right-click menu options.
This was on a Pi5 with labwc enabled.

lurch, 2024-10-10:
Ah-ha! After a bit more head-scratching, myself and @popcornmix have managed to figure out why it works sometimes but not others. :bulb: 

If you've switched pcmanfm into List-view, **and** you're trying to right-click-Paste into your home directory or an immediate subdirectory of your home directory, then the Paste silently fails. But if you try and Paste into a sub-sub-directory (which is what @popcornmix was trying, then the Paste succeeds! How bizarre.

Detailed reproduction steps:
1. Launch pcmanfm (File Manager) and create a new directory in your home folder e.g. `SubDir`
2. Navigate into `SubDir` and create another directory inside there e.g. `SubSubDir` (i.e. so the full path is something like `/home/pi/SubDir/SubSubDir`)
3. Navigate back to your home directory
4. Create two new empty files e.g. `File1` and `File2`
5. Make sure that pcmanfm is in "Icon View" mode
6. Right-click on File1, select Copy
7. Navigate to SubDir
8. Right-click and select Paste - this works and creates File1
9. Navigate back to your home directory
10. Right-click on File2, select Copy
11. Navigate to SubSubDir
12. Right-click and select Paste - this works and creates File2
13. Now delete SubDir/File1 and SubDir/SubSubDir/File2
14. Navigate back to your home directory
15. Switch pcmanfm to "List View" mode
16. Right-click on File1, select Copy
17. Navigate to SubDir
18. Right-click and select Paste - **this silently fails**
19. Navigate back to your home directory
20. Right-click on File2, select Copy
21. Navigate to SubSubDir
22. Right-click and select Paste - this works and creates File2

qrp73, 2024-10-10:
it doesn't depends on sub folder structure. 

The Paste just don't works if you do Copy operation once. 
If you do Copy operation twice for the same file, the Paste operation works.

Also, there is need to restart pcmanfm after setting changes.

qrp73, 2024-10-19:
Tested on old Bullseye image, it don't have such issue. So it appears in Raspi OS Bookworm.

qrp73, 2024-11-04:
Tested it on a clean raspios install. And found additional details on how to reproduce it. 

In order to reproduce this issue, you need to install clipman: ```sudo apt install clipman``` and add it to wayfire or labwc config.

For wayfire ```nano ~/.config/wayfire.ini```:
```
[autostart]
clipman-restore = clipman restore
clipman-store = wl-paste -t text --watch clipman store
```

For labwc ```mkdir -p ~/.config/labwc && nano ~/.config/labwc/autostart```:
```
/usr/bin/wl-paste -t text --watch clipman store &
```

After that reboot the system and you will be able to reproduce the issue with non working Paste operation for the single copy operation.

lurch, 2024-12-08:
> This the most annoying issue of Raspberry Pi OS. You catch it every time when copy/paste files. Any chance to fix it?

> In order to reproduce this issue, you need to install clipman

`clipman` isn't installed by default in Raspberry Pi OS (and we don't mention it anywhere in our [official documentation](https://www.raspberrypi.com/documentation/) ), and this is why fixing this "most annoying issue" isn't very high on our list of priorities.

qrp73, 2024-12-09:
Is there an alternative method that allows copy/paste functionality without needing to keep both applications open until the operation is complete?

By default, Raspberry Pi OS Bookworm loses copied text when you close the window, which is very inconvenient. If you close an unneeded window, the copied data is lost, and you can no longer paste it. This behavior differs from Raspberry Pi OS Bullseye, where the copied text would persist.

Using Clipman can help preserve clipboard data, but it leads to bugs with copy/paste operations in pcmanfm within Raspberry Pi OS. These issues in Raspberry Pi OS leaving users with limited options and unresolved usability problems, because you're limited with just two option - use clipman or not use, but both option have bugs and any choose leads to very annoying usability issues. Please fix it.

Copy/paste is very important part of modern OS and it's sad to see broken copy/paste behavior in Raspberry Pi OS.

lurch, 2024-12-09:
> By default, Raspberry Pi OS Bookworm loses copied text when you close the window, which is very inconvenient. If you close an unneeded window, the copied data is lost, and you can no longer paste it. This behavior differs from Raspberry Pi OS Bullseye, where the copied text would persist.

I believe that this is simply a feature of how Wayland works, rather than a bug in Raspberry Pi OS Bookworm.
Personally, I find that whenever I'm copy'n'pasting text between two applications, I always have both applications still running anyway, i.e. I always do "copy, switch to other application, paste", I never do "copy, close original application, switch to other application, paste".

spl237, 2024-12-09:
Wayland does not support copying between two applications or windows unless they are both still running - this is a consequence of the security design of Wayland and nothing to do with anything we have done.

We don't support detailed list view any more, which is why it is disabled by default. Any bugs which only show up in that view are not going to be fixed by us.

qrp73, 2024-12-10:
> We don't support detailed list view any more, which is why it is disabled by default. Any bugs which only show up in that view are not going to be fixed by us.

this is sad, because this is the only useful view mode for file list, it allows to see file name and its size and other properties.

lurch, 2024-12-29:
Apparently a fix was devised in #343 , so closing this.

closed 2024-12-29 by lurch

ic-dev21, 2025-06-17:
Any headway made with this? I have two raspi OS installs and I am unable to paste folders or files, option is always greyed out.