raspberrypi/bookworm-feedback #308: ETH keeps dropping

bisudw, 2024-10-28:
If LAN is directly connected from Laptop to PI (PI3 in my case) the eth0 keeps dropping every now and them, in every 4-5 minues.

I tried with PI0 where I had a w5500 connected to SPI, there also got the same issue.
Also the network icon keeps going crazy. 

None of these happen when its connected to a swith. Issue happens when connected directly peer to peer network.

Everything works file in Bullseye though.

bisudw, 2024-11-03:
Its Win 11.|
I tried in Win10 also, But the result was same.

qrp73, 2024-11-04:
I had the same issue between RPI Zero 2W and RPI4. It happened due to "green technology feature" for power save... 

Try add ```dtparam=eee=off``` to ```/boot/firmware/config.txt```, it will disable "green technology feature". In my case it solve the issue and I have stable connection.

bisudw, 2024-12-27:
No this did not solve the issue. Still facing the problem.