raspberrypi/bookworm-feedback #309: After changing to pi-greeter-labwc lightdm login background vanished

MattiJarvinen, 2024-10-29:
Greeter background is just a solid color and not the same/similar image as /usr/share/rpd-wallpaper/RPiSystem.png it used to be.

Also I can't seem to find a way to configure a background for pi-greeter-labwc.

MattiJarvinen, 2024-10-31:
Ah, found it there was a recent change done to the pi-greeter that adding drop shadow, it removed the greeter background.

https://github.com/raspberrypi-ui/pi-greeter/commit/82cb5c4647bf427c4145380379e8e9debc991dfe

wallpaper= /usr/share/rpd-wallpaper/RPiSystem.png
wallpaper_mode=crop

spl237, 2024-11-04:
By design - the trick used to give a shadowed undecorated greeter window causes a spurious border on labwc; a plain background hides this.

closed 2024-11-04 by spl237