raspberrypi/bookworm-feedback #310: HDMI Touchscreen Mouse Events Inconsistently Follow Touch Events on Bookworm System (Kernel 6.6.51)

Tiger0215, 2024-10-31:
### Describe the bug

On the latest Bookworm system (kernel version 6.6.51), the HDMI touchscreen can generate touch events normally, but the mouse pointer does not always follow the touch events. This leads to the inability to trigger the double-click effect.

### Steps to reproduce the behaviour

1. Boot the Raspberry Pi with the Bookworm OS (kernel 6.6.51).
2. Connect an HDMI touchscreen.
3. Attempt to perform a double-click using the touchscreen.
4. Observe that the mouse pointer does not consistently follow the touch events, resulting in the double-click effect not being triggered.

### Device (s)

Raspberry Pi 5

### System

pi@raspberrypi:~ $ cat /etc/rpi-issue
Raspberry Pi reference 2024-10-22
Generated using pi-gen, https://github.com/RPi-Distro/pi-gen, 94f7acf599bd0fd119f523b5afefaccf7a7a1024, stage4

pi@raspberrypi:~ $ vcgencmd version
2024/09/23 14:02:56 
Copyright (c) 2012 Broadcom
version 26826259 (release) (embedded)

pi@raspberrypi:~ $ uname -a
Linux raspberrypi 6.6.51+rpt-rpi-v8 raspberrypi/linux#1 SMP PREEMPT Debian 1:6.6.51-1+rpt3 (2024-10-08) aarch64 GNU/Linux

### Logs

pi@raspberrypi:~ $ xinput 
WARNING: running xinput against an Xwayland server. See the xinput man page for details.
Virtual core pointer                    	id=2	[master pointer  (3)]
    Virtual core XTEST pointer              	id=4	[slave  pointer  (2)]
    xwayland-pointer:13                     	id=6	[slave  pointer  (2)]
    xwayland-relative-pointer:13            	id=7	[slave  pointer  (2)]
    xwayland-pointer-gestures:13            	id=8	[slave  pointer  (2)]
    xwayland-touch:13                       	id=10	[slave  pointer  (2)]
Virtual core keyboard                   	id=3	[master keyboard (2)]
    Virtual core XTEST keyboard             	id=5	[slave  keyboard (3)]
    xwayland-keyboard:13                    	id=9	[slave  keyboard (3)]

pi@raspberrypi:~ $ lsusb
Bus 004 Device 001: ID 1d6b:0003 Linux Foundation 3.0 root hub
Bus 003 Device 002: ID 0712:000a WS-070-7EP WS-070-7EP
Bus 003 Device 001: ID 1d6b:0002 Linux Foundation 2.0 root hub
Bus 002 Device 001: ID 1d6b:0003 Linux Foundation 3.0 root hub
Bus 001 Device 001: ID 1d6b:0002 Linux Foundation 2.0 root hub

popcornmix, 2024-10-31:
And single click is working, and double-click is a software feature of the OS, I'm transferring this issue to bookworm-feedback.

My guess is double-click requires the two taps to be close (in pixel coordinates, and time), and your device may not be meeting that. @spl237 may know more about what should and shouldn't work.

pelwell, 2024-10-31:
Install evtest (`sudo apt install evtest`) and use it to look at the events generated from touches. If you can see errors in that output then post it here - you've come to the right place. Otherwise it's a Wayland or desktop bug and we'll have to redirect you.

pelwell, 2024-10-31:
(My comment applied when this was in the linux repo, but I still think the output from evtest may be useful)

lurch, 2024-10-31:
Providing more details about which "HDMI touchscreen" you're using might be very useful too.

> On the latest Bookworm system (kernel version 6.6.51), the HDMI touchscreen can generate touch events normally, but the mouse pointer does not always follow the touch events.

Is this a regression in behaviour (i.e. has this touchscreen ever worked correctly with previous versions of Raspberry Pi OS or earlier versions of Raspberry Pi OS Bookworm?), or has this **always** been "problematic"?

spl237, 2024-10-31:
There are two parameters governing what is detected as a double-click, gtk-double-click-time (in milliseconds) and gtk-double-click-distance (in pixels).

They are set in the file /usr/share/themes/PiXflat/gtk-3.0/settings.ini - we set distance to 25 pixels, and time is the default value of 400ms. You can try increasing one or both of those values to see if that makes a difference. I get consistent double-click detection in the file manager on an HDMI touchscreen with our current values.

Bear in mind these only affect GTK applications - in what application are you finding that double-clicks are not always detected? Is it our file manager (which is GTK-based) or something else?

Tiger0215, 2024-11-01:
After testing with the event program, I found that both touch events and mouse events are responding normally. However, in the Bookworm system, the mouse icon only responds in specific windows. For example, when I touch the top window position of the File Manager, the mouse icon displays correctly and triggers the double-click zoom feature. But once I move the touch event to other folders within the File Manager, the mouse icon does not follow; it remains at the top window position, preventing the double-click to enter the folder. 

Steps to reproduce the behaviour:

1. Boot the Raspberry Pi with the latest Bookworm system.
2. Connect the HDMI touchscreen.
3. Open the File Manager and attempt to touch at the top window position, observing the behavior of the mouse icon.
4. Move the touch event to other folders within the File Manager and observe if the mouse icon follows the response.

The event program listens for both mouse and touch events, and the touch coordinates are responding correctly. I suspect the issue lies in the handling of the mouse icon and mouse events coordination. This problem has existed in previous versions (where even the mouse icon did not appear, making it impossible to double-click to zoom on windows). I noticed that the latest labwc system added support for double-clicking on touchscreens, so I was eager to test it. I hope this information is helpful, and I appreciate everyone's assistance in advance.

Tiger0215, 2024-11-01:
To better illustrate the issue, I’ve attached test images:

“First Touch” shows the touch at the top of the window, where the mouse icon responds normally, and the double-click function works correctly.

“Last Touch” shows the touch at any other position (for comparison, I touched near the bottom right corner at approximately the coordinates 1280, 800). You can see that the mouse icon remains in place and does not move with the new touch position.

By the way, what determines the size of the on-screen keyboard? I’ve noticed that on some touchscreens, the keyboard appears too small, making it difficult to read the letters. I suspect it might be related to the EDID size settings of the touchscreen.
[image]

spl237, 2024-11-01:
Ah - those symptoms sound as if you haven't associated the touchscreen with the display.

Launch the Screen Configuration tool from the Prefefences menu, right-click your display, and from the pop-up menu choose Touchscreen - make sure the driver for your touchscreen is ticked, then close the app. You may need to reboot.

Try that...

Tiger0215, 2024-11-01:
Based on the method you provided, I successfully resolved the issue! However, I have two more questions:

1. When using the official 7-inch DSI Touchscreen from Raspberry Pi without manually specifying the touchscreen, everything responds correctly.
2. When only one touchscreen is connected, is it possible to have the system automatically assign it as the default touchscreen, and only provide the option for the user to specify a touchscreen when the number of touchscreens is ≥2? I believe this would improve the user experience.
In any case, my issue has been resolved. I apologize for my oversight, and I’d like to thank everyone again for their time and assistance.

Tiger0215, 2024-11-01:
> Ah - those symptoms sound as if you haven't associated the touchscreen with the display.
>
> Launch the Screen Configuration tool from the Prefefences menu, right-click your display, and from the [...]

Based on the method you provided, I successfully resolved the issue! However, I have two more questions:

When using the official 7-inch DSI Touchscreen from Raspberry Pi without manually specifying the touchscreen, everything responds correctly.
When only one touchscreen is connected, is it possible to have the system automatically assign it as the default touchscreen, and only provide the option for the user to specify a touchscreen when the number of touchscreens is ≥2? I believe this would improve the user experience.
In any case, my issue has been resolved. I apologize for my oversight, and I’d like to thank everyone again for their time and assistance.

Tiger0215, 2024-11-01:
> Ah - those symptoms sound as if you haven't associated the touchscreen with the display.
>
> Launch the Screen Configuration tool from the Prefefences menu, right-click your display, and from the [...]

Based on the method you provided, I successfully resolved the issue! However, I have two more questions:

1. When using the official 7-inch DSI Touchscreen from Raspberry Pi without manually specifying the touchscreen, everything responds correctly.
2. When only one touchscreen is connected, is it possible to have the system automatically assign it as the default touchscreen, and only provide the option for the user to specify a touchscreen when the number of touchscreens is ≥2? I believe this would improve the user experience.
In any case, my issue has been resolved. I apologize for my oversight, and I’d like to thank everyone again for their time and assistance.

spl237, 2024-11-01:
> > Ah - those symptoms sound as if you haven't associated the touchscreen with the display.
> > Launch the Screen Configuration tool from the Prefefences menu, right-click your display, and from the pop-up menu choose Touchscreen - make sure the driver for your touchscreen is ticked, then close the app. You may need to reboot.
> > Try that...
> 1. When using the official 7-inch DSI Touchscreen from Raspberry Pi without manually specifying the touchscreen, everything responds correctly.

Yes, because we know the details of our own devices, we are able to set up default associations for them.

> 2. When only one touchscreen is connected, is it possible to have the system automatically assign it as the default touchscreen, and only provide the option for the user to specify a touchscreen when the number of touchscreens is ≥2? I believe this would improve the user experience.

See above - we can't predict the driver or name of touch devices other than the ones we manufacture, so we can't pre-associate third party devices.

closed 2024-11-01 by spl237