raspberrypi/bookworm-feedback #313: In labwc menu in the bottom are cut out of the page when there are too many items

archisman-panigrahi, 2024-11-01:
I have a lot of items in my Accessories, and they don't fit within the screen. In X11, I can scroll through the menu to access the items in the bottom.

In labwc, there is no way to scroll, and the menu items which do not fit inside the screen are not shown and there is no way to access them.

It would be nice to have a scrollbar in the menu in labwc.

Consolatis, 2025-01-07:
I assume we are talking about the panel menu, not a labwc built in menu here? If yes there is a weird thing with layershell that I've run into as well that makes xdg_popups not showing scroll bars when created via gtk layershell.
See
- https://github.com/wmww/gtk-layer-shell/issues/148

spl237, 2025-02-27:
Now fixed - should be in apt later on today.

closed 2025-02-27 by spl237