raspberrypi/bookworm-feedback #314: Changing screen orientation crashes desktop with labwc

jatofg, 2024-11-02:
After updating my Raspberry Pi 3B+ with Rasberry Pi OS bookworm 32-bit and enabling the new labwc compositor, attempting to invert the screen orientation crashes the desktop and sends me to the login screen. The desktop subsequently always crashes on startup. I'm using the official Raspberry Pi 7" touchscreen display.

When I try to invert the screen orientation and click Apply in the Screen Configuration tool, a new file .config/kanshi/config is created with the following content (and the crash happens instantly):

```
profile {
        output DSI-1 mode 800x480@60.029Hz position 0,0 transform 180
}
```

When I delete this file, I can log in again, but the screen orientation is not rotated, of course.

I'm attaching the .xsession-errors, an .xsession-errors without the rotation for comparison, and the /boot/firmware/config.txt. If you need any other logs or files to investigate the issue, I'll be happy to provide them.

[config.txt](https://github.com/user-attachments/files/17607062/config.txt)
[xession-errors_withoutRotation.txt](https://github.com/user-attachments/files/17607063/xession-errors_withoutRotation.txt)
[xsession-errors.txt](https://github.com/user-attachments/files/17607064/xsession-errors.txt)

cillian64, 2024-11-05:
Thanks for the report, I think this is caused by a pixman issue we're aware of and working on.  Please could you try the following and see if it resolves the issue?

`sudo apt install libpixman-1-0=0.42.2-1 libpixman-1-dev=0.42.2-1`

jatofg, 2024-11-05:
Yes, that resolves it

Consolatis, 2024-11-05:
@cillian64 Is it fixed upstream and should labwc depend on a later version of pixman?

cillian64, 2024-11-05:
> @cillian64 Is it fixed upstream and should labwc depend on a later version of pixman?

This is actually a bug introduced in our downstream pixman package (which adds some extra ARM optimisations).

Consolatis, 2024-11-05:
> This is actually a bug introduced in our downstream pixman package (which adds some extra ARM optimisations).

Got it, thanks.

cillian64, 2025-01-30:
We've been shipping `libpixman-1-0` `0.42.2-1+rpt3` for a few months which removes these optimisations and fixes this issue.  At some point we'd like to reinstate the optimisations in a fixed form, but this issue can be closed.

closed 2025-01-30 by jatofg

Consolatis, 2025-01-30:
As a daily user of an aarch64 system myself I would appreciate if those fixes could be proposed upstream.

cillian64, 2025-01-30:
> As a daily user of an aarch64 system myself I would appreciate if those optimizations could be proposed upstream.

It doesn't look like we actually have any downstream pixman patches specifically for aarch64.  I believe our aarch64 patches (by Ben Avison) got upstreamed, or otherwise upstream ended up with equivalent aarch64 fast paths.  Almost all of the downstream patches are armv6/armv7, with a couple to the platform-independent fast paths.

That said, I'd still prefer we try to upstream patches if they are worthwhile.  Given that wlroots is now requiring pixman 0.43 (we're still shipping 0.42) I'll have to do some rebasing at some point, and will try then to work out what we can submit upstream.