raspberrypi/bookworm-feedback #315: Screen capture issue with wayland

Intranox, 2024-11-05:
It's nice to see that Raspberry Pi Os is adapting to Wayland. However, in everyday use there are still things that don't work as they should and above all I am referring to software that wants to interact with windows or the screen. If you want to use Flameshot to take a screenshot of a portion of the screen this does not work and there is no way to fix it, even with other similar software. Similar problem if you want to share the screen of a single window, for example on Telegram there is no such possibility which instead is present and working in other DE that also use Wayland. Surely there will be many technical details that I am not aware of but even today with every update I prefer to go back to X11
Ps. Another thing that I would like to have on Wayland, but that I have not yet found, is a lightweight working clipboard manager.

lurch, 2024-11-05:
As I understand it, the security restrictions in Wayland mean that applications are better isolated from each other (as mentioned in 
https://www.raspberrypi.com/news/a-new-release-of-raspberry-pi-os/ ), which might explain some of what you're seeing? @spl237 might be able to provide more technical details.

Note that pressing the PrtScn button on your keyboard **will** take a screenshot of your desktop (using the `grim` tool), and save it in your `Pictures` folder.

spl237, 2024-11-05:
As @lurch says above, you cannot use X screenshot tools such as Flameshot with Wayland, as they cannot access the display buffer. This is a fundamental part of the design of Wayland and cannot be changed.

If you want to take a screenshot, you need to use a Wayland-aware tool such as grim, which is what is linked to the PrtScrn button.

closed 2024-11-05 by spl237

Intranox, 2024-11-05:
> COME@lurchdice sopra, non puoi usare strumenti di screenshot X come Flameshot con Wayland, poiché non possono accedere al buffer di visualizzazione. Questa è una parte fondamentale del design di Wayland e non può essere cambiata.
> 
> Per effettuare uno screenshot, è necessario utilizzare uno strumento compatibile con Wayland, come Grim, che è collegato al pulsante PrtScrn.

Let's say that it's not exactly like that, it's true that the problem arises with Wayland but there are also solutions like those adopted in desktop environments like Gnome or Kde in which both Flameshot and single window sharing work well on Wayland

Intranox, 2024-11-05:
> A quanto ho capito, le restrizioni di sicurezza in Wayland fanno sì che le applicazioni siano meglio isolate le une dalle altre (come menzionato in https://www.raspberrypi.com/news/a-new-release-of-raspberry-pi-os/ ), il che potrebbe spiegare in parte ciò che vedi?@spl237potrebbe essere in grado di fornire maggiori dettagli tecnici.
> 
> Tieni presente che premendo il tasto PrtScn sulla tastiera **verrà** effettuato uno screenshot del desktop (utilizzando lo `grim`strumento) e salvato nella `Pictures`cartella.

Ok, however as I was saying above these programs work on other DE that use Wayland. I could also use Grim but in everyday use it is much easier to be able to take a screenshot of only part of the screen and copy it to the clipboard for quick sending rather than saving the file and cropping it appropriately. Then I mentioned two cases in my everyday use, there may be many other software with limitations of this kind

spl237, 2024-11-09:
It appears from https://flameshot.org/docs/guide/wayland-help/ that Flameshot requires additional packages to be installed to work with Wayland - perhaps you could try following the directions on that page.