raspberrypi/bookworm-feedback #316: Accidental press Alt+Ctrl+Space triggers unattended installation of unwanted packages with no user confirmation and with no cancel option

qrp73, 2024-11-05:
Yesterday, I encountered an annoying issue: accidental pressing Alt+Ctrl+Space triggers the execution of the keybind command ```gui-pkinst orca reboot```. As a result, this command installs a large number of unwanted packages without any confirmation or option to cancel.

I spent a significant amount of time removing these packages, which also led to the removal of essential packages, requiring additional time to restore the system. All of this happened simply because of an accidental press of Alt+Ctrl+Space.

Could you please fix this issue? 
Ideally, add a prompt requiring a password to confirm package installations initiated through a keybind shortcut. Or just remove these keybind.

As a workaround you can fix it by remove this section from ```/etc/xdg/labwc/rc.xml``` and from ```/etc/xdg/openbox/lxde-pi-rc.xml``` and from ~/.config/labwc/rc.xml:
```
    <keybind key="C-A-space">
      <action name="Execute">
         <command>gui-pkinst orca reboot</command>
      </action>
    </keybind>
```

And remove this section from ```/etc/wayfire/template.ini``` and from ```~/.config/wayfire.ini```:
```
binding_orca = <ctrl> <alt> KEY_SPACE
command_orca = gui-pkinst orca reboot
```

lurch, 2024-11-06:
If you boot up your Pi for the first time with speakers attached, you'll hear it tell you "Press Ctrl Alt Space to install a screen reader", so I assume that this is intended behaviour (for the benefit of blind and partially-sighted people).

qrp73, 2024-11-06:
But it should give user the chance to cancel that destructive operation if he pressed it accidentally as in my case.

lurch, 2024-11-06:
If you "accidentally" press the power button on your Pi 5 twice in a row, it automatically shuts down without giving you the chance to "cancel"...

You're the first person to ever complain about accidentally pressing Alt+Ctrl+Space so I'm 99.9% sure we won't be changing this behaviour.

qrp73, 2024-11-06:
I spend about 6 hours to restore the system just due to accidental pressing of this keys combination which I even didn't know about before. That's the issue.

spl237, 2024-11-06:
You can just uninstall in Recommended Software. No need to rebuild entire systems or anything of the like. 

It is unreasonable to expect partially-sighted users to enter passwords or anything of the like. 

It is also not possible to cancel an apt install once it is in progress, or at least not without potentially leaving the system in a broken state.

This behaviour isn’t going to be changed. We have an obligation to make life as easy as possible for physically-disadvantaged users; it is no more than a very occasional minor annoyance for anyone else.

qrp73, 2024-11-06:
I’ve already removed the keybind for this key combination, so hopefully, this won’t happen again. But it’s definitely a dangerous shortcut. Adding at least a confirmation window could prevent a lot of issues. Unfortunately, by default, this key combination triggers an unconditional installation of components, without asking the user if they really want it and without any option to cancel the operation.

And I’ll note that killing the system with an accidental key press definitely doesn’t make life easier - instead, it complicates it.

lurch, 2024-11-06:
I don't know what you did wrong whilst trying to uninstall Orca, but please stop describing this as a "destructive operation" which "kills systems".

spl237, 2024-11-06:
Your feedback is noted. 

But the behaviour will not be changed.

qrp73, 2024-11-06:
> I don't know what you did wrong whilst trying to uninstall Orca, but please stop describing this as a "destructive operation" which "kills systems".

I refer to this as a "destructive operation" because, after enabling this feature, the chaotic word bursts from the speaker left me feeling completely helpless. I was unable to do anything or even analyze what had happened until I turned off the speakers. These random word bursts made it extremely difficult to proceed, I have no idea why, but it works in that way, some kind of hypnotic effect. 

In my attempt to remove the issue quickly, I accidentally deleted a package, and along with it, critical system services. While I noticed additional packages being removed during the process, I immediately tried to reinstall the deleted packages. Unfortunately, after a reboot, the system was appears inoperable.

I hope this helps clarify my experience, and improve this function. As for me, I just removed this dangerous keybind, because I often can use different key shortcuts and don't want to catch such issue again. I added removing this keybind as one of the first steps in my post-installation guide.

ghollingworth, 2024-11-13:
We could change the key sequence to match windows Ctrl-Windows-Enter

But we can't disable Orca, it's kind of difficult when one of your board members is blind, they tend to take it personally!