raspberrypi/bookworm-feedback #317: Labwc crashes. How to enable logging?

CrendKing, 2024-11-06:
I switched from Wayfire to Labwc, and experiencing Labwc crashing when a program of mine which uses [enigo](https://github.com/enigo-rs/enigo/) to simulate keyboard inputs. This program works fine under Wayfire. It also has no problem for mouse inputs.

Before I report this to enigo, I'd like to enable debug logging of Labwc to a file, so I can give them more info. I tried to modify `/usr/bin/labwc-pi` with `exec /usr/bin/labwc -m $@ --debug > /tmp/labwc.log`, but then labwc will fail to start at all.

spl237, 2024-11-06:
Try removing the $@ from that line.

CrendKing, 2024-11-06:
Nothing changed. Just to be sure, this is the line I tried: `exec /usr/bin/labwc -m --debug > /tmp/labwc.log`

CrendKing, 2024-11-06:
Thanks. I got the log at crash:
```
Gdk-Message: 14:49:56.929: Error reading events from display: Broken pipe
(EE) failed to read Wayland events: Broken pipe
ERROR: ../src/main.c: 495: Compositor has gone away. Exiting...
Gdk-Message: 14:49:56.932: Error reading events from display: Broken pipe
Gdk-Message: 14:49:56.933: Error reading events from display: Broken pipe
Gdk-Message: 14:49:56.933: Error reading events from display: Broken pipe
warning: queue 0x7ffef0000ca0 destroyed while proxies still attached:
  zwp_primary_selection_offer_v1#4278190081 still attached
  wl_data_offer#4278190080 still attached
  zwp_primary_selection_device_v1#23 still attached
  zwp_primary_selection_device_manager_v1#19 still attached
  wl_data_device#22 still attached
  wl_seat#21 still attached
  wl_data_device_manager#20 still attached
  wl_registry#18 still attached
warning: queue 0x555598831510 destroyed while proxies still attached:
  wl_buffer#42 still attached
  wl_shm_pool#39 still attached
  xdg_wm_base#27 still attached
  wl_output#13 still attached
  zwp_pointer_constraints_v1#12 still attached
  zwp_relative_pointer_manager_v1#11 still attached
  zxdg_decoration_manager_v1#10 still attached
  xdg_activation_v1#9 still attached
  wl_seat#8 still attached
  zwp_text_input_manager_v3#7 still attached
  wl_subcompositor#6 still attached
  wl_compositor#5 still attached
  wl_shm#4 still attached
  wl_registry#2 still attached
warning: queue 0x555598861e70 destroyed while proxies still attached:
  wl_surface#26 still attached
gtk-nop failed with exit status 1.
X connection to :0 broken (explicit kill or server shutdown).
Error: "Event loop terminated with code: 1"
```

closed 2024-11-06 by CrendKing

Consolatis, 2024-11-06:
Please also report the bug at https://github.com/labwc/labwc. Even if enigo is doing something wrong labwc should not crash because of it. The log you posted in the enigo issue does not contain the relevant part for labwc itself (might need a `2>&1` to also catch stderror).

CrendKing, 2024-11-08:
That's all the log I can get. The command line was something like `labwc -m --debug 2>/tmp/labwc.log`. If I do `2>&1 >/tmp/labwc.log`, labwc would not even start (which I do not understand).