raspberrypi/bookworm-feedback #318: who command displays a large number instead of login date+time

ComputerDoktor, 2024-11-09:
who am i
pi   pts/1        3891110120997922972 (10.0.0.54)
expected output
pi   pts/1        2024-11-09 21:38 (10.0.0.54)
Raspbian GNU/Linux 12 (bookworm)
kernel version
Linux hostname 6.6.51+rpt-rpi-v6 #1 Raspbian 1:6.6.51-1+rpt3 (2024-10-08) armv6l GNU/Linux
coreutils      9.1-1        armhf        GNU core utilities

lurch, 2024-11-10:
How strange. I've just tried this on multiple Pis, and the output always looks correct, e.g.
```
$ who
pi5                   2024-11-07 19:05
pi5      tty1         2024-11-07 19:05
pi5      pts/0        2024-11-10 19:55 (192.168.0.84)
```

Is your system [up to date](https://www.raspberrypi.com/documentation/computers/os.html#manage-software-packages-with-apt) with APT? Do you continue to see this odd behaviour after rebooting your Pi?

ComputerDoktor, 2024-11-11:
The odd behaviour is only on my Raspi Zero W running 32-Bit Raspi-OS. On my Raspberry Pi 3 Model B Plus Rev 1.3 running 64-Bit-OS (kernel 6.6.51+rpt-rpi-v8 aarch64) the output is as expected.
Yes, my system is up to date with apt, and I do observe the odd behaviour after rebooting.

lurch, 2024-11-11:
Thanks for the additional info! I can confirm that I also see strange date values in the `who` output on 32-bit Raspberry Pi OS:
```
$ who
piconnect              1970-01-11 17:52
piconnect tty1         1970-01-02 10:19
piconnect pts/1        -8710239812825976095 (10.3.31.135)
```

I tried investigating it a bit, and `utmpdmp` _does_ display the correct dates:
```
$ utmpdump /run/utmp 
Utmp dump of /run/utmp
[2] [00000] [~~  ] [reboot  ] [~           ] [6.6.51+rpt-rpi-v7   ] [0.0.0.0        ] [2024-11-11T08:53:15,933134+00:00]
[7] [00662] [    ] [piconnect] [            ] [                    ] [0.0.0.0        ] [2024-11-11T08:53:26,924748+00:00]
[1] [00053] [~~  ] [runlevel] [~           ] [6.6.51+rpt-rpi-v7   ] [0.0.0.0        ] [2024-11-11T08:53:27,458497+00:00]
[7] [00789] [tty1] [piconnect] [tty1        ] [                    ] [0.0.0.0        ] [2024-11-11T08:53:28,119980+00:00]
[7] [01462] [ts/1] [piconnect] [pts/1       ] [10.3.31.135         ] [10.3.31.135    ] [2024-11-11T08:57:48,717537+00:00]
[8] [01591] [ts/2] [        ] [pts/2       ] [                    ] [10.3.31.135    ] [2024-11-11T09:30:25,722244+00:00]
```

And so I suspect that this might be because of https://debbugs.gnu.org/cgi/bugreport.cgi?bug=64937

As this is an upstream Debian bug, I suspect your only course of action is to wait for the upstream 32-bit Debian fixes to trickle down into 32-bit Raspberry Pi OS. But I'll leave this bug open for a little while in case @spl237 or @XECDesign or @pelwell have any ideas / suggestions.

ComputerDoktor, 2024-11-11:
Thanks for your information. As I understand the gnu bug report you related to, utmp will be replaced by a systemd interface in order to become y2038 compatible.

lurch, 2025-01-14:
Closing, as this is an upstream bug.

closed 2025-01-14 by lurch