raspberrypi/bookworm-feedback #321: wlopm --on fails to wake up display from screen blank after changing video mode with wlr-randr

qrp73, 2024-11-17:
I'm using RPI4 with raspi-os and labwc, with this labwc/autostart config for display blank:
```
/usr/bin/swayidle -w timeout 300 'wlopm --off \*' resume 'wlopm --on \*' &
```

It works ok, but once I change video mode with wlr-randr with something like this:
```
wlr-randr --output HDMI-A-1 --custom-mode 1280x1024@50Hz
```

it works until first display display blank and unable to wake up anymore.

When this happens, it is possible to wake up the display by changing the display mode again using wlr-randr from remote ssh. However, ```wlopm --on \*``` no longer works.

qrp73, 2024-12-03:
if it helps, here is edid of the display:
```
$ edid-decode /sys/devices/platform/gpu/drm/card1/card1-HDMI-A-1/edid
edid-decode (hex):

00 ff ff ff ff ff ff 00 4c 2d 47 02 ** ** ** **
31 11 01 03 0e 26 1e 78 2a ee 95 a3 54 4c 99 26
0f 50 54 bf ef 80 81 80 81 40 71 4f 01 01 01 01
01 01 01 01 01 01 30 2a 00 98 51 00 2a 40 30 70
13 00 78 2d 11 00 00 1e 00 00 00 fd 00 38 4b 1e
51 0e 00 0a 20 20 20 20 20 20 00 00 00 fc 00 53
79 6e 63 4d 61 73 74 65 72 0a 20 20 00 00 00 ff
00 ** ** ** ** ** ** ** ** ** ** 0a 20 20 00 05

----------------

Block 0, Base EDID:
  EDID Structure Version & Revision: 1.3
  Vendor & Product Identification:
    Manufacturer: SAM
    Model: 583
    Serial Number: ********** (0x********)
    Made in: week 49 of 2007
  Basic Display Parameters & Features:
    Analog display
    Signal Level Standard: 0.700 : 0.300 : 1.000 V p-p
    Blank level equals black level
    Sync: Separate Composite SyncOnGreen
    Maximum image size: 38 cm x 30 cm
    Gamma: 2.20
    DPMS levels: Off
    RGB color display
    First detailed timing is the preferred timing
  Color Characteristics:
    Red  : 0.6396, 0.3300
    Green: 0.2998, 0.5996
    Blue : 0.1503, 0.0595
    White: 0.3134, 0.3291
  Established Timings I & II:
    IBM     :   720x400    70.081663 Hz   9:5     31.467 kHz     28.320000 MHz
    DMT 0x04:   640x480    59.940476 Hz   4:3     31.469 kHz     25.175000 MHz
    Apple   :   640x480    66.666667 Hz   4:3     35.000 kHz     30.240000 MHz
    DMT 0x05:   640x480    72.808802 Hz   4:3     37.861 kHz     31.500000 MHz
    DMT 0x06:   640x480    75.000000 Hz   4:3     37.500 kHz     31.500000 MHz
    DMT 0x08:   800x600    56.250000 Hz   4:3     35.156 kHz     36.000000 MHz
    DMT 0x09:   800x600    60.316541 Hz   4:3     37.879 kHz     40.000000 MHz
    DMT 0x0a:   800x600    72.187572 Hz   4:3     48.077 kHz     50.000000 MHz
    DMT 0x0b:   800x600    75.000000 Hz   4:3     46.875 kHz     49.500000 MHz
    Apple   :   832x624    74.551266 Hz   4:3     49.726 kHz     57.284000 MHz
    DMT 0x10:  1024x768    60.003840 Hz   4:3     48.363 kHz     65.000000 MHz
    DMT 0x11:  1024x768    70.069359 Hz   4:3     56.476 kHz     75.000000 MHz
    DMT 0x12:  1024x768    75.028582 Hz   4:3     60.023 kHz     78.750000 MHz
    DMT 0x24:  1280x1024   75.024675 Hz   5:4     79.976 kHz    135.000000 MHz
    Apple   :  1152x870    75.061550 Hz 192:145   68.681 kHz    100.000000 MHz
  Standard Timings:
    DMT 0x23:  1280x1024   60.019740 Hz   5:4     63.981 kHz    108.000000 MHz
    DMT 0x20:  1280x960    60.000000 Hz   4:3     60.000 kHz    108.000000 MHz
    DMT 0x15:  1152x864    75.000000 Hz   4:3     67.500 kHz    108.000000 MHz
  Detailed Timing Descriptors:
    DTD 1:  1280x1024   60.019740 Hz   5:4     63.981 kHz    108.000000 MHz (376 mm x 301 mm)
                 Hfront   48 Hsync 112 Hback  248 Hpol P
                 Vfront    1 Vsync   3 Vback   38 Vpol P
    Display Range Limits:
      Monitor ranges (GTF): 56-75 Hz V, 30-81 kHz H, max dotclock 140 MHz
    Display Product Name: 'SyncMaster'
    Display Product Serial Number: '**********'
Checksum: 0x05
```

lurch, 2024-12-04:
I've just been trying to reproduce this... I used the "Raspberry Pi Configuration" program to enable "Screen Blanking", and after rebooting I edited `$HOME/.config/labwc/autostart` to change the `600` to `60` (to make this quicker to test) and rebooted again. As expected, my HDMI monitor blanks after 60 seconds and re-awakens when I move the mouse or type on the keyboard.
I then used the "Screen Configuration" program (which uses `wlr-randr` under the hood) to change the screen to different resolutions and frequencies, and in each case the monitor continues to blank after 60 seconds of "idle" and wakes up after any activity on the USB mouse or keyboard.

I see that your EDID claims to support `1280x1024@75Hz` and `1280x1024@60Hz`, but not the `1280x1024@50Hz` that you're trying to use. So is it possible that this is actually a problem with your monitor not being able to wake up to that "custom mode", rather than a problem with Raspberry Pi OS? Are you able to reproduce this problem using a screen-mode set by the "Screen Configuration" GUI program?

qrp73, 2024-12-04:
> I then used the "Screen Configuration" program (which uses `wlr-randr` under the hood) to change the screen to different resolutions and frequencies, and in each case the monitor continues to blank after 60 seconds of "idle" and wakes up after any activity on the USB mouse or keyboard.

When I use arandr (which is named as "Screen Configuration" on raspi->Preferences menu), it also works good for me. 

The issue happens when I change display mode with command from terminal: ```wlr-randr --output HDMI-A-1 --custom-mode 1280x1024@50Hz```. It successfully switch display to a new mode and display blank also triggers ok. But resume from display blank don't works. It just ignore any mouse or keyboard events and display remains in a blank mode.

But I can resume display from a blank mode from remote ssh with execute wlr-randr command to change display mode again.

qrp73, 2024-12-04:
Updated: I can reproduce the issue even for display mode listed in edid:

```
$ wlr-randr --output HDMI-A-1 --custom-mode 1280x1024@75Hz

$ pkill swayidle && swayidle -w timeout 30 'wlopm --off \*' resume 'wlopm --on \*' &
```
It going to sleep and no way to resume it.

qrp73, 2024-12-04:
And some new finding.

Once I do
```
$ wlr-randr --output HDMI-A-1 --custom-mode 1280x1024@75Hz
```
it switch to a new mode and after timeout going to sleep. Resume with keyboard or mouse don't works anymore.

But when I connect with ssh and do:
```
$ wlr-randr --output HDMI-A-1 --custom-mode 1280x1024@75Hz
failed to apply configuration

$ wlr-randr --output HDMI-A-1 --custom-mode 1280x1024@75Hz
failed to apply configuration
```
display still remains in a sleep mode

Now I tried this:
```
$ wlr-randr --output HDMI-A-1 --custom-mode 1280x1024@60Hz
```
and it resumed the display from a blank mode.

Consolatis, 2024-12-05:
> 1. **Custom Display Mode Activation**: When switching to a display mode not included in the standard list, the `wlopm --on` command is failed. Additionally, subsequent attempts to reapply the same mode is failed for an unknown reason. Furthermore, `wlopm --on` only starts functioning correctly after switching to a mode listed in the standard modes. With any custom mode, even very slightly different from standard one, `wlopm --on` fails.

This sounds like a potential labwc bug. Please open a issue at [labwc](https://github.com/labwc/labwc/issues/new/choose).
The only relevant thing here seems to be display being set via `--custom-mode` prevents unblanking via `wlopm --on`. I am not sure why this is happening.

> 2\. **Swayidle Timer Reset**: Upon resuming the display from sleep mode with a display mode change from ssh, the swayidle timer does not restart. As a result, the system does not enter sleep mode again anymore. Manual interaction, such as moving the mouse or pressing a key, is required to reactivate the timer.

This is arguably a labwc UX bug. Although using wlr-randr to unblank is theoretically possible, labwc is not expecting that. Additionally it doesn't count as user activity. This could be changed by making anything that enables a previously disabled display as a user activity. Please open an issue for this as well.

lurch, 2024-12-05:
Thank you @Consolatis ! I'll treat this as an upstream bug and close this downstream issue then.

@qrp73 Just FYI - the "Screen Configuration" GUI _used_ to be `arandr` but is now `raindrop` https://github.com/raspberrypi-ui/raindrop

closed 2024-12-05 by lurch