raspberrypi/bookworm-feedback #322: Possible keyboard map incorrect after latest upate

videosmith, 2024-11-22:
When applying all latest RaspberryPi os updates (as of 2024-11-21), several non-alphanumeric keys are mapped to incorrect glyphs.
ex: single-quote key returns single-quote glyph, however shift-press that same key for double-quote returns the @ glyph.

1. Ensure correct glyphs returned from keys pressed before update.
2. Check keyboard map (Mine is set to):
     Model: Generic 105-key PC
     Layout: English (US)
     Variant: English (US)
4.  Update os.
5.  Check glyphs again, should be correct.
6.  Reboot.
7.  Check glyphs again, mine are incorrectly mapped.
8.  Check keyboard map (mine has not changed).
9.  Change keyboard map to another selection (I chose Generic 101-key PC), apply change.
10.  Change keyboard map back to original, apply change.
11.  Glyphs now properly assigned.

theofficialgman, 2024-11-27:
I can confirm the issue @videosmith . Hopefully this gets fixed ASAP.
This is an issue only on labwc, the keymap is correct on wayfire and x11 (yes I just tested).

lurch, 2024-11-27:
I've just asked @spl237 about this and he says that the problem **has** been fixed, but unfortunately the fix itself is in one of the updates that you're also installing at the same time as all the other updates! And he says that you've already identified the "workaround" for this (i.e. after the switch to labwc, change your locale to something else, and then change it back to the desired locale).

So I'll close this, but I'll re-open if you can still reproduce this when starting with a clean install of the latest Raspberry Pi OS image.

theofficialgman, 2024-11-27:
@lurch @spl237 I have changed the locale using raspberry pi configuration to something else, changed it back to the desired locale using raspberry pi configuration, and then rebooted and the issue still occurs.

theofficialgman, 2024-11-27:
I have also tried changed the locale using raspberry pi configuration to something else, rebooting, changed it back to the desired locale using raspberry pi configuration, and then rebooting again and the issue still occurs.

Please re-open as there is no working documented workaround for the issue (as I have shown) and there is no fix in a package update.

lurch, 2024-11-27:
I've re-opened, but could you please provide more detailed steps about **exactly** what you're doing, what you expect to happen, and what actually happens ?

theofficialgman, 2024-11-27:
> I've re-opened, but could you please provide more detailed steps about **exactly** what you're doing, what you expect to happen, and what actually happens ?

You stated here that a workaround to fix the issue was to:

> change your locale to something else, and then change it back to the desired locale

So I attempting to perform your supplied workaround using 'Raspberry Pi Configuration'. The steps being:
1. Verify the current compositor is labwc
2. Open 'Raspberry Pi Configuration' -> Localisation (btw this is misspelled for US english, it is spelled Localization) -> set any non-current localization
3. Reboot the system when prompted on exiting 'Raspberry Pi Configuration' with OK
4. Open 'Raspberry Pi Configuration' -> Localisation -> set the desired localization that matches the keyboard used
5. Reboot the system when prompted on exiting 'Raspberry Pi Configuration' with OK
6. Observe the issue still occurs.

lurch, 2024-11-27:
Oh, sorry @theofficialgman I've just realised that I made a silly mistake in my previous advice - I should have said "change your keyboard to a different layout, and then change the keyboard back to the desired layout". Does that fix the problem you're seeing?

theofficialgman, 2024-11-27:
Can you specify where I'm supposed to go to change the keyboard layout? That's not clear to me.

lurch, 2024-11-27:
> Can you specify where I'm supposed to go to change the keyboard layout?

1. Click on the "Raspberry icon" at the top-left of the screen
2. Go down to the `Preferences` menu
3. Select `Raspberry Pi Configuration`
4. When it has loaded, click the `Localisation` tab at the top
5. Click on the `Set Keyboard...` button
6. Choose a different keyboard layout from the `Layout:` dropdown
7. Click `OK` to close the `Keyboard` dialog
8. Click `OK` to close the `Raspberry Pi Configuration` program

spl237, 2024-11-28:
Just tested here - took July 24 (pre-labwc) image and set it to French in the wizard, giving an AZERTY layout. Applied all updates, rebooted, at the prompt chose to switch to labwc. When the system came back up in labwc after the reboot, the AZERTY layout was still in effect.

Note - the keyboard layout will only be copied across when switching to labwc from the GUI prompt. If you switch to labwc from raspi-config, you will need to copy the keyboard layout manually - either switch to a different layout and back again, or from a terminal do "sudo raspi-config nonint update_labwc_keyboard"

Working as intended, so closing.

closed 2024-11-28 by spl237

spl237, 2024-11-28:
Ah - it occurs to me that what people are talking about here is not, as I had originally thought, the change in keymap caused by the initial switch to labwc, but a possible loss of the labwc keymap setting caused by more recent changes to the default labwc environment files. That could indeed potentially lose keymap customisations - I'll see if anything can be done about that.

spl237, 2024-11-28:
OK, this should now be fixed in the latest update - when updating with the latest version of ui-mods, which should be going through the build system now and available in a few hours' time, labwc keyboard settings will be preserved. Closing on that basis - please reopen if the problem is not fixed.