raspberrypi/bookworm-feedback #324: web form time field showing as US format (AM/PM) when should be 24h format

ibowditch, 2024-11-28:
I have a web app, running on an AWS server, that uses bootstrap forms in a browser to collect data. One field is for a time, and must be shown in 24h format.
The same form has been working on PIs for several years, without problems, and always shows in 24h format on chromium, chrome, firefox, etc. on PI 4 using bullseye, as well as browsers on on Windows, iOS, Android etc.
However, using bookworm, this time field shows as AM/PM in both chromium and firefox. 
The original locale, set using Raspberry Imager, was en_AU, and I tried changing to en_GB, and en_DK, but the field is always shown in US format - AM/PM.
Further investigation shows that the locale inside chromium always seems to be en_US, regardless of the system locale setting - I confirmed this by typing **console.log(Intl.DateTimeFormat().resolvedOptions().locale);** in the browser console.
The underlying html in the form is fairly straight forward (see below), and works in every other environment I've seen, except in bookworm using chromium or firefox. 

<div class="form-group offset-sm-1 required">Start time: <input type="time" name="eventStart" value="18:01" step="60" class="form-control" required="" id="id_eventStart"></div>

lurch, 2024-11-28:
> Further investigation shows that the locale inside chromium always seems to be en_US, regardless of the system locale setting

Any ideas @XECDesign ?
(also pinging @cillian64 as the OP claims that a similar problem affects Firefox too)

cillian64, 2024-11-28:
I've just tested on bookworm/labwc by putting just that single div into an HTML file.  In both firefox (133.0-1+rpt7) and chromium (1:130.0.6723.116-1~deb12u1+rpt1) the time displays in 24hr format (18:01).  Running the dev-console command in Firefox gives me `en-GB`.  This also happens immediately after I wipe my Firefox profile, so it's not because I have previously set the locale in Firefox somehow.

So either Firefox/chromium are caching the locale they initially saw, something about changing locale is not working (this feels most likely to me), there's something else wonky on your system, or there's something else to do with the website/context making a difference.

Could you try writing a new image with the locale initially set to en_GB and see what happens?  And could you test with an HTML file containing only that single div?

ibowditch, 2024-11-28:
Thanks very much for your prompt response. I boiled this down to the simplest case I could, and the problem is still there.

Here's what I did:

a. Prepared a new PI image downloaded from https://downloads.raspberrypi.com/raspios_arm64/images/raspios_arm64-2024-11-19/2024-11-19-raspios-bookworm-arm64.img.xz
b. Wrote to a fresh SD card using  Pi Imager 1.8.5 as a custom file, with no saved environment
c. Went through the install procedure on the PI 4, leaving everything as default, including locale=en_GB, country UK, though I did connect to my Australian wifi
d. Prepared a very simple html file, with just a form containing a single time field (see attached zip file)
e. Opened the html file on the PI4 with chromium

Observations - (see attached screenshot jpg):
1. The time field is shown as AM/PM, not 24h clock as it should be for locale=en_GB. It displays as 24h clock on all other machines I have tried.
2. The locale and locale -k LC_TIME settings are all set to en_GB as expected, though LANGUAGE and LC_ALL are not set
3. locale -k LC_TIME shows t_fmt="%T", which should show 24h clock for times
4. chromium console.log(Intl.DateTimeFormat().resolvedOptions().locale); shows en_US, not en_UK. There can't be any caching issues here, since this is a fresh install, and the locale wasn't changed from original defaults (en_GB).
5. The form uses bootstrap4 and jquery. I've cut this down as far as i can, and this combination has worked for years on other platforms. 

[image]
[simpleform.zip](https://github.com/user-attachments/files/17952453/simpleform.zip)

ibowditch, 2024-11-28:
PS: If it helps, I created a version of the simple form using bootstrap5 (see attached), but still get the same problem.

[simpleform5.zip](https://github.com/user-attachments/files/17952646/simpleform5.zip)

lurch, 2024-11-29:
@ibowditch Just for the sake of completeness, do you see the same behaviour in Firefox on the Pi too?

ibowditch, 2024-11-29:
Sorry I forgot to check that in the original. 

Firefox is showing the time as 24h (correct), and internal locale as en_GB (correct), but it is not rendering the time control correctly - it should have a clock and time picker on the right of the box (which chromium does get right).

[image]

cillian64, 2024-12-03:
There's a few separate issues here:

1. datetime picker not appearing in Firefox.  I can reproduce that with stock upstream Firefox on my x86_64 debian laptop, so if it is a bug (not clear to me whether it's a bug or just differing browser behaviour) it's an upstream Firefox bug and not rpi specific.
2. AM/PM vs 24-hr time format issue in Firefox.  This appears correctly for both of us with Firefox on a clean image, so if there is a bug it's something to do with changing locale or something bad in your original firefox profile
3. AM/PM vs 24-hr time format in chromium.  I've worked out that the problem here is we changed how we're packaging chromium so the `chromium-l10n` package is no longer installed by default (confusingly older systems which have been upgraded will have that package but newer installs won't have it, that's why I couldn't reproduce the issue at first).  Installing `chromium-l10n` fixes the problem for me, please could you confirm if it does for you too?  If so I'll change it from a `Suggests` to a `Recommends` so it gets pulled in by default.

ibowditch, 2024-12-04:
Thanks very much again for your prompt response. 
I installed chromium-l10n as you suggested, and indeed, both sample forms work just fine now, as does the form with the time fields in my webapp.
So that looks like the fix for this particular problem.
Much appreciated!

cillian64, 2024-12-04:
Glad to hear it, thanks for the report and the help debugging.

@XECDesign please could we make `chromium-l10n` a `Recommends` instead of a `Suggests` from `chromium` and add it to the base image (if the extra size is permissible)?  Without it chromium fails to do en-GB localisation.