raspberrypi/bookworm-feedback #327: strange labwc freeze

qrp73, 2024-11-29:
I catch strange labwc freeze with no way to continue work. The mouse is just frozen and cannot return from that state.

I think it may be due to out of memory event, because it happens when I open large file in audacity. But there is no such errors in the log. I was able to connect with ssh, it shows that there is 440 MB free and system working. No errors in the dmesg. But labwc desktop still remains in frozen state.

I tried to reboot from ssh and it take a long time, here is the log:
```
Nov 29 17:21:01 raspi rtkit-daemon[988]: Supervising 7 threads of 4 processes of 1 users.
Nov 29 17:21:01 raspi rtkit-daemon[988]: Supervising 7 threads of 4 processes of 1 users.
Nov 29 17:21:43 raspi rtkit-daemon[988]: Supervising 7 threads of 4 processes of 1 users.
Nov 29 17:21:43 raspi rtkit-daemon[988]: Supervising 7 threads of 4 processes of 1 users.
Nov 29 17:23:11 raspi rtkit-daemon[988]: Supervising 7 threads of 4 processes of 1 users.
Nov 29 17:23:11 raspi rtkit-daemon[988]: Supervising 7 threads of 4 processes of 1 users.
Nov 29 17:23:28 raspi rtkit-daemon[988]: Supervising 7 threads of 4 processes of 1 users.
Nov 29 17:23:28 raspi rtkit-daemon[988]: Supervising 7 threads of 4 processes of 1 users.
Nov 29 17:25:23 raspi rtkit-daemon[988]: Supervising 7 threads of 4 processes of 1 users.
Nov 29 17:25:23 raspi rtkit-daemon[988]: Supervising 7 threads of 4 processes of 1 users.
Nov 29 17:27:43 raspi pipewire[966]: pw.node: (alsa_output.usb-Synaptics_CX31993_384Khz_HIFI_AUDIO-00.analog-stereo-71) graph xrun not-triggered (0 suppressed)
Nov 29 17:27:43 raspi pipewire[966]: pw.node: (alsa_output.usb-Synaptics_CX31993_384Khz_HIFI_AUDIO-00.analog-stereo-71) xrun state:0x7f860fe008 pending:1/1 s:122335564683083 a:122335564701972 f:122335564773452 waiting:18889 process:71480 status:triggered
Nov 29 17:27:45 raspi pipewire[966]: pw.node: (alsa_output.usb-Synaptics_CX31993_384Khz_HIFI_AUDIO-00.analog-stereo-71) graph xrun not-triggered (2 suppressed)
Nov 29 17:27:45 raspi pipewire[966]: pw.node: (alsa_output.usb-Synaptics_CX31993_384Khz_HIFI_AUDIO-00.analog-stereo-71) xrun state:0x7f860fe008 pending:1/1 s:122964746525488 a:122964746534951 f:122964746566598 waiting:9463 process:31647 status:triggered
Nov 29 17:28:06 raspi pipewire[966]: pw.node: (alsa_output.usb-Synaptics_CX31993_384Khz_HIFI_AUDIO-00.analog-stereo-71) graph xrun not-triggered (2 suppressed)
Nov 29 17:28:06 raspi pipewire[966]: pw.node: (alsa_output.usb-Synaptics_CX31993_384Khz_HIFI_AUDIO-00.analog-stereo-71) xrun state:0x7f860fe008 pending:1/1 s:122971180248788 a:122971180270010 f:122971180295528 waiting:21222 process:25518 status:triggered
Nov 29 17:35:32 raspi sshd[198801]: Accepted password for pi from 192.168.***.*** port 39928 ssh2
Nov 29 17:35:32 raspi sshd[198801]: pam_unix(sshd:session): session opened for user pi(uid=1000) by (uid=0)
Nov 29 17:35:32 raspi systemd-logind[787]: New session 40 of user pi.
Nov 29 17:35:32 raspi systemd[1]: Started session-40.scope - Session 40 of User pi.
Nov 29 17:35:32 raspi sshd[198801]: pam_env(sshd:session): deprecated reading of user environment enabled
Nov 29 17:40:25 raspi systemd[1]: pcscd.service: Deactivated successfully.
Nov 29 17:40:55 raspi sudo[199289]:       pi : TTY=pts/4 ; PWD=/home/pi ; USER=root ; COMMAND=/usr/sbin/reboot
Nov 29 17:40:55 raspi sudo[199289]: pam_unix(sudo:session): session opened for user root(uid=0) by pi(uid=1000)
Nov 29 17:40:55 raspi systemd-logind[787]: The system will reboot now!
Nov 29 17:40:55 raspi systemd-logind[787]: System is rebooting.
Nov 29 17:40:55 raspi sudo[199289]: pam_unix(sudo:session): session closed for user root
Nov 29 17:40:55 raspi systemd[1]: Stopping session-1.scope - Session 1 of User pi...
Nov 29 17:40:55 raspi systemd[1]: Stopping session-3.scope - Session 3 of User pi...
Nov 29 17:40:55 raspi systemd[1]: Stopping session-40.scope - Session 40 of User pi...
Nov 29 17:40:55 raspi systemd[1]: Removed slice system-modprobe.slice - Slice /system/modprobe.
Nov 29 17:40:55 raspi sshd[198801]: pam_unix(sshd:session): session closed for user pi
Nov 29 17:40:55 raspi systemd[1]: Stopped target graphical.target - Graphical Interface.
Nov 29 17:40:55 raspi systemd[1]: Stopped target multi-user.target - Multi-User System.
Nov 29 17:40:55 raspi systemd[1]: Stopped target getty.target - Login Prompts.
Nov 29 17:40:55 raspi systemd[1]: Stopped target rpc_pipefs.target.
Nov 29 17:40:55 raspi systemd[1]: Stopped target sound.target - Sound Card.
Nov 29 17:40:55 raspi systemd[1]: Stopped target timers.target - Timer Units.
Nov 29 17:40:55 raspi systemd[1]: apt-daily-upgrade.timer: Deactivated successfully.
Nov 29 17:40:55 raspi systemd[1]: Stopped apt-daily-upgrade.timer - Daily apt upgrade and clean activities.
Nov 29 17:40:55 raspi systemd[1]: apt-daily.timer: Deactivated successfully.
Nov 29 17:40:55 raspi systemd[1]: Stopped apt-daily.timer - Daily apt download activities.
Nov 29 17:40:55 raspi systemd[1]: dpkg-db-backup.timer: Deactivated successfully.
Nov 29 17:40:55 raspi systemd[1]: Stopped dpkg-db-backup.timer - Daily dpkg database backup timer.
....
Nov 29 17:40:56 raspi systemd[1]: zramswap.service: Deactivated successfully.
Nov 29 17:40:56 raspi systemd[1]: Stopped zramswap.service - Linux zramswap setup.
Nov 29 17:42:25 raspi systemd[1]: session-1.scope: Stopping timed out. Killing.
Nov 29 17:42:25 raspi systemd[1]: session-1.scope: Killing process 938 (lightdm) with signal SIGKILL.
Nov 29 17:42:25 raspi systemd[1]: session-1.scope: Killing process 970 (labwc) with signal SIGKILL.
Nov 29 17:42:25 raspi systemd[1]: session-1.scope: Killing process 1230 (pcmanfm) with signal SIGKILL.
Nov 29 17:42:25 raspi systemd[1]: session-1.scope: Killing process 946 (gmain) with signal SIGKILL.
Nov 29 17:42:25 raspi systemd[1]: session-1.scope: Killing process 1154 (labwc:disk$0) with signal SIGKILL.
Nov 29 17:42:25 raspi systemd[1]: session-1.scope: Killing process 1253 (gdbus) with signal SIGKILL.
Nov 29 17:42:25 raspi systemd[1]: lightdm.service: State 'stop-sigterm' timed out. Killing.
Nov 29 17:42:25 raspi systemd[1]: lightdm.service: Killing process 916 (lightdm) with signal SIGKILL.
Nov 29 17:42:25 raspi systemd[1]: lightdm.service: Killing process 923 (gdbus) with signal SIGKILL.
Nov 29 17:42:25 raspi systemd[1]: lightdm.service: Main process exited, code=killed, status=9/KILL
Nov 29 17:42:25 raspi systemd[1]: lightdm.service: Failed with result 'timeout'.
Nov 29 17:42:25 raspi systemd[1]: Stopped lightdm.service - Light Display Manager.
Nov 29 17:42:25 raspi systemd[1]: lightdm.service: Triggering OnFailure= dependencies.
Nov 29 17:42:25 raspi systemd[1]: Requested transaction contradicts existing jobs: Transaction for plymouth-quit.service/start is destructive (local-fs-pre.target has 'stop' job queued, but 'start' is included in transaction).
Nov 29 17:42:25 raspi systemd[1]: lightdm.service: Failed to enqueue OnFailure= job, ignoring: Transaction for plymouth-quit.service/start is destructive (local-fs-pre.target has 'stop' job queued, but 'start' is included in transaction).
Nov 29 17:42:25 raspi xdg-desktop-por[1310]: Error reading events from display: Broken pipe
[... 7 lines cut ...]
Nov 29 17:42:25 raspi systemd[1]: Stopping systemd-logind.service - User Login Management...
Nov 29 17:42:25 raspi systemd[1]: Stopping user@1000.service - User Manager for UID 1000...
Nov 29 17:42:25 raspi systemd[1]: Started plymouth-reboot.service - Show Plymouth Reboot Screen.
Nov 29 17:42:25 raspi systemd[1]: plymouth-switch-root-initramfs.service - Tell Plymouth To Jump To initramfs was skipped because of an unmet condition check (ConditionPathExists=/run/initramfs/bin/sh).
Nov 29 17:42:25 raspi systemd[950]: Activating special unit exit.target...
Nov 29 17:42:25 raspi systemd[950]: Removed slice app-org.gnome.Terminal.slice - Slice /app/org.gnome.Terminal.
Nov 29 17:42:25 raspi org.freedesktop.secrets[1342]: discover_other_daemon: 1
Nov 29 17:42:25 raspi systemd[950]: app-org.gnome.Terminal.slice: Consumed 7min 20.825s CPU time.
Nov 29 17:42:25 raspi systemd[950]: Stopped target default.target - Main User Target.
Nov 29 17:42:25 raspi systemd[950]: Stopping dbus.service - D-Bus User Message Bus...
Nov 29 17:42:25 raspi systemd[950]: Stopping dconf.service - User preferences database...
...
Nov 29 17:42:26 raspi systemd[1]: Finished systemd-reboot.service - System Reboot.
Nov 29 17:42:26 raspi systemd[1]: Reached target reboot.target - System Reboot.
Nov 29 17:42:26 raspi systemd[1]: Shutting down.
[... cut ...]
Nov 29 17:42:26 raspi systemd-shutdown[1]: Using hardware watchdog 'Broadcom BCM2835 Watchdog timer', version 0, device /dev/watchdog0
Nov 29 17:42:27 raspi systemd-shutdown[1]: Watchdog running with a hardware timeout of 10min.
Nov 29 17:42:27 raspi systemd-shutdown[1]: Syncing filesystems and block devices.
Nov 29 17:42:27 raspi systemd-shutdown[1]: Sending SIGTERM to remaining processes...
Nov 29 17:42:27 raspi systemd-journald[304]: Received SIGTERM from PID 1 (systemd-shutdow).
Nov 29 17:42:27 raspi systemd-journald[304]: Journal stopped
```

qrp73, 2024-12-03:
it happens just randomly. But I suspect it may be related to [this issue](https://github.com/raspberrypi/bookworm-feedback/issues/328), because it leads to the same system lock up with no error in the log.

lurch, 2024-12-03:
Based on comments in that other issue, _perhaps_ when you've got a particularly large audio file, Audacity tries to render it on a single huuuuge texture which is too large for the hardware to handle? So I'll close this (for now) as a duplicate of #328

closed 2024-12-03 by lurch

qrp73, 2024-12-03:
looks like true, because it happened for a large audio file