raspberrypi/bookworm-feedback #328: Open jpeg picture leads to complete labwc desktop crash and freeze

qrp73, 2024-11-30:
Here is test picture in jpeg format created with Krita. This is just an empty image, but large.

Attempt to open it with ImageMagic leads to a complete labwc crash and freeze. 

[image]

Just save the file, right click and select open in ImageMagick.

First, it leads to a short freezes, CPU load indicator on pi-panel going to 100% and then shows "nan" value, and then mouse stopped to respond, display stops to update and no way to exit from this zombie mode. The only way to fix it is just to reboot the system.

I can connect with ssh and it shows that the system works, CPU load is about 6%, free memory is available. And there is no errors in the system log.

So, I'm confused what is going on. Could you please fix it?

lurch, 2024-12-02:
a) Just out of curiosity, why do you need such a **huge** (13980x4645) solid-red picture? :wink: 
b) It opens fine in Eye of MATE (the default image-viewer in Raspberry Pi OS) so ImageMagick must be doing something odd
c) ...but even so, ImageMagick shouldn't be able to cause labwc to totally lock up (I was able to reproduce the problem you described)
d) After waiting for a while, I see this in the `dmesg` output:
```
[  846.822290] INFO: task kworker/u9:2:55 blocked for more than 120 seconds.
[  846.822328]       Tainted: G         C         6.6.62+rpt-rpi-v8 #1 Debian 1:6.6.62-1+rpt1
[  846.822339] "echo 0 > /proc/sys/kernel/hung_task_timeout_secs" disables this message.
[  846.822345] task:kworker/u9:2    state:D stack:0     pid:55    ppid:2      flags:0x00000008
[  846.822375] Workqueue: events_unbound commit_work [drm_kms_helper]
[  846.822514] Call trace:
[  846.822521]  __switch_to+0xe0/0x148
[  846.822545]  __schedule+0x37c/0xd60
[  846.822558]  schedule+0x64/0x108
[  846.822570]  schedule_timeout+0x1a0/0x1b8
[  846.822588]  dma_fence_default_wait+0x1c0/0x1f0
[  846.822606]  dma_fence_wait_timeout+0x1b4/0x220
[  846.822621]  drm_atomic_helper_wait_for_fences+0x180/0x248 [drm_kms_helper]
[  846.822721]  commit_tail+0x40/0x1a0 [drm_kms_helper]
[  846.822817]  commit_work+0x1c/0x30 [drm_kms_helper]
[  846.822912]  process_one_work+0x148/0x3b8
[  846.822927]  worker_thread+0x32c/0x450
[  846.822938]  kthread+0x11c/0x128
[  846.822959]  ret_from_fork+0x10/0x20
```
e) GIMP is also able to open this file okay, so I saved it out as a `.png` version, and Eye of MATE is able to open the PNG okay, but trying to open it with ImageMagick causes the desktop to lock up again
f) If I run `sudo reboot` on the locked-up Pi over SSH, my SSH connection quits (and I'm unable to reconnect, so the SSH server has obviously been shut down), and the locked-up Pi appears to do nothing, but _eventually_ reboots!

spl237, 2024-12-02:
I'd suggest raising this at https://github.com/labwc/labwc/issues

6by9, 2024-12-02:
I'm going to hazard a guess that ImageMagick tries to load the image directly into a texture, and V3D has a maximum texture size that it supports which I think is lower than that (8k x 8k comes to mind).

I would have hoped that the import failed, and something threw an error, but being blocked in `drm_atomic_helper_wait_for_fences` implies the compositor has submitted the output of the composition to DRM to display, and then V3D has never completed, hence blocked waiting on the fence.

cillian64, 2024-12-02:
Could you stop labwc with `sudo systemctl stop lightdm` then run it with debug logging `labwc-pi &>labwc_log` then attach the debug log here?

Agree with Dave that it's probably imagemagick submitting a silly surface which 3d then chokes on.  labwc will probably say this is a wlroots issue, and I suspect wlroots will say that capping texture size is a hardware-specific hack.  But I'll ask them anyway.

qrp73, 2024-12-03:
> a) Just out of curiosity, why do you need such a **huge** (13980x4645) solid-red picture? 😉 

I don't need exactly this solid-red picture, I catch this issue when tried to open a usual hi-res photo taken with smartphone. Then I tried to reproduce it with empty image and get the same issue.

> c) ...but even so, ImageMagick shouldn't be able to cause labwc to totally lock up (I was able to reproduce the problem you described) 

yes, it should not lead to a total lock up. And I also catch similar issues with labwc lock up when worked with Audacity, probably it is the same.

So, it looks that it may happens with different apps.

> f) If I run `sudo reboot` on the locked-up Pi over SSH, my SSH connection quits (and I'm unable to reconnect, so the SSH server has obviously been shut down), and the locked-up Pi appears to do nothing, but _eventually_ reboots!

it takes for about 2-3 minutes for reboot, something is going wrong...

lurch, 2024-12-03:
@cillian64 Logs from both a Pi4 and a Pi5. In each case I was running `labwc-pi -d` (version 0.7.4) and I successfully opened the huge jpeg in Eye of MATE, then closed it, and then tried opening it in ImageMagick (where the Pi locked up).
[labwc_pi4_jpeg.log](https://github.com/user-attachments/files/17993849/labwc_pi4_jpeg.log)
[labwc_pi5_jpeg.log](https://github.com/user-attachments/files/17993850/labwc_pi5_jpeg.log)