raspberrypi/bookworm-feedback #329: labwc screen does not resume from "blanking"

Bylon2, 2024-12-02:
```
$ uname -a
Linux raspberrypi 6.6.62+rpt-rpi-v8 #1 SMP PREEMPT Debian 1:6.6.62-1+rpt1 (2024-11-25) aarch64 GNU/Linux

```
Using **Raspberry Pi 4 - 4GB** (first model)

Since the last move to labwc I have a strange issue.
If the system is freshly booted, the display (Panasonic TV connected to HDMI-A-1) works perfectly.

When I switch off the TV, the next morning when I switch it on again, the screen stays blank (more accurately black!), as if it had no signal on the HDMI input.
Moving the mouse of typing on the keyboard has absolutely no effect, screen stays black.
It is doing so although I have set the display **no to go to blanking** in the raspi-config.

If I connect via SSH or VNC it works fine, but the TV screen persist to stay black.

I have seen some posts using wlopm, and that is even stranger, here is what I get:

```
$ wlopm
HDMI-A-1 off
```
I get that when the TV is working absolutely fine after the first boot.

If I try to switch on the perfectly working HDMI-A-1, this is what I get
```
$ wlopm --on \*
ERROR: Setting power mode for output 'HDMI-A-1' failed.
```
This behaviour is the same be it with Wayfire or with labwc, so maybe I am misinterpreting what it is supposed to say/do!

So labwc means that I have to reboot my Pi each time I switch on my TV, when it has been off for enough time: switch off for 10 minutes is OK. Not convenient!

I have currently _reverted back to wayfire_, and there is absolutely no _not-unblanking_ issue. The display on my TV works fine be it when the Pi is freshly booted, or after an uptime of several weeks.
This should probably also dismiss a "cable issue" since it is the exact same Pi and cable in both cases.

Is there a known solution or what can I do to help debug this issue?

qrp73, 2024-12-03:
I have similar issue, when I switch display mode to some not listed in display edid, it works ok until display blank. Display blank triggers and successfully switch display blank. But when I pressing key to resume, it don't resume.

I'm using labwc with this line in ```~/.config/labwc/autostart```:
```
/usr/bin/swayidle -w timeout 300 'wlopm --off \*' resume 'wlopm --on \*' &
```

With default display mode it works ok and successfully enter and exit from display blank mode. But after display mode switch it fails for some unknown reason.

See details here: https://github.com/raspberrypi/bookworm-feedback/issues/321

Bylon2, 2024-12-03:
So if that helps:

```
$ edid-decode < /sys/devices/platform/gpu/drm/card1/card1-HDMI-A-1/edid
edid-decode (hex):

00 ff ff ff ff ff ff 00 34 a9 96 a2 02 01 01 01
00 21 01 03 80 80 48 78 0a da ff a3 58 4a a2 29
17 49 4b 20 00 00 31 40 01 01 01 01 01 01 01 01
01 01 01 01 01 01 08 e8 00 30 f2 70 5a 80 b0 58
8a 00 00 d0 52 00 00 1e 6f c2 00 a0 a0 a0 55 50
30 20 35 00 00 d0 52 00 00 1a 00 00 00 fc 00 50
61 6e 61 73 6f 6e 69 63 2d 54 56 0a 00 00 00 fd
00 17 78 0f ff 77 00 0a 20 20 20 20 20 20 01 d0

02 03 63 f0 55 61 60 10 1f 05 14 20 21 22 04 13
03 12 76 75 5d 5e 5f 3f 40 01 2c 0d 07 01 15 07
50 57 07 01 67 04 03 83 0f 00 00 6d 03 0c 00 10
00 38 44 2f 08 60 01 02 03 68 d8 5d c4 01 78 80
5b 02 e2 00 cb e3 05 ff 01 e3 0f 03 60 e3 06 0f
01 e5 01 8b 84 90 01 eb 01 46 d0 00 4d 03 9a 7e
5e 6d 95 00 00 00 00 00 00 00 00 00 00 00 00 00
00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 9b

----------------

Block 0, Base EDID:
  EDID Structure Version & Revision: 1.3
  Vendor & Product Identification:
    Manufacturer: MEI
    Model: 41622
    Serial Number: 16843010 (0x01010102)
    Made in: 2023
  Basic Display Parameters & Features:
    Digital display
    Maximum image size: 128 cm x 72 cm
    Gamma: 2.20
    RGB color display
    First detailed timing is the preferred timing
  Color Characteristics:
    Red  : 0.6396, 0.3447
    Green: 0.2910, 0.6347
    Blue : 0.1630, 0.0927
    White: 0.2880, 0.2958
  Established Timings I & II:
    DMT 0x04:   640x480    59.940476 Hz   4:3     31.469 kHz     25.175000 MHz
  Standard Timings:
    DMT 0x04:   640x480    59.940476 Hz   4:3     31.469 kHz     25.175000 MHz
  Detailed Timing Descriptors:
    DTD 1:  3840x2160   60.000000 Hz  16:9    135.000 kHz    594.000000 MHz (1280 mm x 720 mm)
                 Hfront  176 Hsync  88 Hback  296 Hpol P
                 Vfront    8 Vsync  10 Vback   72 Vpol P
    DTD 2:  2560x1440  119.997589 Hz  16:9    182.996 kHz    497.750000 MHz (1280 mm x 720 mm)
                 Hfront   48 Hsync  32 Hback   80 Hpol P
                 Vfront    3 Vsync   5 Vback   77 Vpol N
    Display Product Name: 'Panasonic-TV'
    Display Range Limits:
      Monitor ranges (GTF): 23-120 Hz V, 15-255 kHz H, max dotclock 1190 MHz
  Extension blocks: 1
Checksum: 0xd0

----------------

Block 1, CTA-861 Extension Block:
  Revision: 3
  Underscans IT Video Formats by default
  Basic audio support
  Supports YCbCr 4:4:4
  Supports YCbCr 4:2:2
  Native detailed modes: 0
[... 94 lines cut ...]
      Traditional gamma - SDR luminance range
      Traditional gamma - HDR luminance range
      SMPTE ST2084
      Hybrid Log-Gamma
    Supported static metadata descriptors:
      Static metadata type 1
  Vendor-Specific Video Data Block (HDR10+), OUI 90-84-8B:
    Application Version: 1
    Full Frame Peak Luminance Index: 0
    Peak Luminance Index: 0
  Vendor-Specific Video Data Block (Dolby), OUI 00-D0-46:
    Version: 2 (12 bytes)
    Supports YUV422 12 bit
    DM Version: 5.x
    Backlt Min Luma: 100 cd/m^2
    Interface: Standard + Low-Latency
    Supports 10b 12b 444: Not supported
    Target Min PQ v2: 0 (0.00000000 cd/m^2)
    Target Max PQ v2: 3290 (1604 cd/m^2)
    Unique Rx, Ry: 0.67578125, 0.32031250
    Unique Gx, Gy: 0.24609375, 0.68359375
    Unique Bx, By: 0.14453125, 0.05078125
Checksum: 0x9b  Unused space in Extension Block: 28 bytes

```

lurch, 2024-12-04:
> Is there a known solution or what can I do to help debug this issue?

When your Pi is "working" on your TV (i.e. soon after bootup), if you unplug and then replug the HDMI cable, does the TV come back on again? Does `wlopm` still say `HDMI-A-1 off` ?
When the Pi / TV is "not working" (i.e. when you switch on the TV in the morning), if you unplug and then replug the HDMI cable, does the TV wake up?
Do you have any other HDMI devices you can test with, to see if the Pi _is_ actually outputting an HDMI signal, and it's simply that the TV that isn't "seeing" it?

Bylon2, 2024-12-04:
Thanks for your time to reply.

So now I understand even less why if works with wayfire.

**Part 1)**
TV is on

- Switched to labwc (with raspi-config)
- Reboot
_Splash and all shows on the screen
Image is displayed correctly after boot_

```
$ ps -ef | grep -i 'wayfire\|labwc'
pi        973     926  1 21:47 ?        00:00:02 /usr/bin/labwc -m
pi       1052     973  0 21:47 ?        00:00:00 /usr/bin/ssh-agent /usr/bin/labwc-pi
pi       2158    2146  0 21:50 pts/0    00:00:00 grep --color=auto -i wayfire\|labwc

pi@raspberrypi:~ $ wlopm
HDMI-A-1 off
```

- Unplug cable (TV side)
_Screen goes dark (as expected!)_

(via VNC)
```
pi@raspberrypi:~ $ wlopm
HDMI-A-1 off
```

- Plug cable (TV side)
**Screen stays black** whatever mouse gesture

(via VNC)
```
pi@raspberrypi:~ $ wlopm
HDMI-A-1 off
```

- Revert to **wayfire** (with raspi-config via SSH)
- Reboot
_TV displays the image again of the reboot
Image is displayed correctly after boot_

```
$ ps -ef | grep -i 'wayfire\|labwc'
pi       1041     996  3 21:59 ?        00:00:02 /usr/bin/wayfire
pi       1125    1041  0 21:59 ?        00:00:00 /usr/bin/ssh-agent /usr/bin/wayfire-pi
pi       2185    2173  0 22:00 pts/0    00:00:00 grep --color=auto -i wayfire\|labwc

pi@raspberrypi:~ $ wlopm
HDMI-A-1 off

```

- Unplug cable (TV side)
_Screen goes dark (as expected!)_

(via VNC)
```
pi@raspberrypi:~ $ wlopm
HDMI-A-1 off
```

- Plug cable (TV side)
**Screen stays black**

(via VNC)
```
pi@raspberrypi:~ $ wlopm
HDMI-A-1 off
```

----------------
The only visible difference when plugging the cable is that with labwc the TV sees absolutely nothing and displays only "HDMI1"
With wayfire, when plugged in again, it says "recording device 1" but still no image.
Note that during the boot process the TV says "Videocore", and if using Kodi, it stays unsurprisingly "Kodi"! The TV also says this is 3840x2160@60Hz with 16/9.

This made me come with sort of a **cumbersome workaround**, after the cable being plugged in again, if open a VNC session and start Kodi, there is a display again (of Kodi), and when I exit Kodi the desktop shows again.
True that Kodi does its own display and does not rely on X or Wayland.

So the only difference I see now is that I was using screen blanking on Wayfire. When I swtich on the TV, being used to "computer order of things", I use to switch on the TV first, then move the mouse to unblank.

What I have done is re-enabled blanking under labwc to see if it changes something.

The TV also has a gazillion settings you can change.
Currently the Pi is plugged to HDMI1 (TV has 4) which is set to "Mode 3", description reads high def and fast frame rate.
There is also "Mode 2" saying only high def, and "Mode 1" supposed to be the most compatible with all.

There are also settings for HDR, but I am assuming they can have no effect on "blanking" issues.

**Part2)** 
I'll do the other parts "next day" after enough time of TV off

**Part 3)**
I have a laptop with HDMI but I am not sure the test will have any relevance.
Different cable (not the "mini-HDMI"), and it is running Ubuntu 20.04, hence on X11.
I could still do a test though!

The mini-HDMI cable was purchased at the same time with the Raspberry Pi from an "official" vendor: Kubii.

lurch, 2024-12-04:
> * Plug cable (TV side)
>   **Screen stays black** whatever mouse gesture

Hmm, so there's definitely something going wrong with your TV not detecting the Pi (because the TV stays black) and / or the Pi not detecting the TV (because `wlopm` says `HDMI-A-1 off` even when you have a working display).
Can you try the force_hotplug suggestion from https://github.com/raspberrypi/bookworm-feedback/issues/67#issuecomment-1765094899 and see if that improves things?

> I have a laptop with HDMI but I am not sure the test will have any relevance.

Sorry, I meant a different TV with an HDMI input, so you could check whether the Pi is actually outputting a signal or not. But that's not necessary now, as I believe we've found the culprit.

Bylon2, 2024-12-05:
Thanks for your suggestion, but there must be other things.
"hotplug" seems to be already there:

```
pi@raspberrypi:~ $ cat /boot/firmware/cmdline.txt
console=serial0,115200 console=tty1 root=PARTUUID=54d52123-02 rootfstype=ext4 fsck.repair=yes rootwait quiet splash plymouth.ignore-serial-consoles cfg80211.ieee80211_regdom=FR vc4.force_hotplug=1
```

As for other HDMI devices, I have 2 IImaya monitors that support HDMI on which I had the Pi plugged previously. No such syndrom was observed on those monitors, but that might not be relevant for the current issue.

lurch, 2024-12-05:
> "hotplug" seems to be already there:

Huh, you never mentioned that before. Do you see any different behaviour to the TV unplugging and replugging test with that line removed?

> No such syndrom was observed on those monitors, but that might not be relevant for the current issue.

It at least shows that this is specifically a problem with (the interaction with) your TV, and not a problem with your Pi :wink:

Bylon2, 2024-12-05:
> 
> Huh, you never mentioned that before. Do you see any different behaviour in the TV unplugging and replugging test with that setting removed?
> 
Yes, because I didn't know of this parameter in cmdline.txt!
Or maybe this is added when you set "Enable blanking" with raspi-config?

> 
> It at least shows that this is specifically a problem with (the interaction with) your TV, and not a problem with your Pi 😉
It seems so!

Well, so, having re-enabled "blanking" as it was with wayfire, tonight I did as usual: switched the TV on, it displayed a black screen, then moved the mouse and the Pi went out of blanking (presumably) and the image was there again.
There might also have been an interaction with Kodi since the TV labelled it as Kodi: I had previously launched Kodi as the "cumbersome workaround", then exited Kodi to the desktop.

So I did a clean reboot and will observe in the next days if "enabling blanking" with labwc solves this strange issue with this particular TV.

I will not use Kodi while I "observe" to see it this is fixed or at least "workarounded" with "blanking".

wlopm continues to say "off"... but with wayfire yesterday, while I was testing, one out of 10 times I tested wlopm I had one "on" without noticing anything different in the image displayed by the TV.

So let's say the report is _"suspended for further observation"_ for the moment, and I let you know if this is an OK workaroound!

lurch, 2024-12-05:
> Or maybe this is added when you set "Enable blanking" with raspi-config?

I've checked, and no `raspi-config` doesn't set that option.

> Yes, because I didn't know of this parameter in cmdline.txt!

Hmm, you mentioned you were using Kodi, so maybe it was something in there that set the option?

> There might also have been an interaction with Kodi since the TV labelled it as Kodi

I believe it's [CEC](https://en.wikipedia.org/wiki/Consumer_Electronics_Control) that allows the Pi to tell the TV what it should call the input-source.

> So let's say the report is "suspended for further observation" for the moment, and I let you know if this is an OK workaroound!

Okay! :+1: :crossed_fingers:

popcornmix, 2024-12-05:
> Hmm, you mentioned you were using Kodi, so maybe it was something in there that set the option?

No. `vc4.force_hotplug=1` is only ever manually entered. Nothing in the repo will set it.

Bylon2, 2024-12-07:
> No. `vc4.force_hotplug=1` is only ever manually entered. Nothing in the repo will set it.

I have no recollection of doing so, but maybe I did long ago as a workaround of something!

So far so good.
When I switch the TV on after "some time", I see "HDMI 1 - Video recording device 1", but when I move the mouse, it (presumably) triggers a transition out of blanking and the image is display again, with the TV saying "HDMI 1- Videocore"

I guess transitioning out of blanking is enough for the TV to catch "there is something" and display the image.
Apparently, Kodi is not needed (so far) to trigger CEC and a better dialogue.

I'll continue observing for sometime before closing the report.

The takeaway would be that counter-intuitively, "enabling blanking" can help in some "blanking" situations!..

Bylon2, 2024-12-12:
Comment is my last post above !

closed 2024-12-12 by Bylon2

lurch, 2024-12-12:
> Since the _black syndrome_ happens only when I disable blanking (counter-intuitively), having blanking enabled works for me as a valid workaround.

I dunno why having blanking disabled causes your "black screen syndrome", but it _does_ kinda make sense why having blanking enabled fixes it - I guess with blanking enabled, the Pi will be sending your TV an explicit signal to come out of blanking-mode (which it wouldn't be doing with blanking disabled), which your TV is obviously responding to.

Bylon2, 2024-12-13:
You are right, this is also what I suspect! My conclusion is that it is not worth trying to understand why the TV stays black as long as the "enable blanking" setting does the trick!