raspberrypi/bookworm-feedback #331: ADS7846 in.

noobmaster777, 2024-12-08:


noobmaster777, 2024-12-08:
my config.txt:
dtparam=i2c_arm=on
dtparam=spi=on
dtoverlay=max98357a
camera_auto_detect=1
#display_auto_detect=1
auto_initramfs=1
#dtoverlay=vc4-kms-v3d
#max_framebuffers=2
disable_fw_kms_setup=1
arm_64bit=1
disable_overscan=1
dtoverlay=piscreen,rotate=180,drm
dtparam=pciex1_gen3

Dmesg says this: Is this possibly related to the problem? 
[image]

lurch, 2024-12-08:
@6by9 Is `bookworm-feedback` the right place for this, or should it be transferred to the `linux` repo?

6by9, 2024-12-09:
It's a continuation of https://github.com/raspberrypi/bookworm-feedback/issues/88, which noobmaster777 tagged on the back end of.

It has reminded me of an issue not so long back where an interrupt line definition in an overlay missed out any pinctrl directives, and we ended up with the GPIO not actually connected to the pin. I'll try and refresh my memory on that one.

noobmaster777, 2024-12-09:
Thanks @6by9 i hope its a software problem so I dont have to change hardware. Warm regards!

noobmaster777, 2024-12-10:
> I've just tested my [piscreen clone](https://www.amazon.co.uk/ELEGOO-480x320-Raspberry-Monitor-Interface/dp/B01MRQTMTD) on an 8GB Pi5
> 
> Other than needing to add the `frequency=18000000` override to slow the SPI down slightly, I needed no changes. (Actually it seems stable at the default on this Pi, but I have had variation) As it's my main development system it happens to be on a 6.12 kernel, but I'm not expecting that to make any difference. Feel free to `sudo rpi-update rpi-6.12.y` to grab the test 6.12 kernel if you want to replicate.

Just tested on 6.12 via update and sadly its still not working. Do you mean 'speed=18000000'? Are you using the piscreen overlay?

noobmaster777, 2024-12-10:
After much investigation, I borrowed a friends 3.5 inch touch screen and discovered that both my screens are busted! Apologies for the hassle and thanks for the guidance. Closing this!

closed 2024-12-10 by noobmaster777