raspberrypi/bookworm-feedback #332: Latest `raspberrypi-ui-mods`: rasputin not working on X11 due to missing xinput

lrytz, 2024-12-10:
### Describe the bug

`sudo apt dist-upgrade` upgrades `raspberrypi-ui-mods`, it removes `lxinput` and installs `rasputin`.

`sudo apt autoremove` then removes `xinput`.

Starting `rasputin` then shows `sh: 1: xinput: not found` in the terminal. The mouse sensitivity can no longer be adjusted.

Installing `xinput` manually fixes it.

### Steps to reproduce the behaviour

see above

### Device (s)

Raspberry Pi 5

### System

```
m@pit:~ $ cat /etc/rpi-issue
Raspberry Pi reference 2024-11-19
Generated using pi-gen, https://github.com/RPi-Distro/pi-gen, 891df1e21ed2b6099a2e6a13e26c91dea44b34d4, stage4
```

```
m@pit:~ $ vcgencmd version
2024/11/12 16:10:44
Copyright (c) 2012 Broadcom
version 4b019946 (release) (embedded)
```

```
m@pit:~ $ uname -a
Linux pit 6.6.62+rpt-rpi-2712 raspberrypi/linux#1 SMP PREEMPT Debian 1:6.6.62-1+rpt1 (2024-11-25) aarch64 GNU/Linux
```

popcornmix, 2024-12-10:
Moved this issue as it has nothing to do with the linux kernel repo.

spl237, 2024-12-10:
Thanks for the report - fix pushed; should be in repo later today.

closed 2024-12-10 by spl237