raspberrypi/bookworm-feedback #333: HDMI/CEC physical address set to f.f.f.f when connected AV device switches HDMI input

ymakassiouk, 2024-12-13:
```
yuri@raspberrypi:~ $ uname -a
Linux raspberrypi 6.6.51+rpt-rpi-v7 #1 SMP Raspbian 1:6.6.51-1+rpt3 (2024-10-08) armv7l GNU/Linux
```

I had the HDMI "constellation" working with all the same devices in it, but on a much older Raspbian (can't really remember which now). The issue after installing Bookwork is that my Raspberry loses correct physical HDMI address when the AV receiver it is connected to deselects the respective HDMI input.

When the input is selected, I get the topology correctly reported by both cec-client and cec-ctl, I can send commands and many things work, including changing the routing on other devices. But when the input is changed (either by a command from raspberry or manually on the receiver),  if I run for example a cec-client to monitor HDMI traffic, here's what happens:

```
DEBUG:   [           41394]     CLinuxCECAdapterCommunication::Process - CEC_DQEVENT - CEC_EVENT_STATE_CHANGE - log_addr_mask=0000 phys_addr=ffff
DEBUG:   [           43407]     changing physical address to FFFF
DEBUG:   [           43407]     SetDevicePhysicalAddress - not setting invalid physical address ffff
```

The output is from libcec, but I think there is something below it that is wrong, because same consequences are experienced when I don't even have cec-utils installed, cec-ctl also starts showing physical address f.f.f.f.
And then nothing works any more, not until I switch the input back - and then it works again.

Why I don't think this is the intended behavior? There is a Chromecast sitting on the neighbour HDMI input on the same AV receiver, and when the receiver switches away from Chromecast's input, the Chromecast is still present in the HDMI topology reported by either libcec or cec-ctl on the Raspberry, i.e. Chromecast's physical address doesn't get screwed up.

I created an issue in libcec, but a) they aren't very active these days and b) like I said, I don't think it's their issue, but rather "something underlying", unfortunately I have no ide whom to talk about this to.

popcornmix, 2024-12-13:
This is deliberate behaviour: https://github.com/raspberrypi/linux/blob/80533a952218696c0ef1b346bab50dc401e6b74c/drivers/gpu/drm/vc4/vc4_hdmi.c#L458

As far as the driver knows, hotplug has gone, so hdmi sink is no longer attached, and maintaining the old physical address is questionable.

I think you'll need to find stronger evidence that the current behaviour is wrong (other than other device behaves differently). e.g. a reference from HDMI/CEC spec.

A workaround may be to add `vc4.force_hotplug=1` to cmdline.txt (on existing line) which should stop the driver from seeing the hotplug deassert, and so it won't invalidate the physical address.

popcornmix, 2024-12-13:
Might be worth reading the HDMI spec ("High-Definition Multimedia Interface"), which includes "Supplement 1 Consumer Electronics Control (CEC)"

ymakassiouk, 2024-12-13:
The problem is that even though Supplement 1 doesn't say anything about the unregistered device with physical address f.f.f.f being somehow cut off from everything, I haven't yet any seen any cec-ctl commands that would produce anything until the input is switched back to the Raspberry's input (and it gains valid physical and a logical address). What, in your experience, can cec-ctl send in order to re-activate itself?

popcornmix, 2024-12-13:
> What, in your experience, can cec-ctl send in order to re-activate itself?

This isn't something I've ever investigated.

ymakassiouk, 2024-12-13:
In the code that you refer to, the invalidation of physical address happens upon status becoming "disconnected" and that does look reasonable. Although, this apparently happens when there's no actual disconnection, the device physically remains a part of the HDMI topology (and should be able to continue sending CEC traffic - maybe that's not in the CEC specs, but I would argue it would make sense). Sorry, I know you didn't like this reference, but - the same way Chromecast does, when the selected input goes away from it.

popcornmix, 2024-12-13:
The code I showed occurs after a the hdmi sink deasserts the hotplug line.

In general that means the device is "unplugged".
But it's vendor specific exactly what a sink does with the hotplug line.

I'd say the normal behaviour is for hotplug to remain asserted when the AV input is changed, or the device is placed in standby.

But some devices deassert hotplug when placed in standy, and some when AV input is changed. I feel this behaviour is unhelpful, as there is no way to distinguish hotplug deasserted from hdmi cable is unplugged.

popcornmix, 2024-12-13:
> Sorry, I know you didn't like this reference, but - the same way Chromecast does, when the selected input goes away from it.

We'll need stronger evidence we are doing something wrong than there exists a device that does things differently that you prefer. On a quick search of linux kernel these drivers do the same as us (a cec phys_addr_invalidate when disconnected):
```
drivers/gpu/drm/bridge/adv7511/adv7511_drv.c
drivers/gpu/drm/bridge/synopsys/dw-hdmi.c-
drivers/gpu/drm/i915/display/intel_hdmi.c
drivers/gpu/drm/sun4i/sun4i_hdmi_enc.c
drivers/gpu/drm/tegra/output.c
drivers/gpu/drm/vc4/vc4_hdmi.c
```
We are unlikely to diverge from their behaviour If you want this changed, we'd need sufficient evidence that all these drivers should be change (e.g. a section from the spec that says the current behaviour is wrong).

ymakassiouk, 2024-12-13:
Right, I understand your position, thank you for explaining. I suppose this boils down to an older device having CEC implemented the way they thought was correct. I sure wonder what more modern Pioneer devices do, and what other manufacturers implement. 
However, for my purposes, it is completely fine that the Raspberry device will sit with `force_hotplug=1` as per your workaround suggestion. I tested this now and can confirm that it helps. When the input is deselected, my device remains on its physical address and is able to manage its surroundings (power up the receiver and switch it to the particular input). Thank you for the idea!

closed 2024-12-13 by ymakassiouk