raspberrypi/bookworm-feedback #334: wlopm --on fails to wake up display from screen blank

qrp73, 2024-12-13:
Previously I reported the same issue which always happens for any custom video mode: https://github.com/raspberrypi/bookworm-feedback/issues/321

The bug didn't fixed, but closed.

Now I catch the same issue with standard video mode. Usually it works ok. But once I catch the issue that I cannot wake up display from speed. This time I was used standard display mode (not custom). 

Since I remember how to wakeup the system from previous bug, I tried to wake up it through ssh remotely with executing wlr-randr and it helps. The next time it going to sleep and wake up with no issue.

So, it looks like there is floating bug which prevents to wake up the display even when it works in standard mode.

There is no error messages in the log. It happens when I connected USB3 high speed flash disk during display blank and then tried to wake up it with mouse and keyboard, so it may be related.

lurch, 2024-12-14:
> Usually it works ok. But once I catch the issue that I cannot wake up display from speed.

What does this mean?

qrp73, 2024-12-14:
> What does this mean?

Usually for standard video mode it going to sleep and wake up with no issue. But as described above, sometime it fails to wake up even when display setting is for standard video mode.

lurch, 2024-12-14:
It was the "wake up display from speed" that I was confused by, but now I realise that you probably meant "wake up display from sleep"?

> But once I catch the issue...

Have you got a reliable way of reproducing this, or is this just something that you saw once with no way to reproduce it? If it's the latter, then I'm afraid there will be no way for us to investigate this further.

qrp73, 2024-12-16:
just catch this issue again. 
And interestingly it happens again when I connected USB flash drive during display sleep.

Here is the system log, it seems there is no error messages:
```
[278713.706749] vc4-drm gpu: swiotlb buffer is full (sz: 524288 bytes), total 32768 (slots), used 112 (slots)
[280014.756923] bcm2835-codec bcm2835-codec: bcm2835_codec_flush_buffers: Timeout waiting for buffers to be returned - 0 outstanding
[366448.345923] sda: detected capacity change from 31277232 to 0
[366451.434896] usb 2-1: USB disconnect, device number 7
[366880.871276] usb 2-1: new SuperSpeed USB device number 8 using xhci_hcd
[366880.892257] usb 2-1: New USB device found, idVendor=0781, idProduct=5580, bcdDevice= 0.10
[366880.892276] usb 2-1: New USB device strings: Mfr=1, Product=2, SerialNumber=3
[366880.892283] usb 2-1: Product: Extreme
[366880.892289] usb 2-1: Manufacturer: SanDisk
```

> It was the "wake up display from speed" that I was confused by, but now I realise that you probably meant "wake up display from sleep"?

yes, I'm talking abot wake up display from sleep mode.

> Have you got a reliable way of reproducing this, or is this just something that you saw once with no way to reproduce it? If it's the latter, then I'm afraid there will be no way for us to investigate this further.

I'm not sure, but I catch it 2 times and both cases happened when I connected USB flash drive during display sleep and then tried to wake up display with mouse move.

lurch, 2024-12-16:
Thanks, I'll see if I can recreate this later. (It seems unlikely, but I guess it's possible that the disk-automount or "new drive inserted" notification-popups might be doing something weird?)