raspberrypi/bookworm-feedback #335: Systemd automount is trying to unmount storage when used

EdiDD, 2024-12-14:
systemd 252 (252.31-1~deb12u1)

```
[Automount]    
Where=/mnt/disk1
TimeoutIdleSec=10m
```

```
[Mount]
What=/dev/disk/by-uuid/...
Where=/mnt/disk1
```

When stream data are read by some process like watching movie from disk1 errors like below appears in log every 10m:

```
Failed unmounting mnt-disk1.mount
Failed to expire automount, ignoring: No such device
```

Systemd automount should try unmounting storage only if no other process is using it.

lurch, 2025-01-09:
So does anything actually go wrong, or is this just a warning message which you could ignore without any ill effects?
Are you sure that this is a bug specific to Raspberry Pi OS Bookworm, and not a bug with the upstream systemd project?

EdiDD, 2025-01-10:
It is not a warning. These are errors flooding log (but without any harmful consequences). I don't know if it is specific only to Raspberry Pi OS Bookworm. I have also submitted this bug report to upstream systemd.

lurch, 2025-01-14:
Are you able to provide more detailed reproduction steps please? What software are you using to play the movie? Which logfile is getting flooded with errors?
I'm not very familiar with systemd and automount, but I found [this guide](https://community.hetzner.com/tutorials/automount-filesystems-with-systemd) which got me going. As a test I tried setting the timeout to just 20 seconds, and I found that simply opening a terminal window and `cd`-ing into `/mnt/disk1` was enough to stop it automatically unmounting the disk.

> I have also submitted this bug report to upstream systemd.

Could you provide a link here to that bug-report? Thanks.