raspberrypi/bookworm-feedback #336: How to calibrate touch screens using two identical screens?

yuuuu-u, 2024-12-17:
**Prospect:**
I bought two screens from Waveshare, link: [https://www.waveshare.com/8dp-caplcd.htm](https://www.waveshare.com/8dp-caplcd.htm,) connected to Raspberry Pi 5, and used the following system kernel version:
`Linux raspberrypi 6.6.62+rpt-rpi-v8 #1 SMP PREEMPT Debian 1:6.6.62-1+rpt1 (2024-11-25) aarch64 GNU/Linux`

**Problem:**
I want to use these two screens in extended screen mode and map their touch functions to their respective touchpads. Since the two screens are identical, they are also recognized identically in the Touchscreen of Screen Configuration, which results in the touch of the two screens being mapped to only one of them after calibration. 
[image]
I tried to use the `xinput` command to calibrate the two screens, but the command could not recognize the touch device I was currently connected to

**Temporary solution:**
I used `raspi-config` to switch the system environment to X11, and the xinput calibration command can be used normally. The touch screens of the two screens can be mapped accordingly. 
However, a new problem has occurred. The VNC interface remains gray and cannot be used.

Could you please give me some suggestions on this issue? I look forward to receiving your reply soon.

lurch, 2024-12-17:
ping @spl237 and @6by9 to see if they have any ideas, but most of us are finishing for the Christmas holidays soon, so you may have to wait a while for a reply!

In your screenshot, even though both menu entries show "Waveshare Waveshare", are you able to tick the "upper" entry on one screen, and the "lower" entry on the other screen?

6by9, 2024-12-17:
It's a limitation of how USB HID (Human Interface Devices) advertise themselves, and the way libinput relies solely on the string name which isn't guaranteed to be unique.

I did look at this with https://forums.raspberrypi.com/viewtopic.php?t=376569. There's a slightly hacky kernel patch at https://github.com/raspberrypi/linux/pull/6356 which does make the names unique (as something like "usb-1-1 Waveshare Waveshare" and "usb-3-1 Waveshare WaveShare" in your case).

Ideally it would be fixed in the main libinput / wlroots / labwc layers though.
@cillian64 for any thoughts.

cillian64, 2024-12-17:
There's various prior art around disambiguating input device names in wlroots and compositors (eg. https://gist.github.com/tdaniel22/166301c48a2e50502ba34877f509dcfb).  But the screen config tool (arandr) gets its names from running `libinput list-devices`.  If we did this in libinput then I think just a single patch would cover everything, but I'm not sure if changing the libinput name would break anything else.

Let's wait until Simon is back in the new year and discuss with him.

closed 2025-09-22 by lurch