raspberrypi/bookworm-feedback #338: Problem using 4 USB ports at the same time: "could not open port"

Melek-Cherif, 2024-12-20:
Hello,
I am encountering a weird problem with my RPi communicating with 4 esp32 via USB serial communication. When using 3 or less esp32, the system works perfectly, but once I tried to use 4 ones at the same time, things got messy.

The system consists of an IoT device where the raspberry handle receives messages from esp32. So the code lists the ports and sends for each one of them a start message to start sending data once and then waits for the next start message.
At first the system worked for 2 or 3 cycles, then the transmission got slower (the transmission time went from 2 seconds to more than 12 seconds), and after that I received the error that I couldn't open the port as if the esp32 was not plugged. And it persists until I reboot the raspberry.
I think that there is some problem in the traffic of messages and an error accumulation that ends with shutting down the port. Also, I checked the power supply and the power delivered for every port, and it is sufficient and should work. I think there is something missing in the logic of transmission.

lurch, 2024-12-20:
(Transferred the issue to the `bookworm-feedback` repo, as it has nothing to do with documentation.)

After a reboot, and with all 4 devices attached, and before you start running your own code, what output do you get from the following commands:
```
cat /etc/os-release
```
```
uname -a
```
```
ls /dev/tty*
```
```
lsusb -v
```

And after you've run your code, and you've got the "could not open port" error, what does `dmesg` report? Also, what model of Raspberry Pi are you using?

Melek-Cherif, 2024-12-20:
I am using Raspberry Pi 4 model B. Here is what i got after following your advice
[image]

- Also, this is the output of `lsusb -v`:

[image]

- then after running the code it got here is what I got: **it runs perfectly for 4 or 5 times**

[image]

- Then it went slow, and after that, it got me this error:

[image]

- And this is the output from `dmesg`:

[image]

I don't find any logical explanation for this behavior!!

lurch, 2024-12-21:
Thanks for the logs :+1: 
I see that you're using Linux kernel 6.6.31, however the latest kernel in the Raspberry Pi OS repo is 6.6.62. So can you please do an [apt full-upgrade](https://www.raspberrypi.com/documentation/computers/os.html#manage-software-packages-with-apt), then reboot, and see if the error messages change at all?

Are you using a USB Hub, or are all the esp32 devices plugged directly into the USB ports on your Raspberry Pi 4?

Melek-Cherif, 2024-12-23:
Thank you for the advice but the problem still persists even after a full upgrade, Also, I am not using a USB Hub.

lurch, 2024-12-23:
Hmmm, I can't think of any obvious reason why the `cp210x` driver might work fine with three devices connected, but apparently become unreliable when four devices are connected.
Are the esp32 devices also all powered from the Raspberry Pi? Maybe with four devices plugged in at once the Pi 4B isn't able to supply enough current over the USB ports to keep all four devices active at once, and one or more of the devices are suffering from a brown-out? What PSU are you powering the Raspberry Pi with? In your original comment you said "Also, I checked the power supply and the power delivered for every port, and it is sufficient and should work." - could you provide more details about that? (I did a quick internet search and found https://lastminuteengineers.com/esp32-sleep-modes-power-consumption/ which says that a single ESP32 can draw between 240mA and 790mA, whereas https://www.raspberrypi.com/documentation/computers/raspberry-pi.html#typical-power-requirements says that Raspberry Pi 4 can supply a maximum of 1.2A over USB, which works out at 300mA per port.)
If you have a powered USB hub (e.g. https://www.raspberrypi.com/products/usb-3-hub/ in combination with https://www.raspberrypi.com/products/type-c-power-supply/ ) that might help track down whether it's a software-driver problem or a insufficient-power problem.

Melek-Cherif, 2024-12-24:
I think it's a software-driver problem because I was running another code that communicates with four esp32 with no problem, but each code is following a different approach. The first one is a listener that waits for messages from a microcontroller. But in the second approach, the microcontroller is in listening mode, and the raspberry controls the four of them by triggering a start message. As for the power supply, I am using a 5V, 3A raspberry charger, and the output of `lsusb -v` shows that the maximum power for the USB is 100 mA, and I checked that the esp32 consumes over 200 mA when she is executing a wifi-related task, but when executing a serial transmission task, it needs about 30 mA. That's why I think it is related to software and code management in the raspberry.

lurch, 2024-12-24:
I suspect that the 100mA reported by `lsusb` is **just** the current required for the CP210x UART->USB chip, and doesn't take the current-draw of the ESP32 into account too? I'm not saying that it _isn't_ a software-driver problem, but if you're able to power the ESP32s from a separate external power supply, whilst still connecting them to the Raspberry Pi 4's USB ports, that might be an interesting experiment to try? (don't forget to connect the GND from the external PSU to your Raspberry Pi's GND too)

> I am using 2GB RAM i will increase the memory and see what happens

You could try running `htop` in a separate terminal window whilst running your tests, to see if you're getting low on memory the longer your code is running for.

Melek-Cherif, 2024-12-25:
Thank you, @lurch, for this information. Actually, I changed the methodology of communication, and it became unidirectional. So the raspberry triggers the microcontrollers only once, and the esp32 program sends data repeatedly, which is different from my previous approach. Previously, I was triggering them each iteration. So I think with 4 USB ports and bidirectional transmissions, the ports couldn't manage the flow. Sadly, I still didn't get a logical explanation for this behavior, but I am happy that things are working right now.

closed 2024-12-25 by Melek-Cherif

lurch, 2024-12-26:
> Actually, I changed the methodology of communication, and it became unidirectional.

As far as USB is concerned, the connection is always "bidirectional", because it works on a request-response basis (i.e. the Raspberry Pi continually asks "give me your latest data" and the CP210x either replies with "I haven't got any data" or "here's the latest data").

> Sadly, I still didn't get a logical explanation for this behavior

If you're now triggering your ESP32 devices just once at startup, maybe the natural clock-drift on each device means that the readings from your devices are now being "interleaved" (i.e. never all happening at the same time), whereas previously (when you were triggering each iteration from the Pi) the readings from you devices were being "synchronised"?
I obviously don't have any hard evidence, but I still _suspect_ that a lack-of-power leading to one or more of the ESP32 devices resetting is a more likely explanation for why it "works fine" for 3 devices but "goes wrong" for 4 devices.

>  but I am happy that things are working right now.

Cool, good to hear that :+1: