raspberrypi/bookworm-feedback #340: Unable to open text file with encoding different than UTF8 in Geany editor

qrp73, 2024-12-21:
## Steps to reproduce

1) Save one of the following text files
2) Open with Geany editor
3) Make sure that encoding is incorrect (see screenshot how it should look)
4) Try to change encoding with menu Document->Set Encoding->East European-> select according to file name

**Expected result:** encoding is changed and text is properly displayed

**Actual result:** nothing happens. No way to read text files from files with encoding different than UTF8.

## Test files with different encodings

[cp-866.txt](https://github.com/user-attachments/files/18217386/cp-866.txt)
[cp-1251.txt](https://github.com/user-attachments/files/18217387/cp-1251.txt)
[cp-KOI8R.txt](https://github.com/user-attachments/files/18217388/cp-KOI8R.txt)
[cp-UTF8.txt](https://github.com/user-attachments/files/18217389/cp-UTF8.txt)

## Proper displaying

[image]

lurch, 2024-12-21:
Are you sure that this is a problem specific to Bookworm on Raspberry Pi OS, and not just a problem with the upstream Geany package in Debian Bookworm?

qrp73, 2024-12-23:
I'm not sure, but Geany has encoding selection menu and it appears that it don't works. Maybe due to old version in raspberry pi repository, maybe due to some encoding issue in OS itself, I don't know.

lurch, 2024-12-23:
I'll take a look when I'm back in the office next year...